autowarefoundation/autoware_universe #4505: `/planning/scenario_planning/trajectory` stops to generate trajectory while driving.

Kim-mins, 2023-08-03:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

Hello team,

I'm currently running Autoware on Carla, and I found that **sometimes the planning module does not generate a path and makes the ego vehicle stop.**

Here's a video for RViz: [[video](https://drive.google.com/file/d/1N40d4FtEIKhL1CpqO4kAOkE83qF8tHVR/view?usp=sharing)]

From 0:18 of the video, you can see that the trajectory is not updated and the car eventually stops after passing the last point of the trajectory.

### Expected behavior

I expected the trajectory to be generated well, so that the car could drive well without stuck.

### Actual behavior

The trajectory is not generated and the ego vehicle stops.

### Steps to reproduce

- Ros2 bag record file: [[rosbag](https://drive.google.com/drive/folders/1ihQTqVg5yolf1RVwYQBIWJbUtDEZZMTe?usp=sharing)]
- Map file: [[lanelet](https://drive.google.com/file/d/1vm9SvalJe7Bc9sh_8jK-0cPulVAov5uD/view?usp=drive_link)] [[pcd](https://drive.google.com/file/d/1MvJlfjkw3LWFUs5hdE_KQ2_CQTusU8UN/view?usp=drive_link)]

### Versions

I ran the experiment on the container from the image [here](https://github.com/autowarefoundation/autoware/pkgs/container/autoware-universe/70625594?tag=humble-20230215-prebuilt-cuda-amd64).
Carla version: 0.9.13

### Possible causes

After some investigation by myself, I found that the messages from the topic `/planning/scenario_planning/trajectory` is no longer generated.

maxime-clem, 2023-08-03:
It looks like one of the planning node crashes. Are you able to share your log files ? You can find usually find it in `$HOME/.ros/log/[[DATE_AND_USER]]/launch.log`.

Kim-mins, 2023-08-03:
Thank you for the response @maxime-clem!!

I could find many logs in the directory you mentioned! thank you!
But sadly, I ran the scenario about 2 weeks ago, so I cannot find the log file corresponds to the video.

So I just ran Autoware with the same scenario, and it made a similar situation.
Here's the log file: [[log](https://drive.google.com/file/d/14CHSW41UU2y-2VkkuKk20eZZsPPwXwkb/view?usp=sharing)].

maxime-clem, 2023-08-03:
The `lidar_centerpoint_node` dies first but it may not be related to the issue.
> 1691037289.4402609 [ERROR] [lidar_centerpoint_node-42]: process has died  ...

The issue comes from the motion planner crashing.
> 1691037432.1712468 [component_container_mt-59] terminate called after throwing an instance of 'std::invalid_argument'
> 1691037432.1713648 [component_container_mt-59]   what():  Points is empty.
> 1691037432.2907925 [ERROR] [component_container_mt-59]: process has died [pid 1064, exit code -6, cmd '/opt/ros/humble/lib/rclcpp_components/component_container_mt --ros-args -r __node:=motion_planning_container -r __ns:=/planning/scenario_planning/lane_driving/motion_planning -p use_sim_time:=True -p wheel_radius:=0.383 -p wheel_width:=0.235 -p wheel_base:=2.79 -p wheel_tread:=1.64 -p front_overhang:=1.0 -p rear_overhang:=1.1 -p left_overhang:=0.128 -p right_overhang:=0.128 -p vehicle_height:=2.5 -p max_steer_angle:=0.7'].

I think this crash has already been fixed with the latest version of universe.
Are you able to try a more recent version (the one you are using is 6 months old) ?

Kim-mins, 2023-08-03:
Thank you for more investigation!!
Yes sure. I'll try the latest version and tell you the result soon.
By the way, is the crash from my bad configurations or originated from the Autoware itself?
Maybe I should change my settings if the crash is from my side.

maxime-clem, 2023-08-03:
I think the crash is an issue with the Autoware code as the nodes should never crash.
You should not have to change any setting on your side.
If the issue still occurs with a more recent version, then there may be a bug that needs to be fixed.

Kim-mins, 2023-08-03:
I see. Thank you for the answer.
I'll try the latest version soon.

maxime-clem, 2023-08-07:
I am happy to hear that using the latest version solved the issue.
From the video and the log it does look like the same issue (some motion planning module crashes and the trajectory is no longer updated).

Kim-mins, 2023-08-08:
Thank you for answering!! I really appreciate to your help!
I'll close this issue since this is not the issue now.

closed 2023-08-08 by Kim-mins