autowarefoundation/autoware_universe #4538: The control_performance_analysis module has the phenomenon of Aborted (core dumped).

nainaigetuide, 2023-08-07:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

Use` ros2 launch control_performance_analysis controller_performance_analysis.launch.xml` for control performance analysis.
control_performance_analysis terminates when running close to target.

https://github.com/autowarefoundation/autoware.universe/assets/22877692/f678d3e0-06f3-4d7d-b9cf-cd5ddd04f619

### Expected behavior

control_performance_analysis work properly

### Actual behavior

```
[control_performance_analysis_exe-1] terminate called after throwing an instance of 'std::out_of_range'
[control_performance_analysis_exe-1]   what():  vector::_M_range_check: __n (which is 82) >= this->size() (which is 80)

```

### Steps to reproduce

1.  `ros2 launch autoware_launch planning_simulator.launch.xml map_path:=./autoware_map/sample-map-planning vehicle_model:=sample_vehicle sensor_model:=sample_sensor_kit`
2. `ros2 launch control_performance_analysis controller_performance_analysis.launch.xml`
3.  Set the start and end points, and run

### Versions

OS: 20.04
ROS2: Humble

maxime-clem, 2023-08-07:
Maybe related to https://github.com/autowarefoundation/autoware.universe/issues/4498

brkay54, 2023-08-10:
I will work on this.

brkay54, 2023-08-13:
@nainaigetuide Sorry I couldn't re-create the problem in my local. Can you provide me a test file for scenario_simulator? Link: https://tier4.github.io/scenario_simulator_v2-docs/

maxime-clem, 2024-03-12:
I will close this issue but feel free to reopen it if needed

closed 2024-03-12 by maxime-clem