autowarefoundation/autoware_universe #4680: roi pointcloud fusion for small objects detection

badai-nguyen, 2023-08-21:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

- Vehicle sensor configuration with low resolution LiDAR is limited to detect small objects such as traffic cones since too few points are hit the object and Euclidean cluster failure to cluster those objects. 
- The analysis shows that current roi_cluster_fusion on sample vehicle is only able detected traffic cones in range between 19m~49m in ideal flat road. It's narrower in the real test and clustering is applied after ground segmentation filter and compare map segmentation filter.

| cone size | IoU > 0 range | 
| - | - |
| [image] |[image]|

 
- [min_cluster_size: 10](https://github.com/autowarefoundation/autoware_launch/blob/2796d0a2115398f5479d95990dc72deabd09942d/autoware_launch/config/perception/object_recognition/detection/clustering/voxel_grid_based_euclidean_cluster.param.yaml#L6) could be change to small to improve small object clustering. However, Euclidean cluster FP unknown objects output also increase rapidly because of noise pointcloud.

### Purpose

The 2D image detection model like YOLOX may be strong to handle such unexpected small objects even in quite long distance. 
A new approach which more relies on  2D image detector result and use LiDAR cloud as validation as well as depth information could solve this issue.

### Possible approaches

Implement a fusion of pointcloud and unclassified ROIs .

### Definition of done

- Successful detected small objects such as traffic cones with less points cloud if recoginzed by 2d image detector.

miursh, 2023-11-10:
@badai-nguyen Is this issue already resolved? Or this includes current development item?

badai-nguyen, 2023-11-10:
@miursh I think the neccessary PRs for this issue are allready merged. The accuracy of this approach also is depend on the performace of Yolox-s model which may need more time to collect and annotate the data, and retrain the model. So I will closed this issue.

closed 2023-11-10 by badai-nguyen