autowarefoundation/autoware_universe #4686: No handling of MGRS grid borders in gnss_poser

kasper-helm, 2023-08-21:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

Currently the UTM to MGRS [conversion](https://github.com/autowarefoundation/autoware.universe/blob/main/sensing/gnss_poser/include/gnss_poser/convert.hpp#L161) in gnss_poser does not properly handle the case where you cross from one grid into another. I have seen cases where e.g. the mgrs.x position crosses the 100000m mark and then wraps around to 0m, creating a large jump. 

### Expected behavior

Pose output from gnss_poser should be smooth and continuous, without large jumps. 

### Actual behavior

gnss_poser outputs large jumps in position (100 km).

### Steps to reproduce

- Drive in an area where an MGRS grid line is crossed (e.g. I saw this happen when crossing from 10SEF to 10SFF, south of the SF Bay Area.
- Have a gnss unit running which is feeding lat/long data to gnss_poser, and the latter is set to use MGRS

### Possible causes

This seems to be handled properly in Eagleye/llh_converter https://github.com/MapIV/llh_converter/blob/main/src/llh_converter.cpp#L433
but it would be good to have it here as well since we encountered other issues that prevent us from using Eagleye.

VRichardJP, 2023-08-22:
One change I think I have proposed long ago is to move coordinate conversion logic to its own library, put coordinate system related params in a yaml file along with the PCD and lanelet2 maps, and load them as global parameters (like the vehicle description).

I have been using this internally for a while and I think it makes more sense: first because lanelet2 map and PCD map points are both expressed in a fixed coordinate system, so it makes little sense to infer e.g. MGRS code at runtime; second because it is then easy to add coordinate conversion features to other nodes (e.g. FMS interface node).

As a side effect, all map coordinates are always expressed in a fixed reference system (whether it is UTM, MGRS or arbitrary), so there is no wrap around issue (e.g you can have x=12345.0 and y=-123.45).

kminoda, 2023-08-22:
Hi kasper, 
I am in charge of this topic, and are now merging the patches to address such situation. Although we are having a bit of delay, I suppose we can merge all the patches within September.
This discussion may also be a help: https://github.com/orgs/autowarefoundation/discussions/3582

YamatoAndo, 2023-09-29:
@kminoda 
>  I suppose we can merge all the patches within September.

Is it done?

kminoda, 2023-10-01:
(@YamatoAndo Thanks for checking)

This issue has been overcome with multiple projection type support. If you are operating in some areas that overlaps multiple MGRS grids, you may use `LOCAL_CARTESIAN_UTM` or `TRANSVERSE_MARCATOR` as stated in https://github.com/autowarefoundation/autoware.universe/blob/main/map/map_projection_loader/README.md. By setting `map_projector_info.yaml` properly, the whole Autoware system now converts all the coordinates accordingly (except PCD file, you still need to prepare PCD in the desired coordinate beforehand). Note that eagleye is currently not supported, but will be shortly.

kminoda, 2023-10-11:
@kasper-helm I will close this issue since it was resolved as describe above. But feel free to open this again if you still have troubles. Thanks.

closed 2023-10-11 by kminoda