autowarefoundation/autoware_universe #4760: Error when the program(tensorrt_yolo) loads the Yolov5m6 model onnx

phoenixGG, 2023-08-26:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

Hello,I converted yolov5m6 to tensorrt onnx model, but the autoware tensorrt_yolo program reported an error after starting it, does the current version of AUTOWARE not support Yolov5m6 model (size: 1280)?
[image]

### Expected behavior

tensorrt_yolo work well

### Actual behavior

tensorrt_yolo report error when loading Yolov5m6 model onnx file

### Steps to reproduce

1.tensorrt_yolo report error when loading Yolov5m6 model onnx file
2.
3.

miursh, 2023-08-28:
@manato @wep21 
Do you have any ideas?

wep21, 2023-08-28:
@phoenixGG The onnx file you have provide via email seems to be broken. Could you check if the onnx file can be loaded by some tool such as `netron`.
[image]

phoenixGG, 2023-08-28:
> @phoenixGG The onnx file you have provide via email seems to be broken. Could you check if the onnx file can be loaded by some tool such as `netron`. [image]

I'm gonna send it again in a minute.

wep21, 2023-08-28:
I have requested the file access.

wep21, 2023-09-09:
I resolved the issue in the conversation via email.

closed 2023-09-09 by wep21