autowarefoundation/autoware_universe #4851: fix(ndt_scan_matcher): skipping_publish_num does not switches error state

yvzksgl, 2023-09-01:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

In the timer_diagnositcs() warning condition of the skipping_publish_num is incomplete. It just checks is skipping_publish_num greater than 1, but it should check also is skipping_publish_num smaller than 5.

### Expected behavior

When skipping_publish_num is greater or equal to 5 ndt_scan_matcher should be in error state.

### Actual behavior

When skipping_publish_num is greater or equal to 5 ndt_scan_matcher stays in warning state.

### Steps to reproduce

It can be reproduced with setting skipping_publish_num  to a value that greater than 5 

### Additional context

I have an also feature request which i will create an issue for it and i have ready to create pr's both of them.

YamatoAndo, 2023-09-04:
When it's 5 or more, it will properly enter an error state.
https://github.com/autowarefoundation/autoware.universe/blob/main/localization/ndt_scan_matcher/src/ndt_scan_matcher_core.cpp#L277-L279

yvzksgl, 2023-09-05:
https://github.com/autowarefoundation/autoware.universe/pull/4853