autowarefoundation/autoware_universe #4912: ar_tag_based_localizer: could not find file <aruco/aruco.hpp>

xianglunkai, 2023-09-07:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

--- stderr: ar_tag_based_localizer                                                                                                                                                                                             
In file included from /home/xlk/autoware/src/universe/autoware.universe/localization/ar_tag_based_localizer/src/ar_tag_based_localizer_node.cpp:45:
/home/xlk/autoware/src/universe/autoware.universe/localization/ar_tag_based_localizer/include/ar_tag_based_localizer/ar_tag_based_localizer_core.hpp:54:10: fatal error: aruco/aruco.h: 没有那个文件或目录
   54 | #include <aruco/aruco.h>
      |          ^~~~~~~~~~~~~~~
compilation terminated.

### Expected behavior

success

### Actual behavior

error

### Steps to reproduce

colcon build --symlink-install

SakodaShintaro, 2023-09-07:
Thanks for the report.

Is the `aruco` library installed in your environment?

If it has been installed, you should get the results shown in the following command.

```bash
$ dpkg -l | grep ros-humble-aruco
ii  ros-humble-aruco                                  5.0.0-1jammy.20230721.204844            amd64        The ARUCO Library has been developed by the Ava group of the Univeristy of Cordoba(Spain).
```

If not installed, try the following command in the `autoware` directory.

```
source /opt/ros/humble/setup.bash
rosdep install -y --from-paths src --ignore-src --rosdistro $ROS_DISTRO
```

I hope this solves the problem.

xianglunkai, 2023-09-07:
@SakodaShintaro 
Thank you very much!

xianglunkai, 2023-09-07:
May I ask if the planning module is considering introducing a pilot project for reinforcement learning？
Txs!

SakodaShintaro, 2023-09-07:
I'm glad you got it to work.

> May I ask if the planning module is considering introducing a pilot project for reinforcement learning？

There is no particular one that can be introduced to you.

If you are able to make such a thing, even a simple one, please introduce it to us at https://github.com/orgs/autowarefoundation/discussions/categories/show-and-tell :+1:

xianglunkai, 2023-09-07:
rosdep install -y --from-paths src --ignore-src --rosdistro $ROS_DISTRO
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
common_sensor_launch: Cannot locate rosdep definition for [nebula_sensor_driver]

nebula_sensor_driver?

SakodaShintaro, 2023-09-08:
Sorry I'm not familiar with that package.

As for everything I know about it

(1) `rosdep update` may help you

(2) `nebula_sensor_driver` is related to https://github.com/tier4/nebula.

The latest `autoware.repos` will load it.

https://github.com/autowarefoundation/autoware/blob/0e5ec3109c752b3cd92deaaad4c0b51b3b8d2cf1/autoware.repos#L78-L81

xianglunkai, 2023-09-08:
Thank you again.

Also, I am very familiar with the Apollo.auto algorithm. Do you think universe has any advantages over Apollo?

SakodaShintaro, 2023-09-08:
Sorry, I am aware of Apollo.auto, but I am not familiar enough with its contents to make a direct comparison.

One of the general advantages of autoware is its connection to peripheral software services.

* [AWSIM](https://tier4.github.io/AWSIM/GettingStarted/QuickStartDemo/) : open source simulator
* [web.auto](https://web.auto/ja/) : development and vehicle management tool as a paid service of TIER IV
* [fanfare](https://solutions.tier4.jp/fanfare) : a reference of vehicle configurations that autoware can work

and so on.

We think that autoware is still lacking in many areas in terms of performance when viewed as a stand-alone software for autonomous mobility. We are constantly developing. (I myself am mainly in charge of developing the localization function)

closed 2023-09-11 by xianglunkai