autowarefoundation/autoware_universe #4915: Vehicle can not move because steering is not controlable when stopped

brkay54, 2023-09-07:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

In vehicle_cmd_gate, if the vehicle is in the stopped state it sends the current steering to the low level. However, this causes a bug for the longitudinal controller because of keep_stopped_until_steering_converge parameter. 

Because of this, in planning_simulator, we can not control the steering when velocity is zero in the manual controller.

I think moving keep_stopped_steering_convergence parameter from pid_controller to vehicle_cmd_gate is a good option to fix it.

### Expected behavior

The vehicle is moving when it is in autonomous mode.

### Actual behavior

The vehicle is not moving when it is in autonomous mode.

### Steps to reproduce

1. open planning simulator
2. add manual controller panel
3. try to control steering when velocity is 0

closed 2024-02-09 by brkay54

feng123a52, 2024-07-26:
请问使用禾赛的激光雷达实车在外部传入自己的点云数据，autoware出现了这么个错，The pointcloud layout is not compatible with PointXYZIRCAEDT/PointXYZIRC. Aborting该如何解决

knzo25, 2024-07-26:
@feng123a52 
Ever since https://github.com/autowarefoundation/autoware.universe/pull/6996
we use a different pointcloud type. You could either modify Hesai drivers or use Nebula (tier4's driver).
Also, please create new issues/discussions and if you want to ease communications, I highly recommend sticking to English