autowarefoundation/autoware_universe #4978: How to integrate the official ouster ros2 driver into autoware.universe project

TZECHIN6, 2023-09-13:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

I would like to integrate my ouster lidar (os-1 64 beam I believe) into autoware.universe project. However, I am having trouble to get the pointcloud data after the preprocessing node.

I first tried to pass the raw point cloud into the self cropped box node as below:
```
    cropbox_parameters = create_parameter_dict("input_frame", "output_frame")
    cropbox_parameters["negative"] = True

    vehicle_info = get_vehicle_info(context)
    cropbox_parameters["min_x"] = vehicle_info["min_longitudinal_offset"]
    cropbox_parameters["max_x"] = vehicle_info["max_longitudinal_offset"]
    cropbox_parameters["min_y"] = vehicle_info["min_lateral_offset"]
    cropbox_parameters["max_y"] = vehicle_info["max_lateral_offset"]
    cropbox_parameters["min_z"] = vehicle_info["min_height_offset"]
    cropbox_parameters["max_z"] = vehicle_info["max_height_offset"]

    nodes.append(
        ComposableNode(
            package="pointcloud_preprocessor",
            plugin="pointcloud_preprocessor::CropBoxFilterComponent",
            name="crop_box_filter_self",
            remappings=[
                ("input", "/ouster/points"),
                ("output", "self_cropped/pointcloud"),
            ],
            parameters=[cropbox_parameters],
            extra_arguments=[{"use_intra_process_comms": LaunchConfiguration("use_intra_process")}],
        )
    )
  ```

The output topic of /sensing/lidar/top/self_cropped/pointcloud has no response.

At this point, the issue might caused by the missing of required data field, thus I tried to just directly pass into `concatenate_data` as below:

```
def launch_setup(context, *args, **kwargs):
    # set concat filter as a component
    concat_component = ComposableNode(
        package="pointcloud_preprocessor",
        plugin="pointcloud_preprocessor::PointCloudConcatenateDataSynchronizerComponent",
        name="concatenate_data",
        remappings=[
            ("~/input/twist", "/sensing/vehicle_velocity_converter/twist_with_covariance"),
            ("output", "concatenated/pointcloud"),
        ],
        parameters=[
            {
                "input_topics": [
                    "/ouster/points",
                    "/sensing/lidar/left/outlier_filtered/pointcloud",
                    "/sensing/lidar/right/outlier_filtered/pointcloud",
                ],
                "output_frame": LaunchConfiguration("base_frame"),
            }
        ],
        extra_arguments=[{"use_intra_process_comms": LaunchConfiguration("use_intra_process")}],
    )
```

Sadly, no luck for me to see the pointcloud at topic `/sensing/lidar/concatenated/pointcloud`...

I have checked the raw points topic `/ouster/points`, and ensure it has data output and can be seen in rviz.
[... 219 lines cut ...]
    y: 0.0
    z: 0.0
    roll: 0.0
    pitch: 0.0
    yaw: 0.0
```

4. I havent integrated the ouster into autoware workspace yet, so right now I first launching ouster, then launching autoware.launch

### Versions

- OS: Ubuntu 22
- ROS2 Humble
- Autoware universe (`0e5ec3109c752b3cd92deaaad4c0b51b3b8d2cf1` 2023-08-08)
- ouster ros2 driver [link](https://github.com/ouster-lidar/ouster-ros/tree/ros2)

### Possible causes

Might due to unmatch data field and also tf issue.

As the ouster driver package is now installed at other workspace, it doesnt launch with node_container. This might also an issue... but I have difficult time to create a custom node_container. 

### Additional context

Any rosbag or information is needed to further debug this issue, please free to let me know, I would try my best to provided the required information.

VRichardJP, 2023-09-14:
The problem might comes from the data layout:
```
fields:
- name: x
  offset: 0
  datatype: 7
  count: 1
- name: y
  offset: 4
  datatype: 7
  count: 1
- name: z
  offset: 8
  datatype: 7
  count: 1
- name: intensity
  offset: 16
  datatype: 7
  count: 1
- name: t
  offset: 20
  datatype: 6
  count: 1
- name: reflectivity
  offset: 24
  datatype: 4
  count: 1
- name: ring
  offset: 26
  datatype: 4
  count: 1
- name: ambient
  offset: 28
  datatype: 4
  count: 1
- name: range
  offset: 32
  datatype: 6
  count: 1
```

Point cloud data layout is defined here: https://autowarefoundation.github.io/autoware-documentation/main/design/autoware-architecture/sensing/data-types/point-cloud/#point-cloud-fields

Which is now implemented here: https://github.com/tier4/nebula/blob/main/nebula_common/include/nebula_common/point_types.hpp

```c++
struct PointXYZIRCAEDT
{
  float x;
  float y;
  float z;
  std::uint8_t intensity;
  std::uint8_t return_type;
  std::uint16_t channel;
  float azimuth;
  float elevation;
  float distance;
  std::uint32_t time_stamp;
};
```

As you can see, the layout is not exactly the same. There are some fields you don't have, and some you 
 have that don't exist.

Unfortunately, [the new nebula module seems not to support Ouster Lidar (yet)](https://github.com/tier4/nebula/tree/main#supported-sensors), so you need to modify the driver you have (or add a conversion node) so you get the same fields than above.

Not all fields are equaly important. The most important are 'x', 'y', 'z' and 'time_stamp'. In your case you have a time field, but it is named 't'. You should also check what is the format used by this field and make sure it is what Autoware expect. AFAIK, 'channel', 'azimuth' and 'distance' are only used by the ring outlier filter (which is not that important so you could remove it), and the other fields are not used.

TZECHIN6, 2023-09-14:
Thanks for the guidance, I will try modify the driver to match the data field. 

Sorry but i have a question about `'channel', 'azimuth' and 'distance' are only used by the ring outlier filter `, as I saw the graph saying each node input points msg are all required `XYZIRC`, channel seem cannot be ignored. Also, for the  first cropbox filter node, `XYZIRCADT` points is needed, does this mean I have to prepare all data field at the very beginning?

If i have three lidars combo, left-right velodyne lidar and top ouster lidar, whats the least required field to directly pass ouster lidar points to `Cloud Transformer` and `Cloud Concatenator`?

VRichardJP, 2023-09-14:
It is better to follow the expected format. That said, if you remove the ring outlier filter, I think you can fill fields you don't have with 0.

TZECHIN6, 2023-09-15:
Understood. I am trying to work on that now

isouf, 2024-02-26:
@TZECHIN6 may I please ask if you managed to overcome this issue? I am facing the same issue and I believe it comes from the field type differences.

TZECHIN6, 2024-02-27:
Currently, I don't have an ouster lidar with me. I will resume this integration task once I borrow one from the University.

In the meantime, you could check this out to see the difference between the published pointcloud type structure [link](https://github.com/ros-drivers/ros2_ouster_drivers/blob/humble/ros2_ouster/include/ros2_ouster/client/point.h).

Remark: there is no nubela driver when at the time I use autoware, so things might change right now.

TZECHIN6, 2024-02-28:
@isouf 

https://github.com/ouster-lidar/ouster-ros/issues/217

actually I have asked related question in ouster's repository, and seem they have merged a feature to allow selecting different presentation of point cloud type.

Please have a look at here: https://github.com/ouster-lidar/ouster-ros/pull/216

I haven't checked it out in detail. I will come back for this after receiving the lidar.

isouf, 2024-02-28:
@TZECHIN6 thank you very much for your prompt response. I managed to resolve the issue by creating a node to convert the format from [Ouster](https://github.com/ros-drivers/ros2_ouster_drivers/blob/humble/ros2_ouster/include/ros2_ouster/client/point.h#L18) to [Autoware](https://github.com/elm-europe-am/elm_sensor_kit_launch/blob/isouf/ds_dev/elm_sensor_kit_launch/launch/sensing.launch.xml#L47). However, I need to investigate the ring outlier results further because they do not seem to be sensible. For now I have disabled the ring outlier.

TZECHIN6, 2024-02-29:
@isouf  Great to hear from you about that. Actually I do not have experience on formatting the point cloud type before, would you mind sharing a bit how you able to do such transform? I would be very appreciated.

isouf, 2024-02-29:
@TZECHIN6 
Here is the _"quick"_  method that I made to convert the pointcloud format. Any improvements recommendations are welcomed. 

```cpp
// Receive the pointcloud from ouster lidar and change the format of the message so that it matches
// the format the is used in autoware pointcloud_preprocessor filters
void PointcloudFormatAdaptorNode::onPointcloud(
  const sensor_msgs::msg::PointCloud2::ConstSharedPtr pcd)
{
  // Initiate output messages
  auto pointcloud_msg = std::make_shared<sensor_msgs::msg::PointCloud2>();

  // Initiate pcl pointcloud message for the input point cloud
  pcl::PointCloud<pointcloud_format_adaptor::PointXYZITRRAR>::Ptr input_pointcloud(
    new pcl::PointCloud<pointcloud_format_adaptor::PointXYZITRRAR>);

  // Convert ros to pcl
  pcl::fromROSMsg(*pcd, *input_pointcloud);

  // Initiate pcl pointcloud message for the input point cloud
  pcl::PointCloud<nebula::drivers::PointXYZIRADT>::Ptr output_pointcloud(
    new pcl::PointCloud<nebula::drivers::PointXYZIRADT>);
  output_pointcloud->reserve(input_pointcloud->points.size());

  // Convert pcl ouster to pcl autoware
  nebula::drivers::PointXYZIRADT point{};
  for (const auto & p : input_pointcloud->points) {
    point.x = p.x;
    point.y = p.y;
    point.z = p.z;
    point.intensity = p.intensity;
    point.return_type = 0;
    point.ring = p.ring;
    point.azimuth = std::atan2(p.y, p.x);
    point.distance = p.range;
    point.time_stamp = p.t;
    output_pointcloud->points.emplace_back(point);
  }
  output_pointcloud->header = input_pointcloud->header;
  output_pointcloud->height = 1;
  output_pointcloud->width = output_pointcloud->points.size();

  // Convert pcl to ros
  pcl::toROSMsg(*output_pointcloud, *pointcloud_msg);

  // Publish updated pointcloud message
  pub_pcd_->publish(*pointcloud_msg);
}
```

The `PointXYZITRRAR` point type is defined on a separate header file as follows:

```cpp
struct PointXYZITRRAR
{
  float x;
  float y;
  float z;
  float intensity;
  std::uint32_t t;
  std::uint16_t reflectivity;
  std::uint16_t ring;
  std::uint16_t ambient;
  uint32_t range;
  EIGEN_MAKE_ALIGNED_OPERATOR_NEW
};
```

TZECHIN6, 2024-03-01:
@isouf  Thanks for your sharing! I will definitely have a look about it.

If there is any update or improvement I have made, I will let you know for sure.

DarrenQu, 2024-05-31:
@isouf  Thanks for you sharing! Quick question, where in the autoware.universe code do you add this node and how do you run this node?

DarrenQu, 2024-05-31:
@isouf  To follow up my last questions. Cause I'm using Ouster LiDAR with the following topic: 
[image]
However, autoware requires the following topic:
[image]

So, where in the autoware.universe code do you integrate the converter node?

isouf, 2024-06-01:
@DarrenQu you will need to replace the [default Velodyne drivers](https://github.com/autowarefoundation/sample_sensor_kit_launch/blob/main/sample_sensor_kit_launch/launch/lidar.launch.xml#L14) with Ouster and then feed the `PointCloud2` message (output of Ouster drivers) directly to the [`pointcloud_preprocessos`](https://github.com/autowarefoundation/sample_sensor_kit_launch/blob/main/sample_sensor_kit_launch/launch/pointcloud_preprocessor.launch.py#L40).

skill121790, 2024-06-26:
@isouf Could you perhaps share what you lidar.launch.xml file looks like? I have incorporated a node that changes the format from ouster to autoware, and I have confirmed that it is outputting data in the correct format. I changed the input topics in pointcloud_preprocessor.launch.py to the output topic of my converter node, but I am still not getting any output from /sensing/lidar/concantenated/pointcloud

TZECHIN6, 2024-07-31:
> @DarrenQu you will need to replace the [default Velodyne drivers](https://github.com/autowarefoundation/sample_sensor_kit_launch/blob/main/sample_sensor_kit_launch/launch/lidar.launch.xml#L14) with [...]

@isouf if I directly pass the raw pointcloud from the modified driver to the pointcloud_proprocessors like the way you did, then what the nebula driver does? as seem like the raw pointcloud should firstly be published to the cropbox filter...outlier ring filter, then concatenated at the end. Right now seem it bypass all those preprocess steps.

knzo25, 2024-07-31:
@TZECHIN6 
The point type that autoware expects is now this one
https://github.com/autowarefoundation/autoware.universe/blob/14fa0f3692551ec6af460284537473c5c9be6d66/common/autoware_point_types/include/autoware_point_types/types.hpp#L95
So you may need to take extra steps

TZECHIN6, 2024-07-31:
> @TZECHIN6 
> 
> The point type that autoware expects is now this one
> 
> https://github.com/autowarefoundation/autoware.universe/blob/14fa0f3692551ec6af460284537473c5c9be6d66/common/autoware_point_types/include/autoware_point_types/types.hpp#L95
> 
> So you may need to take extra steps

@knzo25 thanks for the update! I have switched to Livox lidar recently, and successfully publish the required point cloud structure after modifying the official driver.

However when blocking me now is the integration to the point cloud preprocessing component container nodes, I have posted a question about it in the discussion https://github.com/orgs/autowarefoundation/discussions/5048

I know it's a bit off the topic of this issue.

FranzAlbers, 2024-07-31:
I've adapted the method from @isouf a bit for the current Autoware PointXYZIRCAEDT point type and built a package around it so that one can directly use a composable node to subscribe to the PointCloud from the Ouster driver and provide the output to the Crop Box Filter (and the Autoware point cloud preprocessing pipeline):
https://github.com/FranzAlbers/ouster_point_type_adapter

subramanians1, 2024-12-02:
Hello @FranzAlbers ,
I have been trying to use the package in https://github.com/FranzAlbers/ouster_point_type_adapter  to convert ouster messages however i an running into the following error:

fatal error: autoware_point_types/types.hpp: No such file or directory
7 | #include "autoware_point_types/types.hpp"

I tried to include it in the sensor_component/external directory of autoware and also in a separate workspace with ouster driver. i run into the same error in both cases. Could you please advice how i should build this package?

FranzAlbers, 2024-12-02:
Hello @subramanians1, 

Autoware changed the namespace for point_types. I've pushed the changes to the repo and it should be building correctly now.

subramanians1, 2024-12-02:
Thanks for your prompt action @FranzAlbers . That indeed fixed the issue.

TZECHIN6, 2024-12-05:
@subramanians1  Have you rebuild the project or at least `colcon build` the new created description package for your ouster? As the error statement state the package not found.

subramanians1, 2024-12-05:
Yeah...I rebuilt autoware