autowarefoundation/autoware_universe #4998: Unable to init pose with RViz button after latest pull

yaukaizhi, 2023-09-14:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

I'm unable to give an initial pose in Rviz when running with use_sim_time=true and logging_simulation for system_run_mode. 

Log file here:
[
[launch.log](https://github.com/autowarefoundation/autoware.universe/files/12607855/launch.log)
](url)

### Expected behavior

Init pose using Rviz

### Actual behavior

Unable to initialize localization even after giving init pose

### Steps to reproduce

Use sim time = true
System run mode = logging_simulation
Give init pose using Rviz initialpose button
Use a rosbag to test

YamatoAndo, 2023-09-14:
@yaukaizhi Hi!
When checking the log, I noticed that it contains the message `[localization.pose_estimator.ndt_scan_matcher]: No InputSource.`
This occurs when the ndt_scan_matcher is unable to subscribe to LiDAR point clouds.
So please check if LiDAR point clouds are being subscribed correctly.

yaukaizhi, 2023-09-14:
May I know where the Lidar topic is for localization? It has been moved around a few times haha

YamatoAndo, 2023-09-14:
The ndt_scan_matcher subscribes to `/localization/util/downsample/pointcloud`.
You can know which topic it is subscribing to by using the following command.
```
$ ros2 node info /localization/pose_estimator/ndt_scan_matcher 
/localization/pose_estimator/ndt_scan_matcher
  Subscribers:
    /clock: rosgraph_msgs/msg/Clock
    /localization/kinematic_state: nav_msgs/msg/Odometry
    /localization/pose_twist_fusion_filter/biased_pose_with_covariance: geometry_msgs/msg/PoseWithCovarianceStamped
    /localization/util/downsample/pointcloud: sensor_msgs/msg/PointCloud2
    /parameter_events: rcl_interfaces/msg/ParameterEvent
    /sensing/gnss/pose_with_covariance: geometry_msgs/msg/PoseWithCovarianceStamped
```

Alternatively, please refer to [node-diagram](https://autowarefoundation.github.io/autoware-documentation/main/design/autoware-architecture/node-diagram/).

VRichardJP, 2023-09-14:
I have faced a similar problem recently after the switch to nebula. I would first check the `/sensing/lidar/*` topics for missing data (e.g. `ros2 topic hz` or `ros2 topic info -v`).

yaukaizhi, 2023-09-15:
ros2 topic hz /lidar_topic shows 10Hz when playing my rosbag. But once Autoware launches, ros2 topic hz /lidar_topic says the topic isn't being published. What could cause this?

YamatoAndo, 2023-09-15:
@yaukaizhi Could you share your rosbag? I will try it.

yaukaizhi, 2023-09-15:
Here you go, I realised it's only when perception in autoware_launch is set to true
https://1drv.ms/u/s!Apo0xLUwYEZdmVClyQuu04hhSK72?e=5P9zYw

YamatoAndo, 2023-09-15:
@yaukaizhi 
> I realised it's only when perception in autoware_launch is set to true
Is this isuse resolved?

YamatoAndo, 2023-09-15:
It's not ideal, but I was able to execute the initial position estimation using the following method.

terminal1
`ros2 launch autoware_launch logging_simulator.launch.xml map_path:=$HOME/map/sample_map vehicle_model:=sample_vehicle sensor_model:=sample_sensor_kit`

termainal2
`ros2 run tf2_ros static_transform_publisher 0 0 0 0 0 0 base_link velodyne`

termainal3
`ros2 bag play mranti_full.db3 --clock 200 --remap /points_raw:=/sensing/lidar/top/outlier_filtered/pointcloud`

And give an initial pose from Rviz's 2DPoseEstimate

closed 2023-09-25 by yaukaizhi