autowarefoundation/autoware_universe #5128: tf2_autoware_auto_msgs.hpp:60:13: error: redefinition of ‘void tf2::doTransform(const T&, T&, const TransformStamped&) [with T = geometry_msgs::msg::Point32_<std::allocator<void> >; geometry_msgs::msg::TransformStamped = geometry_msgs::msg::TransformStamped_<std::allocator<void> >]’

L-Net-1992, 2023-09-26:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

[image]

### Expected behavior

build error

### Actual behavior

build error

### Steps to reproduce

1.vcs import src < autoware.repos
2.vcs pull src
3.source /opt/ros/humble/setup.bash
4.rosdep install -y --from-paths src --ignore-src --rosdistro $ROS_DISTRO
5.colcon build --symlink-install --cmake-args -DCMAKE_BUILD_TYPE=Release

### Versions

Ubuntu22.04
Humble
46f871fdc6df53f9dce615ce8a132e3d3d139c14

SakodaShintaro, 2023-09-26:
The same error occurs in my environment.
This is caused by a recent update to ros2/geometry2.
https://github.com/ros2/geometry2/compare/4597504ef9138f780c877f7b83b1f7fcd806dc34...a73160df358c52d1f506fce21f15f8243e9b87f4

There are pull requests created by wep21-san to deal with this issue. I hope this information is helpful.
https://github.com/autowarefoundation/autoware.universe/pull/5127
https://github.com/autowarefoundation/autoware.universe/pull/5089

L-Net-1992, 2023-09-26:
@SakodaShintaro I feel that there are some dependencies besides issues. You see, the underlying dependencies are ROS2, which should be version updates, and some are bug free. Because this package has not been updated for two months.

SakodaShintaro, 2023-09-27:
Yes, I agree that Autoware has various problems around package versioning.
If you have any good ideas, please feel free to suggest them on the [discussion page](https://github.com/orgs/autowarefoundation/discussions).

In my understanding,  this issue has been resolved since the pull request was merged.
If the problem still remains, please re-open.

closed 2023-09-27 by SakodaShintaro