autowarefoundation/autoware_universe #5154: Recurrent Crashes in goal_planner: Segmentation Fault Linked to freespace_planning_algorithms::AstarNode Deallocation

kyoichi-sugahara, 2023-09-27:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

While executing the goal_planner, the program crashes due to a segmentation fault. Based on the stack trace, the issue seems to arise when an std::unordered_map, holding values of type freespace_planning_algorithms::AstarNode, is being deallocated.

(I will share the photo screenshot later)

### Expected behavior

The intended behavior is for the nodes to remain alive, and for freespace_planning_algorithms::AstarNode to successfully generate a path to the goal and reach it without issues.

### Actual behavior

It's not guaranteed to be reproducible 100% of the time when generating paths using freespace_planning_algorithms, but after several repetitions, the node eventually crashes.

Here is the stack trace:

```
Program terminated with signal SIGSEGV, Segmentation fault.
#0  __gnu_cxx::new_allocator<std::__detail::_Hash_node<std::pair<unsigned int const, freespace_planning_algorithms::AstarNode>, false> >::deallocate (__t=1, __p=0x7f1cb8926eb0, this=0x7f1ce82e3778) at /usr/include/c++/11/ext/new_allocator.h:132
132       deallocate(_Tp* __p, size_type __t __attribute__ ((__unused__)))
[Current thread is 1 (Thread 0x7f1d32506540 (LWP 669228))]
(gdb) bt
#0  __gnu_cxx::new_allocator<std::__detail::_Hash_node<std::pair<unsigned int const, freespace_planning_algorithms::AstarNode>, false> >::deallocate (__t=1, __p=0x7f1cb8926eb0, this=0x7f1ce82e3778) at /usr/include/c++/11/ext/new_allocator.h:132
#1  std::allocator_traits<std::allocator<std::__detail::_Hash_node<std::pair<unsigned int const, freespace_planning_algorithms::AstarNode>, false> > >::deallocate (__n=1, __p=0x7f1cb8926eb0, __a=...) at /usr/include/c++/11/bits/alloc_traits.h:496
#2  std::__detail::_Hashtable_alloc<std::allocator<std::__detail::_Hash_node<std::pair<unsigned int const, freespace_planning_algorithms::AstarNode>, false> > >::_M_deallocate_node_ptr (__n=0x7f1cb8926eb0, this=0x7f1ce82e3778)
    at /usr/include/c++/11/bits/hashtable_policy.h:1905
#3  std::__detail::_Hashtable_alloc<std::allocator<std::__detail::_Hash_node<std::pair<unsigned int const, freespace_planning_algorithms::AstarNode>, false> > >::_M_deallocate_node (__n=0x7f1cb8926eb0, this=0x7f1ce82e3778)
    at /usr/include/c++/11/bits/hashtable_policy.h:1895
#4  std::__detail::_Hashtable_alloc<std::allocator<std::__detail::_Hash_node<std::pair<unsigned int const, freespace_planning_algorithms::AstarNode>, false> > >::_M_deallocate_nodes (__n=0x7f1cba511170, this=0x7f1ce82e3778)
    at /usr/include/c++/11/bits/hashtable_policy.h:1916
#5  std::_Hashtable<unsigned int, std::pair<unsigned int const, freespace_planning_algorithms::AstarNode>, std::allocator<std::pair<unsigned int const, freespace_planning_algorithms::AstarNode> >, std::__detail::_Select1st, std::equal_to<unsigned int>, std::hash<unsigned int>, std::__detail::_Mod_range_hashing, std::__detail::_Default_ranged_hash, std::__detail::_Prime_rehash_policy, std::__detail::_Hashtable_traits<false, false, true> >::clear (this=0x7f1ce82e3778)
    at /usr/include/c++/11/bits/hashtable.h:2320
#6  std::_Hashtable<unsigned int, std::pair<unsigned int const, freespace_planning_algorithms::AstarNode>, std::allocator<std::pair<unsigned int const, freespace_planning_algorithms::AstarNode> >, std::__detail::_Select1st, std::equal_to<unsigned int>, std::hash<unsigned int>, std::__detail::_Mod_range_hashing, std::__detail::_Default_ranged_hash, std::__detail::_Prime_rehash_policy, std::__detail::_Hashtable_traits<false, false, true> >::~_Hashtable (this=0x7f1ce82e3778, 
    __in_chrg=<optimized out>) at /usr/include/c++/11/bits/hashtable.h:1532
#7  std::unordered_map<unsigned int, freespace_planning_algorithms::AstarNode, std::hash<unsigned int>, std::equal_to<unsigned int>, std::allocator<std::pair<unsigned int const, freespace_planning_algorithms::AstarNode> > >::~unordered_map (
    this=0x7f1ce82e3778, __in_chrg=<optimized out>) at /usr/include/c++/11/bits/unordered_map.h:102
#8  freespace_planning_algorithms::AstarSearch::~AstarSearch (this=0x7f1ce82e3550, __in_chrg=<optimized out>)
    at /home/kyoichi-sugahara/workspace/pilot-auto.awf-latest/src/autoware/universe/planning/freespace_planning_algorithms/include/freespace_planning_algorithms/astar_search.hpp:106
#9  freespace_planning_algorithms::AstarSearch::~AstarSearch (this=0x7f1ce82e3550, __in_chrg=<optimized out>)
    at /home/kyoichi-sugahara/workspace/pilot-auto.awf-latest/src/autoware/universe/planning/freespace_planning_algorithms/include/freespace_planning_algorithms/astar_search.hpp:106
#10 0x00007f1d09e33cfb in std::default_delete<freespace_planning_algorithms::AbstractPlanningAlgorithm>::operator() (
    __ptr=<optimized out>, this=<optimized out>) at /usr/include/c++/11/bits/unique_ptr.h:79
#11 std::unique_ptr<freespace_planning_algorithms::AbstractPlanningAlgorithm, std::default_delete<freespace_planning_algorithms::AbstractPlanningAlgorithm> >::~unique_ptr (this=0x7f1ce805edc8, __in_chrg=<optimized out>)
    at /usr/include/c++/11/bits/unique_ptr.h:361
#12 behavior_path_planner::FreespacePullOver::~FreespacePullOver (this=0x7f1ce805e990, __in_chrg=<optimized out>)
    at /home/kyoichi-sugahara/workspace/pilot-auto.awf-latest/src/autoware/universe/planning/behavior_path_planner/include/behavior_path_planner/utils/goal_planner/freespace_pull_over.hpp:35
#13 behavior_path_planner::FreespacePullOver::~FreespacePullOver (this=0x7f1ce805e990, __in_chrg=<optimized out>)
    at /home/kyoichi-sugahara/workspace/pilot-auto.awf-latest/src/autoware/universe/planning/behavior_path_planner/include/behavior_path_planner/utils/goal_planner/freespace_pull_over.hpp:35
#14 0x00007f1d09cb5538 in std::default_delete<behavior_path_planner::PullOverPlannerBase>::operator() (
    __ptr=<optimized out>, this=<optimized out>) at /usr/include/c++/11/bits/unique_ptr.h:79
--Type <RET> for more, q to quit, c to continue without paging--c
#15 std::unique_ptr<behavior_path_planner::PullOverPlannerBase, std::default_delete<behavior_path_planner::PullOverPlannerBase> >::~unique_ptr (this=0x7f1ce8055950, __in_chrg=<optimized out>) at /usr/include/c++/11/bits/unique_ptr.h:361
#16 behavior_path_planner::GoalPlannerModule::~GoalPlannerModule (this=0x7f1ce80552a0, __in_chrg=<optimized out>) at /home/kyoichi-sugahara/workspace/pilot-auto.awf-latest/src/autoware/universe/planning/behavior_path_planner/include/behavior_path_planner/scene_module/goal_planner/goal_planner_module.hpp:111
#17 0x00007f1d09cb560d in behavior_path_planner::GoalPlannerModule::~GoalPlannerModule (this=0x7f1ce80552a0, __in_chrg=<optimized out>) at /home/kyoichi-sugahara/workspace/pilot-auto.awf-latest/src/autoware/universe/planning/behavior_path_planner/include/behavior_path_planner/scene_module/goal_planner/goal_planner_module.hpp:111
#18 0x000055da144e6832 in std::_Sp_counted_base<(__gnu_cxx::_Lock_policy)2>::_M_release() ()
#19 0x00007f1d09b7d605 in std::__shared_count<(__gnu_cxx::_Lock_policy)2>::~__shared_count (this=0x7f1cf402d5d8, __in_chrg=<optimized out>) at /usr/include/c++/11/bits/shared_ptr_base.h:705
#20 std::__shared_ptr<behavior_path_planner::SceneModuleInterface, (__gnu_cxx::_Lock_policy)2>::~__shared_ptr (this=0x7f1cf402d5d0, __in_chrg=<optimized out>) at /usr/include/c++/11/bits/shared_ptr_base.h:1154
#21 std::shared_ptr<behavior_path_planner::SceneModuleInterface>::~shared_ptr (this=0x7f1cf402d5d0, __in_chrg=<optimized out>) at /usr/include/c++/11/bits/shared_ptr.h:122
#22 std::_Destroy<std::shared_ptr<behavior_path_planner::SceneModuleInterface> > (__pointer=0x7f1cf402d5d0) at /usr/include/c++/11/bits/stl_construct.h:151
Program terminated with signal SIGSEGV, Segmentation fault.
#0  __gnu_cxx::new_allocator<std::__detail::_Hash_node<std::pair<unsigned int const, freespace_planning_algorithms::AstarNode>, false> >::deallocate (__t=1, __p=0x7f1cb8926eb0, this=0x7f1ce82e3778) at /usr/include/c++/11/ext/new_allocator.h:132
132       deallocate(_Tp* __p, size_type __t __attribute__ ((__unused__)))
[Current thread is 1 (Thread 0x7f1d32506540 (LWP 669228))]
(gdb) bt
[... 32 lines cut ...]
#18 0x000055da144e6832 in std::_Sp_counted_base<(__gnu_cxx::_Lock_policy)2>::_M_release() ()
[... cut ...]
3. At some point, I encountered the problem described in this issue.

### Possible causes

- `mutex` in the goal_planner, which runs in multi-threading, may not be operating correctly.
- During the execution of freespace_planning_algorithms, there is a release of memory in the planner manager module.
- There might be a standalone bug in Astar.

### Additional context

This issue results in an occasional crash of the node, affecting the reliability of the path planning process, and thereby requires prompt attention and resolution.

kyoichi-sugahara, 2023-09-27:
@NorahXiong 
Hello, and sorry for sudden mention.
I haven't been able to identify the cause yet, and I apologize if this turns out to be unrelated, but I was wondering if there might be a possibility that the issue is being influenced by this [PR](https://github.com/autowarefoundation/autoware.universe/pull/2040)? If this change is found to be unrelated, I sincerely apologize for any inconvenience.

NorahXiong, 2023-09-29:
> @NorahXiong Hello, and sorry for sudden mention. I haven't been able to identify the cause yet, and I apologize if this turns out to be unrelated, but I was wondering if there might be a possibility [...]

I tried many times but the crash never happened and no clue was found in the related code, Is there any special step not mentioned in the reproduce steps?
[image]

NorahXiong, 2023-09-29:
> Thanks for the issue! Similar problems occurred with both `start/goal planner`. The logs are a little different, but I think it is the same kind of bug.
> 
> It seems to be caused by the fact that `open_list_` sometimes contains `0x0`. [image]

I think the 0x0 elements are in the underlying wrapping data structure (vector) rather than in the queue.  It would not be very likely that the empty pointers are pushed into the queue as the pointers have all been visited before being pushed.

kyoichi-sugahara, 2023-10-02:
@NorahXiong 
Thank you so much for the reponse.
I tried to reproduce problem again and I successfully reproduced the probelem(It's very difficult to reproduce problem though...)
The reproducibility is not perfect. 
The situation is 
- goal is inside of bus
- ego vehicle is stucked
- change currnet pose

[goal_planner_node_die.webm](https://github.com/autowarefoundation/autoware.universe/assets/32741405/ed44c234-9f1a-496f-bf1d-be025be9d43d)

kosuke55, 2023-10-02:
@NorahXiong 
Thanks for the reply and for trying!
In my case, I was able to reproduce it by putting the goal many times.

https://github.com/autowarefoundation/autoware.universe/assets/39142679/aad32ff6-d14f-4b5d-90f4-c6cf6d9fdb67

kosuke55, 2023-12-14:
@NorahXiong 
We have not been able to fix this issue.  We are sorry to bother you, but could you please try to reproduce it? :pray:

kosuke55, 2023-12-14:
@VRichardJP 
I think you are familiar with memory, so if you could advise me on this I would appreciate it. :bow:

VRichardJP, 2023-12-14:
It looks like a concurrency issue, as the program seems to crash at different places:
- In the [report video](https://github.com/autowarefoundation/autoware.universe/issues/5154#issue-1914884971), the segfault is in `Astar::search()`
- In the [report backtrace](https://github.com/autowarefoundation/autoware.universe/issues/5154#issue-1914884971), the segfault is in `~Astar()`
- In the [last video](https://github.com/autowarefoundation/autoware.universe/issues/5154#issuecomment-1743054799), it happens again in `Astar::search()`

Did you try to run the program with valgrind? e.g. with `launch-prefix="gnome-terminal -- valgrind --tool=memcheck --leak-check=yes "`

kosuke55, 2023-12-14:
@VRichardJP 
thanks for the advice. We did use valgrind so we will try!

VRichardJP, 2023-12-14:
I am not sure I totally understand how the modules are running, the `StartPlannerModule` runs its own callback queue for the `FreespacePullOut` here:
https://github.com/autowarefoundation/autoware.universe/blob/22522261d381cf445ab9cf119cdbc8e79c1f14f1/planning/behavior_path_start_planner_module/src/start_planner_module.cpp#L79

Is my understanding correct?
- behavior path planner manager create/destroy planning modules at runtime depending on the situation.
- the manager calls the modules `run()` function one after the other.

Then, what happens to the callback created by `FreespacePullOut`? I am not sure what is the behavior when the default callback queue is used. Maybe the module share the same callback queue than the manager. In such case the timed callbacks are mutually exclusive and the freespace pull out timed callback cannot run at the same time than the manager.

But here, the callback is running in a different queue, so you may have a thread running inside `FreespacePullOut` timed callback while the manager is trying to destroy it, or updating some internal data.

For instance, what happens if you put a sleep right before the `planFreespacePath()` line here:

https://github.com/autowarefoundation/autoware.universe/blob/22522261d381cf445ab9cf119cdbc8e79c1f14f1/planning/behavior_path_start_planner_module/src/start_planner_module.cpp#L99-L101

kosuke55, 2023-12-15:
@VRichardJP 
Thank you very much for your very detailed look!! 

> Is my understanding correct?
・behavior path planner manager create/destroy planning modules at runtime depending on the situation.
・the manager calls the modules run() function one after the other.
Obviously, both don't happen at the same time.

yes, your understanding is correct.

The manager deletes or std::move modules depending on the situation.
For example, when a new route is received, the module instance is cleared. 
https://github.com/autowarefoundation/autoware.universe/blob/feat/avoidance_pull_over/planning/behavior_path_planner/src/behavior_path_planner_node.cpp#L379

If FreespacePullOut is running in a separate thread at this time, it is possible that the data could be rewritten and crashed.

So, I feel that locking the clear from manager while FreespacePullOut's callback is running, or as a separate instance of FreespacePullOut (building a server), etc. might be a solution.

> For instance, what happens if you put a sleep right before the planFreespacePath() line here:
[autoware.universe/planning/behavior_path_start_planner_module/src/start_planner_module.cpp](https://github.com/autowarefoundation/autoware.universe/blob/22522261d381cf445ab9cf119cdbc8e79c1f14f1/planning/behavior_path_start_planner_module/src/start_planner_module.cpp#L99-L101)
Lines 99 to 101 in [2252226](https://github.com/autowarefoundation/autoware.universe/commit/22522261d381cf445ab9cf119cdbc8e79c1f14f1)
 if (isStuck() && is_new_costmap) { 
   planFreespacePath(); 
 } 
I guess it will crash right away.For instance, what happens if you put a sleep right before the planFreespacePath() line here:

I would like to confirm this as well, but I wonder if it dies during the planFreespacePath process and then sleeps before that process? Is the intention to generate a time delay so that clearing is more likely to occur during the planFreespacePath process ?Specifically, should I perform the same reproduction method by doing the following?

```
 if (isStuck() && is_new_costmap) { 
   std::this_thread::sleep_for(std::chrono::seconds(10));
   planFreespacePath(); 
 } 
```

VRichardJP, 2023-12-15:
@kosuke55 
Yes, if the issue is what I think it is, then while you have one thread sleeping before `planFreespacePath()` the behavior path planner manager will continue doing its work:
```C++
 if (isStuck() && is_new_costmap) { 
   std::this_thread::sleep_for(std::chrono::seconds(10));
   planFreespacePath(); 
 } 
```

In particular, if you reset the goal in that 10s window, the freespace object (or things it is refering to) are likely to be destroyed/moved, and I guess you will get some sort of segmentation fault.

lilyrosecarnation, 2023-12-18:
> @NorahXiong We have not been able to fix this issue. We are sorry to bother you, but could you please try to reproduce it? 🙏

Have you found out the reason leading to the segmentation fault? I'm sorry I have to try it again later if you still need.

NorahXiong, 2023-12-18:
> @NorahXiong We have not been able to fix this issue. We are sorry to bother you, but could you please try to reproduce it? 🙏

Have you found out the reason leading to the segmentation fault? I'm sorry I have to try it again later if you still need.

kosuke55, 2024-01-31:
@NorahXiong 
Sorry for the delay. No we have not been able to proceed with any analysis yet.

lilyrosecarnation, 2024-02-05:
@kosuke55 @kyoichi-sugahara 
I tried again by following your tips but still no segmentation fault occurred. Here is the [video link](https://drive.google.com/file/d/1oV4lulN7c-UkADtgVmEfENOtAwYBfvDf/view?usp=sharing).
Any suggestions to help me reproduce the bug?

kosuke55, 2024-02-05:
@NorahXiong 
Thank you very much for trying again.
ego vehicle is needed to be in parking_lot to run FreespacePullOver().
parking_lot is light yellow area and the red rectangle is expample of ego positon.

[image]

kosuke55, 2024-02-05:
@NorahXiong 
oh, sorry currently it hve to be close enough to goal to execute the goal planner, and the braking distance determines that distance. If this is made large enough, it will be triggered regardless of the braking distance.

https://github.com/tier4/autoware_launch/blob/awf-latest/autoware_launch/config/planning/scenario_planning/lane_driving/behavior_planning/behavior_path_planner/goal_planner/goal_planner.param.yaml#L45

```
minimum_request_length: 100.0
```

And the goal is needed to be put in the `road_shoulder`

kosuke55, 2024-02-05:
@NorahXiong 

As @VRichardJP indicated, sleep could easily produce a crash.

```
  if (isStuck() && is_new_costmap && needPathUpdate(path_update_duration)) {
    std::this_thread::sleep_for(std::chrono::seconds(10));
    planFreespacePath();
  }
```

https://github.com/autowarefoundation/autoware.universe/assets/39142679/a4ba48ac-545c-4d6c-b466-7bd82806bb80

kosuke55, 2024-02-05:
https://github.com/autowarefoundation/autoware.universe/pull/6322 may fix the issue, we will test more

NorahXiong, 2024-02-06:
@kosuke55 
I followed your steps by
1. setting minimum_request_length to 100.0;
2. setting sleeping time before freespace planning;
3. setting the goal on the road shoulder.
Test result: No crash occurred. 

Information may help you confirm the cause:
1. My latest tests(today and yesterday) were taken in native environment(Ubuntu22.04).
2. My previous tests were taken in a docker environment and the lane-driving related node died once or twice during tests.

[image]
[image]

https://github.com/autowarefoundation/autoware.universe/assets/103234047/23f0d8d7-cd7f-41a0-ada3-83aa91adc20d