autowarefoundation/autoware_universe #5221: behavior_planning_container process died after the second goal reached

annb3, 2023-10-04:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

I have download the last updates /main branch of autoware where I insert my maps, lanelet, vehicle and sensor. 
In the previous version everything works fine, but now, launching the planning simulation, the behavior_planning_container died after have reached the second goal, so the new goal can be set, but the AUTO button isn't enable, the trajectory seems to be identified but not completely set and starting to publish ERROR on the terminal related to the not met departure condition.

[image]

[image]

From now on the vehicle is freezed and nothing can be done to avoid this situation.
It is not a problem of vehicle or sensor because it happens also with the sample-vehicle and sample-sensor and with your map.

### Expected behavior

I expected a different behavior, exactly as in the previous version of the planning: where the vehicle move and reach the goal in the simulation without problem or constraints. It is able also to go out of lane in the parking phase and going out from the park slot. 
Also trying to reset the initial position nothing changes, the system crashed.

### Actual behavior

Now the system crashed after the vehicle has reached one/two goals and nothing can be do to avoid it even if I try to reset the initial position.

### Steps to reproduce

1. launch the docker container: `rocker --nvidia --x11 --network host --devices /dev/vid* --user --volume $HOME/autoware --volume $HOME/autoware_map --volume /dev/shm/:/dev/shm -- ghcr.io/autowarefoundation/autoware-universe:latest-cuda &&
export RCUTILS_COLORIZED_OUTPUT=1 &&
export RCUTILS_CONSOLE_OUTPUT_FORMAT="[{severity} {time}] [{name}]: {message} ({function_name}() at {file_name}:{line_number})"`
2. `vcs import src < autoware.repos &&
sudo apt update &&
rosdep update &&
rosdep install -y --from-paths src --ignore-src --rosdistro $ROS_DISTRO`
3. launch the planning simulation
4. set the initial position and goal poses more than once after have reached the previous goal

### Versions

- OS: Ubuntu 20.04.6 LTS
- ROS 2: HUMBLE
- Autoware: docker developer download - your branch is up to date with 'origin/main'
-  Docker container launched as: `rocker --nvidia --x11 --network host --devices /dev/vid* --user --volume $HOME/autoware --volume $HOME/autoware_map --volume /dev/shm/:/dev/shm -- ghcr.io/autowarefoundation/autoware-universe:latest-cuda &&
export RCUTILS_COLORIZED_OUTPUT=1 &&
export RCUTILS_CONSOLE_OUTPUT_FORMAT="[{severity} {time}] [{name}]: {message} ({function_name}() at {file_name}:{line_number})"`

### Possible causes

Not finished to develop the planning stack?

greensky0802, 2023-11-07:
you can try to change RMW to cyclone dds

annb3, 2023-11-07:
Thanks a lot @greensky0802, now it is more stable, thanks also to the pushed updates (planning package related), so it doesn't crash anymore for this reason! 
I can close the issue.

closed 2023-11-07 by annb3

luojiaxiang11, 2024-06-14:
> Thanks a lot @greensky0802, now it is more stable, thanks also to the pushed updates (planning package related), so it doesn't crash anymore for this reason! I can close the issue.

Hello, have you not encountered this issue again after that? My autoware 1.0 version still encounters this issue.