autowarefoundation/autoware_universe #5230: No route generated when using 2 way lanelet map when robot is facing backwards

yaukaizhi, 2023-10-05:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

On a lanelet map with oneway = no in vector map builder. Robot can go from A to B, but not from B to A. Route not generated. Has this issue been faced before? 

### Expected behavior

Able to go bothways

### Actual behavior

Only goes oneway

### Steps to reproduce

Set oneway = no in vector map builder
Set goal from A to B, then B to A

idorobotics, 2023-10-05:
Are you referring to driving in reverse?

xmfcx, 2023-11-02:
Hello @yaukaizhi ,

Thank you for bringing this to our attention. After some further investigation, I'd like to clarify that, as of now, Autoware does not support bi-directional lanes. This limitation is also mentioned in a post by @mitsudome-r on [this GitHub issue](https://github.com/autowarefoundation/autoware.universe/issues/1850#issuecomment-1252611247).

I understand that this might not be the answer you were hoping for. As a potential workaround, you could consider dividing the lane into two separate one-way lanes, if feasible for your use case. We'll also keep track of this request and consider it for future updates.

I will also ask the Planning team for their thoughts on bi-directional lane support and workarounds.

@TakaHoribe @mehmetdogru could you let us know if this is in the roadmap?
Also, is there a workaround for bi-directional lane support in current Autoware?

mehmetdogru, 2023-11-02:
@xmfcx 

This topic has been worked on in past by LeoDrive engineers but not concluded yet and due to lack of sources we cannot work on this feature in near future. We can bring the topic up to see is anybody up to work on this task.

And I agree with Mitsudome-san that the best way to go at the moment if somebody want to implement bi-directional road is to annotate 2 overlapping but opposite directional lanes.

xmfcx, 2023-11-02:
Thank you @mehmetdogru for your insights.

In the interest of keeping our issues organized and actionable, I propose closing this issue for now. However, if in the future, anyone is keen on implementing this feature, I'd encourage opening a new, dedicated issue for the feature request. This will ensure that we can track and manage it more effectively.

closed 2023-11-02 by xmfcx