autowarefoundation/autoware_universe #5252: The frequency of crop box filter node is lower than normal

beginningfan, 2023-10-08:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

I am using a lidar which is robosense-m1. There are many NaN points in its pointcloud.
The input frequency of `crop box node` is 10hz, but the output is about 6hz.
It is related to #3112
[image]

### Expected behavior

The output frequency is as same as the input frequency

### Actual behavior

The output frequency is much lower than the input frequency.
After after commenting the `RCLCPP_WARN` in [here](https://github.com/autowarefoundation/autoware.universe/blob/83821d4818cf83ea3bcb14511fc64482859ed7f1/sensing/pointcloud_preprocessor/src/crop_box_filter/crop_box_filter_nodelet.cpp#L141C7-L141C18), the output frequency is normal

### Steps to reproduce

Launch Autoware
Run recorded bag file

### Versions

Ubuntu 22.04
Humble
Autoware Universe : main

### Possible causes

There are too many NaN points. Frequent printing leads to excessive system usage
[image]

### Additional context

We should use `RCLCPP_WARN_ONCE` or `RCLCPP_INFO_THROTTLE` instead of `RCLCPP_WARN`

VRichardJP, 2023-10-08:
Yes! to add a print in the main loop was not a bright idea.

Throttling could do, but I would rather count the number of rejected points and move the warning outside the loop. If you are not in a rush, I can do propose a PR next week.

On another topic, I don't think it is normal to have so many NaNs in your data :)

beginningfan, 2023-10-08:
@VRichardJP There is a pcd file of lidar rs-m1. It has 4743 NaN points in 78750 points. Overall, point clouds can also be used normally. As a comparison, there are also 2117 zero values in the files of Oster64.
[pcd.zip](https://github.com/autowarefoundation/autoware.universe/files/12840102/pcd.zip)

VRichardJP, 2023-10-12:
@beginningfan 
I have created a PR that should fix your issue. Could you confirm?

beginningfan, 2023-10-12:
@VRichardJP  Thanks for your PR. It works well.
I think [there](https://github.com/VRichardJP/autoware.universe/blob/2bc2257566bae2a3d69cfc6215caa566298fcdbf/sensing/pointcloud_preprocessor/src/crop_box_filter/crop_box_filter_nodelet.cpp#L170) should be 100 instead of 1000. The frequency of the pointcloud is 10Hz, so printing once per second seems a bit slow

VRichardJP, 2023-10-12:
I added the throttle to avoid spamming the output/log at 10Hz.

The point of the message is to warn the user there may be a problem with its data. I think 1 Hz is enough so the warning won't be unnoticed. Still, I don't want to bother the user too much if they decide to ignore the warning.

Or maybe I'm overthinking? Printing for each input cloud is fine to me too (I don't have NaNs in my sensor data anyway). What do you think?

beginningfan, 2023-10-13:
@VRichardJP You are right. 1Hz is good enouth

PR #5283 (2023-10-19): perf(pointcloud_preprocessor): move print out of hot loop
## Description

fixes #5252

## Tests performed

@beginningfan

## Effects on system behavior

None

## Pre-review checklist for the PR author

The PR author **must** check the checkboxes below when creating the PR.

- [x] I've confirmed the [contribution guidelines].
- [x] The PR follows the [pull request guidelines].

## In-review checklist for the PR reviewers

The PR reviewers **must** check the checkboxes below before approval.

- [x] The PR follows the [pull request guidelines].

## Post-review checklist for the PR author

The PR  [... cut]
closed 2023-10-19 by miursh