autowarefoundation/autoware_universe #5254: test_tf2_autoware_auto_msgs : compile error

xianglunkai, 2023-10-09:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

It seems to that some functions have not defined!

### Expected behavior

No problems

### Actual behavior

--- stderr: autoware_auto_tf2                                                                                          
/usr/bin/ld: CMakeFiles/test_tf2_autoware_auto_msgs.dir/test/test_tf2_autoware_auto_msgs.cpp.o: in function `Tf2AutowareAuto_DoTransformPoint32_Test::TestBody()':
test_tf2_autoware_auto_msgs.cpp:(.text+0x2c7): undefined reference to `void tf2::doTransform<geometry_msgs::msg::Point32_<std::allocator<void> > >(geometry_msgs::msg::Point32_<std::allocator<void> > const&, geometry_msgs::msg::Point32_<std::allocator<void> >&, geometry_msgs::msg::TransformStamped_<std::allocator<void> > const&)'

### Steps to reproduce

 checkout current main branch：
colcon build --symlink-install --cmake-args -DCMAKE_BUILD_TYPE=Release

### Versions

OS: ubuntu22.04

### Possible causes

l

VRichardJP, 2023-10-12:
I could fix the issue by reverting https://github.com/autowarefoundation/autoware.universe/pull/5232.
@taikitanaka3 any idea?

taikitanaka3, 2023-10-12:
@VRichardJP
if i am right, since this issue is occured without upgrading humble, so sudo apt upgrade will solve this.

VRichardJP, 2023-10-12:
Yes indeed, I get the following compilation error without #5232 after upgrading the `ros-humble-tf2-*` packages:

```
In file included from /home/sig/autoware/src/universe/autoware.universe/common/autoware_auto_tf2/test/test_tf2_autoware_auto_msgs.cpp:17:
/home/sig/autoware/src/universe/autoware.universe/common/autoware_auto_tf2/include/autoware_auto_tf2/tf2_autoware_auto_msgs.hpp:60:13: error: redefinition of ‘void tf2::doTransform(const T&, T&, const TransformStamped&) [with T = geometry_msgs::msg::Point32_<std::allocator<void> >; geometry_msgs::msg::TransformStamped = geometry_msgs::msg::TransformStamped_<std::allocator<void> >]’
   60 | inline void doTransform(
      |             ^~~~~~~~~~~
In file included from /home/sig/autoware/src/universe/autoware.universe/common/autoware_auto_tf2/include/autoware_auto_tf2/tf2_autoware_auto_msgs.hpp:36,
                 from /home/sig/autoware/src/universe/autoware.universe/common/autoware_auto_tf2/test/test_tf2_autoware_auto_msgs.cpp:17:
/opt/ros/humble/include/tf2_geometry_msgs/tf2_geometry_msgs/tf2_geometry_msgs.hpp:260:6: note: ‘void tf2::doTransform(const T&, T&, const TransformStamped&) [with T = geometry_msgs::msg::Point32_<std::allocator<void> >; geometry_msgs::msg::TransformStamped = geometry_msgs::msg::TransformStamped_<std::allocator<void> >]’ previously declared here
  260 | void doTransform(
      |      ^~~~~~~~~~~
In file included from /home/sig/autoware/src/universe/autoware.universe/common/autoware_auto_tf2/test/test_tf2_autoware_auto_msgs.cpp:17:
/home/sig/autoware/src/universe/autoware.universe/common/autoware_auto_tf2/include/autoware_auto_tf2/tf2_autoware_auto_msgs.hpp:81:13: error: redefinition of ‘void tf2::doTransform(const T&, T&, const TransformStamped&) [with T = geometry_msgs::msg::Polygon_<std::allocator<void> >; geometry_msgs::msg::TransformStamped = geometry_msgs::msg::TransformStamped_<std::allocator<void> >]’
   81 | inline void doTransform(
      |             ^~~~~~~~~~~
In file included from /home/sig/autoware/src/universe/autoware.universe/common/autoware_auto_tf2/include/autoware_auto_tf2/tf2_autoware_auto_msgs.hpp:36,
                 from /home/sig/autoware/src/universe/autoware.universe/common/autoware_auto_tf2/test/test_tf2_autoware_auto_msgs.cpp:17:
/opt/ros/humble/include/tf2_geometry_msgs/tf2_geometry_msgs/tf2_geometry_msgs.hpp:460:6: note: ‘void tf2::doTransform(const T&, T&, const TransformStamped&) [with T = geometry_msgs::msg::Polygon_<std::allocator<void> >; geometry_msgs::msg::TransformStamped = geometry_msgs::msg::TransformStamped_<std::allocator<void> >]’ previously declared here
  460 | void doTransform(
      |      ^~~~~~~~~~~
ninja: build stopped: subcommand failed.
---
Failed   <<< autoware_auto_tf2 [7.85s, exited with code 1]
```

@xianglunkai 
So I guess `sudo apt update && sudo apt upgrade` should also fix the issue on your side.

xianglunkai, 2023-10-13:
Txs!
Do we have any way to ensure the correct and complete dependencies required for each build？

VRichardJP, 2023-10-13:
Yes and no.

According to the [package.xml specification](https://ros.org/reps/rep-0149.html#id20), we could add attributes such as `version_eq="VERSION"`. So yes, it is possible to add a version number to each library... but it would be very painful to maintain. Not only that, but I am not even sure it would bring anything. Many ROS2 packages are installed through the OS package manager, which most likely will install the latest version available.

xianglunkai, 2023-10-16:
@VRichardJP 
Txs!

closed 2023-10-16 by xianglunkai

luojiaxiang11, 2024-04-01:
I also encountered this issue, but when I use 'apt upgrade' or 'apt upgrade ros-humble-tf2-*', a new error   appeared::
`Preparing to unpack .../008-tzdata_2024a-0ubuntu0.20.04_all.deb ...
Unpacking tzdata (2024a-0ubuntu0.20.04) over (2023c-0ubuntu0.20.04.2) ...
dpkg: error processing archive /tmp/apt-dpkg-install-zt6sBA/008-tzdata_2024a-0ubuntu0.20.04_all.deb (--unpack):
 unable to make backup link of './usr/share/zoneinfo/Etc/UTC' before installing new version: Invalid cross-device 
link
dpkg-deb: error: paste subprocess was killed by signal (Broken pipe)`
How should I solve this problem? Looking forward to everyone's help.