autowarefoundation/autoware_universe #5293: Obstacle Detection and Avoidance happening even if obstacle(trees) is not in path or drivable area.

YashuMessi18, 2023-10-12:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

Hello.
We've just switched from autoware.ai to autoware.universe. While running the vehicle in autonomous mode, the vehicle is reacting to obstacles which are not in path or outside drivable area especially trees. Like the point cloud clusters are falling inside the path, and it is treating it as an obstacle. We are using only LIDAR and IMU. We do not have any other sensors.

Either after detection "It is stopping Or It is trying to avoid that " where generally, it won't be needed. We need the vehicle to not treat it as an obstacle and move as it is until the goal pose.

Please suggest what to do now or what parameters we need to change such that it doesn't consider it as an obstacle.
PFA rviz and real time pictures.

If you see IMG.jpeg, it is real time picture taken by phone, where a small plant is on left side outside drivable area, not in the path as well but vehicle is considering it as an obstacle.
If you see Screenshot.png, the snippet is taken from rviz, where path is turned yellow and while testing it is stopping over there.

Similarly, if you see Screenshot(2).png, there are trees across both sides and some of the point clusters formed by that is getting treated as an obstacle and vehicle is getting stopped.
[image]
[image]
[image]

### Expected behavior

 We need the vehicle to not treat it as an obstacle and move as it is until the goal pose. 

### Actual behavior

The vehicle is stopping and chattering a lot or else deviating from the path(avoidance).

### Steps to reproduce

Launch autoware 
Running with our interface

### Possible causes

Maybe the points cluster created by the lidar is effecting it. Need ideas where we can implement such that the leaves/ branches of the tree way too much above from the vehicle or trees on border of the drivable area isn't being considered an obstacle.

VRichardJP, 2023-10-19:
I think your obstacle detection problem can easily be fixed by cheating on the height of your vehicle. For example, the [sample vehicle launch](https://github.com/autowarefoundation/sample_vehicle_launch/blob/main/sample_vehicle_description/config/vehicle_info.param.yaml) defines a vehicle that is 2.5 meter high. If you set the height for example to 2.0 (or lower), then any points above the threshold will be ignored. There is another parameter for maximum height detection somewhere, but I think "cheating" on the vehicle height is the simplest solution.

As for avoidance you have multiple options:
- make sure your PCD map includes all trees/bushes: autoware detection module filters out lidar points that are on the map. Unfortunately, this won't do if there is a lot of wind in your area.
- reduce the size of the lanelet, so that nothing ever appear on top of the road. Of course, because the lanelet shape is modified, the vehicle may look like it is avoiding the bushes/trees, but at least in a controlled way.
- if you have no real need for obstacle avoidance, you could also simply disable it in the `autoware_launch` parameters.

mehmetdogru, 2023-11-09:
@YashuMessi18 

As they are already mentioned;

- About the obstacles above z axis filter in obstacle_stop_planner is the simplest solution.

- About the plants hanging on the sides you can modify [this section ](https://github.com/autowarefoundation/autoware_launch/blob/0cd5d891a36ac34a32a417205905c109f2bafe7b/autoware_launch/config/planning/scenario_planning/lane_driving/behavior_planning/behavior_path_planner/avoidance/avoidance.param.yaml#L73-L83)not to avoid unknown objects at all or with much smaller lateral margins

Does these solve your problem. Is it ok if close the issue?

mehmetdogru, 2024-01-18:
I am closing the issue since it is stale

closed 2024-01-18 by mehmetdogru