autowarefoundation/autoware_universe #5311: Multiple ASAN reports in autoware

Rrooach, 2023-10-15:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

Hi, when I fuzz testing the autoware, I use address sanitizer (ASAN) to compile the application, and I encounter multiple ASAN errors. the detailed information is listed in the following link
[aged_object_queue](https://pastebin.com/7TW9TCHs)
[csv_loader](https://pastebin.com/rxvatCWh)
[lapjv](https://pastebin.com/6s4L4TC0)
[lowpass_filter](https://pastebin.com/VUSV6FJw)
[range](https://pastebin.com/imuH7wD0)
[strack](https://pastebin.com/9Zveq2tc)

It would be so nice of you to review this bug report. Thank you again for taking the time. Any suggestions or feedback you can provide would be very helpful.

### Expected behavior

Programs run without any errors

### Actual behavior

Programs crash with ASAN report

### Steps to reproduce

Compile the autoware with ASAN, Run corresponding components, and get the corresponding errors.

VRichardJP, 2023-10-19:
I am not familiar with fuzzing techniques, but it seems the reports mention files that don't exist? For example in the ekf related one:
```
#2  __wfuzz_entrypoint_adapter_back(void*, unsigned long)   /home/changfeng/workspace/shuimu/fuzz-autoware/fuzz_detail/fuzz.autoware.universe/localization/ekf_localizer/fuzz_aged_object_queue.cpp:45:1
```

Where does this `fuzz_aged_object_queue.cpp` comes from? There is not even a `aged_object_queue.cpp` file in the project. 

Then, that very report seems to point out an illegal read in the `back()` function here:

https://github.com/autowarefoundation/autoware.universe/blob/b0196439a149318a10f4cb3a22d2ee7a541ff886/localization/ekf_localizer/include/ekf_localizer/aged_object_queue.hpp#L31

The only case where this function could segfault is if `object_` is empty, which is the "expected" undefined behavior of many C++ standard containers. Or am I missing something?

xmfcx, 2023-11-01:
@Rrooach Welcome to the Autoware project!

> Compile the autoware with ASAN, Run corresponding components, and get the corresponding errors.

Could you detail these steps please? We could add this as a test to the CI too.

xmfcx, 2023-11-01:
Also checked aged_object_queue bug over @VRichardJP 's investigation,

The overloaded back function is only used in 2 places as far as I could look up:

- https://github.com/autowarefoundation/autoware.universe/blob/b0196439a149318a10f4cb3a22d2ee7a541ff886/localization/ekf_localizer/src/ekf_localizer.cpp#L642-L653

```cpp
  if (!pose_queue_.empty()) {
    geometry_msgs::msg::PoseStamped p;
    p.pose = pose_queue_.back()->pose.pose;
...
  }
...
  if (!pose_queue_.empty()) {
    pose_yaw = tf2::getYaw(pose_queue_.back()->pose.pose.orientation);
  }
```
And both check the size before accessing it.

The results you share with us is interesting but we would like to know it in more detail to replicate them and address the issues.

Rrooach, 2023-11-02:
> Could you detail these steps please? We could add this as a test to the CI too.

Hi, sorry for the late response, I'm little preoccupied by some other work.

Basically, fuzzing is an automatic testing approach, it uses randomly generated input (most cases are a random buffer) to the program under test,  with the help of Address Sanitizer (ASAN), we detect program's memory error.

I'd be happy to provide some of the detailed steps: 
Here we use our customized fuzzer to extract some of the interfaces within autoware, and compile them during alone with the entire autoware project.  
For the ASAN, we can instrument it by adding external ```CXX_FLAG``` augment with ```-fsanitize=address``` 
The following is the source code: https://drive.google.com/file/d/1kITL_eZA7gpJj5tZky0D-rYS8TgeGmQR/view?usp=sharing
and the following are the related file path:
```bash
./fuzz.autoware.universe/perception/bytetrack/lib/fuzz_lapjv.cpp 
./fuzz.autoware.universe/perception/bytetrack/lib/fuzz_strack.cpp 
./fuzz.autoware.common/autoware_utils/math/fuzz_range.cpp 
./fuzz.autoware.universe/control/mpc_lateral_controller/fuzz_lowpass_filter.cpp 
./fuzz.autoware.universe/vehicle/raw_vehicle_cmd_converter/fuzz_csv_loader.cpp 
./fuzz.autoware.universe/localization/ekf_localizer/fuzz_aged_object_queue.cpp  
./fuzz.autoware.universe/common/motion_utils/fuzz_distance.cpp
``` 
The fuzzing process involves using our developed fuzzing tool, but here Im sorry that we can not completely open source that due to some commercial reasons.

The problem is that the code called some of the tool's API, like
```C
WFUZZ_TEST_ENTRYPOINT(linspace);
```
You can bypass that by directly call them in a main function, and passing external input to the target function.

> Also checked aged_object_queue bug over @VRichardJP 's investigation,

For this, since we extract some of the API for testing, here we may introduce some false positive during the test, and in that case Im so sorry for the inconvenience.

MUCH THANK AGAIN for your kindness and detailed response! 
For any following question I'd like to provide more information you might need.

best regard.