autowarefoundation/autoware_universe #5362: Centerpoint tiny does not detect any object

VRichardJP, 2023-10-20:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

I am using a velodyne HDL-32E LIDAR sensor.

When I use the default perception model (centerpoint tiny), Autoware is unable to detect any obstacle. For example, there is one person and one car on the left of the vehicle in the following situation: 

[image]

The centerpoint node seems to be running with no error. It just outputs no object:
```bash
$ $ ros2 topic echo /perception/object_recognition/detection/centerpoint/objects
header:
  stamp:
    sec: 1697777914
    nanosec: 491376896
  frame_id: base_link
objects: []
---
header:
  stamp:
    sec: 1697777914
    nanosec: 591464448
  frame_id: base_link
objects: []
---
header:
  stamp:
    sec: 1697777914
    nanosec: 691005184
  frame_id: base_link
objects: []
---

```

No issue when `centerpoint_model_name` is set to `centerpoint`:
[image]

or when `apollo` is used as the `lidar_detection_model`:

[image]

### Expected behavior

Clearly visible objects should be easily detected

### Actual behavior

Nothing is detected

### Steps to reproduce

Run autoware with centerpoint?

miursh, 2023-10-26:
@yukke42 Do you have any ideas about this?

yukke42, 2023-10-27:
@VRichardJP It may be because centerpoint is trained on nuscenes dataset (32 beams lidar) and centerpoint_tiny is trained on argovese dataset. (64 beams lidar)

VRichardJP, 2023-10-27:
I see. 

If the number of layers is important, I guess it would be nice to document it and name the models accordingly. For example "centerpoint_32". 

Then it also means it would be best to provide weights trained with different sensors (16,32,64,etc). But I guess finding datasets with the correct input is a problem...

yukke42, 2023-11-01:
> If the number of layers is important, I guess it would be nice to document it and name the models accordingly. For example "centerpoint_32".

We haven't investigated with different lidars, but it shuould be important. And I agree with you.

> Then it also means it would be best to provide weights trained with different sensors (16,32,64,etc). But I guess finding datasets with the correct input is a problem...

Yes, that's right ...

amadeuszsz, 2024-10-17:
Closing this issue due to inactivity. Please feel free to reopen it if necessary.

closed 2024-10-17 by amadeuszsz