autowarefoundation/autoware_universe #5373: Autoware drives slower than before

VRichardJP, 2023-10-23:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

After updating my code base to the latest version, I observed Autoware is driving slower than before. For example, Autoware used to drive up to 30km/h on the following road:

[Screencast from 2023年10月23日 13時42分32秒.webm](https://github.com/autowarefoundation/autoware.universe/assets/18645627/9c57a49d-0c3a-4a20-8114-c7f108fcbcf2)

After updating the code, Autoware is now driving very slowly on that same road:

[Screencast from 2023年10月23日 14時14分34秒.webm](https://github.com/autowarefoundation/autoware.universe/assets/18645627/8176ccb6-1b6c-4a0b-b300-7efcdf3902d8)

In both examples, I am using the planning simulator with the exact same map, vehicle and sensor kit. The only difference is that the first example code base is a few months old (latest merged commits from main branch are from July), while the second one is up-to-date.

It looks like a planning related issue, however there are so many modules it is difficult for me to identify who is responsible for slowing down the vehicle.

Does anyone has any idea? 

### Expected behavior

Driving as fast as before

### Actual behavior

Slower

### Steps to reproduce

Compare today's vs snapshot from 3~4 months ago

VRichardJP, 2023-10-23:
After some more troubleshooting, it looks like the default `jerk_weight` it too big:

https://github.com/autowarefoundation/autoware_launch/blob/main/autoware_launch/config/planning/scenario_planning/common/motion_velocity_smoother/JerkFiltered.param.yaml

Reducing the value to ~1.0 made the vehicle accelerate more and felt more natural (a human driver would still accelerate more).

@satoshi-ota 
It looks like in [this PR](https://github.com/autowarefoundation/autoware_launch/commit/20d27d3502e31573cfa3203ca3ece353a58d1ff0), all weights have been multiplied by 10, except `jerk_weight` which has been multiplied by 100. Is it on purpose?

satoshi-ota, 2023-10-26:
> After some more troubleshooting, it looks like the default `jerk_weight` it too big: [...]

@VRichardJP Thanks for your comment. I updated those parameters in order to align value between launcher config and package one. But there is no spesific reason so please tune up them in your project:+1: 

https://github.com/autowarefoundation/autoware.universe/blob/98fe02b9d6cd618a002d9dbbf2e6707d2f95498f/planning/motion_velocity_smoother/config/JerkFiltered.param.yaml#L1-L7

closed 2023-10-26 by VRichardJP