autowarefoundation/autoware_universe #5427: Strange behaviors while using rviz ManualController plugin

evshary, 2023-10-27:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

While using the rviz2 plugin ManualController, the vehicle will lose control if I set positive and negative velocity repeatedly

[image]

### Expected behavior

The vehicle should work no matter in the positive or negative velocity.

### Actual behavior

The vehicle will lose control

### Steps to reproduce

1. Set the vehicle to 10 km/h
2. Then set to -10 km/h
3. Set to 10 km/h again
4. The vehicle will accelerate and move forward crazily

### Versions

Ubuntu 22.04 with latest humble Autoware.

### Possible causes

I guess this is something wrong with handling acceleration in ManualController, but haven't taken a deeper look.

### Additional context

Autoware galactic doesn't have the issue.

mehmetdogru, 2023-11-09:
@evshary 

Thank you for creating the issue.

I was not able to reproduce the issue by following `Steps to reproduce ` in the description.

Could you share a video that is happening?

evshary, 2023-11-09:
Sure, here is the video

https://github.com/autowarefoundation/autoware.universe/assets/456210/15778ec9-1b92-4487-80b7-5b99373a4cc0

mehmetdogru, 2023-11-09:
Thank you for the video @evshary, @TakaHoribe will ask TIER IV engineer who is responsible for the plugin.

maxime-clem, 2023-11-30:
I found this issue can be fixed by forcing the car to go through the `PARK` gear when switching from reverse to drive.

https://github.com/autowarefoundation/autoware.universe/blob/3ac90b08c7a57dca222b5c65734b3ffddbad4d60/common/tier4_control_rviz_plugin/src/tools/manual_controller.cpp#L136-L143

This can be accomplished by changing the condition 
https://github.com/autowarefoundation/autoware.universe/blob/3ac90b08c7a57dca222b5c65734b3ffddbad4d60/common/tier4_control_rviz_plugin/src/tools/manual_controller.cpp#L136

to 

```C++
    if (ackermann.longitudinal.speed > eps && current_velocity_ > -eps) {
```

The switch to `PARK` forces the car to completely stop when switching from negative to positive velocity (similar to the current switch from positive to negative) so it is not ideal.

Unrelated, but there might be a bug with the calculation of the time step `dt`.
https://github.com/autowarefoundation/autoware.universe/blob/3ac90b08c7a57dca222b5c65734b3ffddbad4d60/common/tier4_control_rviz_plugin/src/tools/manual_controller.cpp#L225

Here it is constant but it should be calculation from the difference in the timestamps of the received `VelocityReport`.

mehmetdogru, 2024-01-18:
@maxime-clem to see if a PR needs to be created

maxime-clem, 2024-01-30:
I opened a PR to fix this issue: https://github.com/autowarefoundation/autoware.universe/pull/6230

mehmetdogru, 2024-02-15:
@evshary 

Could you confirm if https://github.com/autowarefoundation/autoware.universe/pull/6230 solves your issue so that we can close it?

maxime-clem, 2024-02-29:
@evshary friendly ping.

evshary, 2024-02-29:
Hi @mehmetdogru @maxime-clem Sorry that I didn't have time to test it recently. I hope I can give it a try next week.
But it's ok to close the issue without waiting for me.
Thank you for the help on solving the issue.

maxime-clem, 2024-02-29:
@evshary It is fine if you can check it next week. I just want to make sure the proposed solution is suitable.
Thank you

evshary, 2024-03-04:
@maxime-clem After some tests, I believe the bug is fixed now. Thank you!

closed 2024-03-04 by evshary

maxime-clem, 2024-03-04:
@evshary thank you for checking.
Are you able to approve the PR ?

evshary, 2024-03-04:
Done.