autowarefoundation/autoware_universe #5443: Inconsistent driving area between normal driving and avoidance

VRichardJP, 2023-10-30:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

Example situation: a bus is driving on a curvy narrow road. Because the vehicle is big and the road is narrow, Autoware estimate the vehicle would not be able to drive the curve without leaving the driving area. However if any object is detected close to the driving lane, avoidance module kicks in and extend the driving area, letting the vehicle drive where it would normally stop.

[avoidance-2023-10-30_17.11.56.webm](https://github.com/autowarefoundation/autoware.universe/assets/18645627/a075b087-45b1-418c-94ea-47ffa4a6b71c)

I don't mind Autoware stopping because the road is too narrow, but if the drivable area can increase when there is an obstacle, why should it be different if there nothing around?

### Expected behavior

If autoware can't drive when the road is too narrow, the existence or not of an obstacle around the path should not change that decision.

### Actual behavior

Drivable area increases.

### Steps to reproduce

Planning simulator with big vehicle on a narrow road

zulfaqar-azmi-t4, 2023-11-02:
@VRichardJP 
The area can be expanded dynamically when `dynamic expansion` [is enabled](https://github.com/autowarefoundation/autoware_launch/blob/2255356e0164430ed5bc7dd325e3b61e983567a3/autoware_launch/config/planning/scenario_planning/lane_driving/behavior_planning/behavior_path_planner/drivable_area_expansion.param.yaml#L10C1-L11C1).

VRichardJP, 2023-11-02:
@zulfaqar-azmi-t4 Thank you for pointing out.
My point here is the lack of consistency of the default features. Normal lane driving is "simpler"/"safer" than avoidance, so it makes more sense for autoware to be able to do "more complex" things during normal lane driving than during avoidance manoeuvers. 

Or maybe the dynamic expansion feature is new/experimental and will be enabled by default in the future?

zulfaqar-azmi-t4, 2023-11-06:
> Or maybe the dynamic expansion feature is new/experimental and will be enabled by default in the future?

@VRichardJP 
Yes, it was found that there are degradation during scenario testing when dynamic expansion is activated. Given that this feature primarily benefits longer vehicles, the decision was made to deactivate it as a default setting. 

The feature is will be enabled by default once it reaches more mature level.

zulfaqar-azmi-t4, 2024-01-17:
Hi @VRichardJP 
The parameter now is [turned on](https://github.com/autowarefoundation/autoware_launch/blob/faef67ad27511d9bd604e3757203e119a7940967/autoware_launch/config/planning/scenario_planning/lane_driving/behavior_planning/behavior_path_planner/drivable_area_expansion.param.yaml#L10) by default.
So it should solve this issue.

Is it okay for me to close this?

VRichardJP, 2024-01-17:
No problem!

closed 2024-01-17 by VRichardJP