autowarefoundation/autoware_universe #5449: Obstacle stop planner causes slow creep and crash

MrRubyRed, 2023-10-31:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

Creating a simple scene in scenario_v2 of ego trailing a fake agent via ACC where the fake agent eventually stops, it causes ego to stop pass the stop point and to slowly creep forward onto the stopped agent.

### Expected behavior

After the other agent stops, ego should stop and not creep forward.

### Actual behavior

Ego stops past the obstacle stop line and it subsequently starts to creep forward.

### Steps to reproduce

1. Create an agent that goes forward at constant speed and then stops.
2. Have ego in the same lane trailing the vehicle with ACC.
3. Make agent stop.

### Versions

- OS: Ubuntu 22.04
- ROS2

### Possible causes

Condition `if (node_param_.enable_slow_down && latest_slow_down_section_)` inside `ObstacleStopPlannerNode::insertVelocity` is the culprit. The logic makes it so the condition is always satisfied.

MrRubyRed, 2023-11-05:
@xmfcx I finally tracked the problem down. The issue happens in the `searchPredictedObjects` function, particularly this line:  https://github.com/autowarefoundation/autoware.universe/blob/main/planning/obstacle_stop_planner/src/node.cpp#L904C1-L910C10 

The issue is twofold:
1. The intersection polygon previously computed is not really a polygon. A quick debug_ptr_->pushPolygon shows that the intersection is actually a segment of the edge of the collision predicted object. I'm fairly sure the boost intersection geometry operations within the the `searchPredictedObjects` is not quite correct.

2. The `collision_point` computed as a result of the `bg::intersection` call is a point on the edge of the `one_step_move_vehicle_polygon` which will (with a bit of noise) cause the previously provided code snippet with the condition `bg::within(collision_point, one_step_move_vehicle_polygon)` to fail, which in turn causes the `planner_data.found_collision_points` to be false.

mehmetdogru, 2023-11-09:
@MrRubyRed 

Thanks for creating the issue.

Could you create a PR that solves the problem or could you share a scenario where the issue can be reproduced.

MrRubyRed, 2023-11-09:
https://github.com/autowarefoundation/autoware.universe/pull/5545

brkay54, 2023-12-06:
Hi @MrRubyRed, I created a draft PR that may solve this problem. I wrote the idea behind this PR [here](https://github.com/autowarefoundation/autoware.universe/pull/5545#issuecomment-1837268436). Can you try this PR for your case?

mehmetdogru, 2024-02-15:
@MrRubyRed I will be closing the issue since it is stale. Please open up again if the issue is still there.

closed 2024-02-15 by mehmetdogru