autowarefoundation/autoware_universe #5453: cmake error of TRTEXEC_DEFAULT_BIN string value

felixf4xu, 2023-11-01:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

In cmake file of autoware.universe/common/trtexec_vendor/CMakeLists.txt, this line has a bug:
```
set(TRTEXEC_DEFAULT_BIN /usr/src/tensorrt/bin/trtexec)
if(NOT EXISTS TRTEXEC_DEFAULT_BIN)
```
it should be:
```
if(NOT EXISTS ${TRTEXEC_DEFAULT_BIN})
```
please notice the `${}`

### Expected behavior

cmake can detect trtexec bin file 

### Actual behavior

cmake failes to detect trtexec even if the file exists

### Steps to reproduce

run cmake of colcon for this package

### Versions

Ubuntu 22
humble
autoware main branch

xmfcx, 2023-11-01:
Thank you for bringing this to our attention.

You're right; it's a typo in the `CMakeLists.txt` file which can cause the described issue.

Would you be interested in contributing a small PR to fix this?
If not, we'll address it promptly. We appreciate your help in improving Autoware.

isouf, 2023-11-22:
I am facing the same issue when trying to build the workspace (inside Docker) of the latest autoware `main` branch. I also confirm that the solution proposed by @felixf4xu resolves problem :+1: . Would be good to check and merge the PR https://github.com/autowarefoundation/autoware.universe/pull/5464 created by @felixf4xu.

amadeuszsz, 2024-07-08:
Closing with reference to https://github.com/autowarefoundation/autoware.universe/pull/5749 and https://github.com/autowarefoundation/autoware.universe/pull/5464#issuecomment-1839922422

closed 2024-07-08 by amadeuszsz