autowarefoundation/autoware_universe #5455: static_centerline_optimizer launch error

harishkumarbalaji, 2023-11-01:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

When I try to run the static_centerline_optimizer launch file, I get an error in rviz plugins error.

### Expected behavior

To view the optimized vector_map output in rviz

### Actual behavior

3 errors in rviz plugin
```
 [rviz2]: PluginlibFactory: The plugin for class 'rviz_plugins/PathWithLaneIdFootprint' failed to load. Error: According to the loaded plugin descriptions the class rviz_plugins/PathWithLaneIdFootprint with base class type rviz_common::Display does not exist.
```
```
[rviz2]: PluginlibFactory: The plugin for class 'rviz_plugins/TrajectoryFootprint' failed to load. Error: According to the loaded plugin descriptions the class rviz_plugins/TrajectoryFootprint with base class type rviz_common::Display does not exist.
```
```
[rviz2]: PluginlibFactory: The plugin for class 'rviz_plugins/PathFootprint' failed to load. Error: According to the loaded plugin descriptions the class rviz_plugins/PathFootprint with base class type rviz_common::Display does not exist.
```

### Steps to reproduce

ros2 launch static_centerline_optimizer run_planning_server.launch.xml vehicle_model:=sample_vehicle

xmfcx, 2023-11-01:
You should source Autoware in the terminal you are running the RViz, before calling it.

`source your_autoware_path/install/setup.bash`

then call:

`ros2 launch static_centerline_optimizer run_planning_server.launch.xml vehicle_model:=sample_vehicle`

If you already tried this and it still gives this error, we'll mark it as a bug.

harishkumarb-tw, 2023-11-01:
Yes, I sourced the Autoware in the terminal. I was able to run with `rviz:=false`, but getting this error when running with `rviz:=true` I hope the issue is with Rviz config or plugins. 

Also, the web interface method didn't work, 
I am getting this when I open the website `http://localhost:4010/`
```
Not Found
The requested URL was not found on the server. If you entered the URL manually please check your spelling and try again.
```
So, Ultimately I couldn't visualise the optimized path.

harishkumarbalaji, 2023-11-01:
Yes, I sourced the Autoware in the terminal. I was able to run with `rviz:=false`, but getting this error when running with `rviz:=true` I hope the issue is with Rviz config or plugins. 

Also, the web interface method didn't work, 
I am getting this when I open the website `http://localhost:4010/`
```
Not Found
The requested URL was not found on the server. If you entered the URL manually please check your spelling and try again.
```
So, Ultimately I couldn't visualise the optimized path.

xmfcx, 2023-11-01:
I get the same for web interface but I think it is expected.

https://github.com/autowarefoundation/autoware.universe/tree/64e2ab33dfcfd8a32dc76fe9de0396209a0f5c41/planning/static_centerline_optimizer#vector-map-builder-interface

> Note: This function of Vector Map Builder has not been released. Please wait for a while. Currently there is no documentation about Vector Map Builder's operation for this function.

For me, it didn't give me the RViz plugin errors.

[image]

So your plugin errors are interesting. Did you build the entire autoware?

Nevertheless, you should run it with the Command Line Interface instructions anyways due to reasons above.

Could you follow: https://github.com/autowarefoundation/autoware.universe/tree/64e2ab33dfcfd8a32dc76fe9de0396209a0f5c41/planning/static_centerline_optimizer#command-line-interface

harishkumarbalaji, 2023-11-01:
Thanks for the instructions, I just realized that I was working in the [awsim-stable](https://github.com/autowarefoundation/autoware/tree/awsim-stable) branch. 
After checking, I found out that the problem persists only in `awsim-stable` branch and it is working well in `main` branch.

xmfcx, 2023-11-01:
Probably currently you don't need to be on awsim-stable branch to have autoware working with AWSIM.

We need to test and move the awsim-stable branch to latest main. But this is outside of the scope of this issue.

I'm glad your problem is solved. If you have any trouble on main branch with AWSIM, please let us know.

closed 2023-11-01 by xmfcx