autowarefoundation/autoware_universe #5495: vehicle moving in a zig-zag manner even if the path is straight

KRajeswarRao, 2023-11-06:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

Hi
We've just switched from autoware.ai to autoware.universe. While running the vehicle in autonomous mode, we have observed that the vehicle is moving in a zig-zag manner although the path is completely straight.

when we set the estimate pose(starting point) of the ego vehicle slant or somewhat tilted, and the goal pose (end point) far away in a straight orientation. generally vehicle should try to align quick and better with centerline. 

Here in our case, the vehicle is trying to align exactly in the path but it is moving as shown in the images below attached. it is continuously trying to do that and in the process it is crashing out of the path/drivable area.
But when we keep the vehicle perfectly aligned with the center line and the steering angle is also zero degree then the vehicle is moving in a straight line

Please suggest what parameters or code needs to be changed.
[image]
[image]

### Expected behavior

vehicle has to align with centerline and move straight even when the initial pose is slightly tilted.

### Actual behavior

The vehicle is moving in a zig-zag manner when we try to give the initial pose as slightly tilted or the initial steering angle has some degree

### Steps to reproduce

Launch Autoware

### Versions

ROS2
Autoware universe

kosuke55, 2023-11-22:
@KRajeswarRao 
If you are using MPC for steer control, it may be helpful to tune the parameters with reference to the following page.
https://autowarefoundation.github.io/autoware.universe/main/control/mpc_lateral_controller/

kyoichi-sugahara, 2023-12-01:
@KRajeswarRao 
If you have further questions related this issue, please feel free to open new issue with pasting this link.
I will close this issue, for now.

closed 2023-12-01 by kyoichi-sugahara