autowarefoundation/autoware_universe #5530: route_handler fails when enable_correct_goal_pose param is true

beyzanurkaya, 2023-11-09:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

There are 2 reasons for this problem:

1. The route_handler does not check for overlap lanelets for goal_lanelet, it only checks the closest lanelet to the goal_pose.
2. When `enable_correct_goal_pose` is true, the goal_pose is calculated with the closest lanelet to it, disregarding the overlap lanelets. Since this lanelet is not found in the `all_route_lanelets` obtained from the route_handler, the error message` Goal is not valid! Please check position and angle of goal_pose` is sent. No route is being generated.

https://github.com/autowarefoundation/autoware.universe/assets/32412808/27591fcc-230e-451f-bfa4-1b3a60014d0e

### Expected behavior

All overlap lanelets should be checked to generate a proper route, the shortest among the possible routes should be selected, and the route should be generated.

### Actual behavior

There is a proper route available based on the given start and goal positions, but it is not being generated.

### Steps to reproduce

1. set enable_correct_goal_pose to true
2. download [map.zip](https://github.com/autowarefoundation/autoware.universe/files/13305659/map.zip)
3. run Psim
4. give a goal pose on red dot with a single click  
[image]

kosuke55, 2023-11-22:
@beyzanurkaya 
> All overlap lanelets should be checked to generate a proper route, the shortest among the possible routes should be selected, and the route should be generated.

thanks, I agree with your idea! Do you plan to work on this task?

kosuke55, 2023-11-22:
@beyzanurkaya
wow, thanks the PR https://github.com/autowarefoundation/autoware.universe/pull/5533!!

kosuke55, 2023-11-30:
some scenarios related shoulder_lanes fails, but I did not check closely :pray: