autowarefoundation/autoware_universe #5539: Reduce excessive error and warning logs on Autoware launch

xmfcx, 2023-11-09:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

When Autoware is executed, it generates a significant number of warning and error logs, despite the system running as expected. The logs are not only excessive but also misrepresent the stability of the running application, as they appear even in the absence of actual functional issues. This creates a noisy debugging environment, making it challenging to identify and address genuine issues.

The issue was replicated using the latest `main` branch with the planning simulator demo as per the Autoware documentation: [Planning Simulation Tutorial](https://autowarefoundation.github.io/autoware-documentation/main/tutorials/ad-hoc-simulation/planning-simulation/).

A video demonstrating the issue is available here:
- [Autoware Error and Warning Logs Video](https://www.youtube.com/watch?v=JDXDh1oNaMg)

The logs from the run that showcase the problem can be found here:
- [Autoware Logs Gist](https://gist.github.com/xmfcx/aebd2fe8a9a7c08a388aafae69dd110a)

### Purpose

The primary goal is to clean up the logs by either resolving the underlying causes of the warnings and errors or by adjusting the threat levels if the logs are not indicative of real issues.
This will aid in de-cluttering the console output, making it easier to debug actual problems, and generally improve the developer experience.

### Possible approaches

1. **Log Analysis**: Go through the logs to determine if the warnings/errors are justified. If they represent real issues, then they should be addressed with bug fixes.
2. **Adjust Log Levels**: If the logs are overly cautious and do not represent actual errors or critical warnings, then the log levels should be adjusted accordingly.
3. **Code Refactoring**: Inspect the sections of the code-base that are generating these logs and refactor if necessary to prevent false positives.
4. **Configuration Tweaks**: Check if the verbosity of the logs can be configured, allowing users to set the desired level of log detail.
5. **Documentation Update**: If certain logs are expected and do not indicate problems, this should be clearly documented to avoid confusion. (last resort, not preferred)
6. **Incremental Fixing**: Tackle the errors and warnings in batches, creating multiple issues or pull requests to address them systematically.

### Definition of done

- [ ] The error and warning logs that are not indicative of actual issues are eliminated or their threat levels are adjusted.
- [ ] The logging output should be clear and should accurately reflect the operational status of Autoware.
- [ ] Documentation should be updated, if necessary, to reflect any changes made to log management.
- [ ] Additional issues should be created for any unresolved logs that require further investigation or long-term changes.
- [ ] The Autoware community agrees that the logging is now informative and not excessively verbose, allowing for effective debugging.

xmfcx, 2023-11-09:
To clarify, this task encapsulates warning and error logs coming from any part of the Autoware for any reason.

ahmeddesokyebrahim, 2023-12-26:
I would like to share updates regarding this issue.

The target of this issue is to reduce excessive log message in Autoware for the following  tutorials with default parameters
- [Planning Simulation](https://autowarefoundation.github.io/autoware-documentation/main/tutorials/ad-hoc-simulation/planning-simulation/)
- [Rosbag Replay Simulation](https://autowarefoundation.github.io/autoware-documentation/main/tutorials/ad-hoc-simulation/rosbag-replay-simulation/)
- [AWSIM](https://autowarefoundation.github.io/autoware-documentation/main/tutorials/ad-hoc-simulation/digital-twin-simulation/awsim-tutorial/)

In this [sheet](https://docs.google.com/spreadsheets/d/1z6RTyKQuaERNli5b-iCdj1ottrK84h_RUKbKmkc7YxU/edit?usp=sharing), we are categorize and following up the log messages during the fixing process of this issue. 

The initial plan is to have the task into 5 phases : 

Phase | Definition
-- | --
Phase 1 | **Planning simulation** : starting planning simulator and before setting vehicle start location. This phase produces a lot of recurrent message while nodes are initializing
Phase 2 | **Planning simulation** : setting vehicle start location, goal location, and basic driving scenario. This phase where actually the nodes are starting to work and communicate.
Phase 3 | **Planning simulation** :  handing special driving scenarios in planning simulator like parking, …
Phase 4 | **Rosbag replay simulation**
Phase 5 | **AWSIM**

The following is the list of relevant PRs for proposing solution fixing the different log messages. 
_For completing this issue, these PRs must be successully reviewed and merged [list can be updated during development]_ 
- [ ] [autoware.universe](https://github.com/autowarefoundation/autoware.universe/pull/5971)
- [ ] [autoware_launch](https://github.com/autowarefoundation/autoware_launch/pull/760)
- [ ] [autoware_common](https://github.com/autowarefoundation/autoware_common/pull/224)

ahmeddesokyebrahim, 2024-02-05:
this is the list of bugs opened asking the community to collaborate in fixing some of the warning or error messages:

- https://github.com/autowarefoundation/autoware.universe/issues/6204
- https://github.com/autowarefoundation/autoware.universe/issues/6205

ahmeddesokyebrahim, 2024-02-07:
The PRs for the first 3 phases are ready for review
- [autoware.universe](https://github.com/autowarefoundation/autoware.universe/pull/5971)
- [autoware_launch](https://github.com/autowarefoundation/autoware_launch/pull/760)
- [autoware_common](https://github.com/autowarefoundation/autoware_common/pull/224)
- [tier4/pointcloud_to_laserscan](https://github.com/tier4/pointcloud_to_laserscan/pull/6)

The following phases will be addressed in separate PRs.

ahmeddesokyebrahim, 2024-02-13:
### List of PR associated to this issue 

Merged :
- https://github.com/autowarefoundation/autoware.universe/pull/5971
- https://github.com/autowarefoundation/autoware_launch/pull/760
- https://github.com/autowarefoundation/autoware_common/pull/224
- https://github.com/autowarefoundation/autoware.universe/pull/6298
- https://github.com/autowarefoundation/autoware.universe/pull/6384

Ready for Review
- https://github.com/tier4/pointcloud_to_laserscan/pull/6

WIP
- x

mitsudome-r, 2026-03-24:
I'm closing the issue that was inactive for a long time. Feel free to reopen if the issue is still relevant