autowarefoundation/autoware_universe #5607: Rendering error in Image projection based fusion

sezan92, 2023-11-16:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

I was following the Autoware Universe Documentation. I have found this link has a table , but the table has not been rendered properly https://autowarefoundation.github.io/autoware.universe/main/perception/image_projection_based_fusion/

I have checked the raw file , I copied the table to my `VS code` , there the table is correctly rendered.

### Expected behavior

from Markdown tables generator
[image]

### Actual behavior

https://autowarefoundation.github.io/autoware.universe/main/perception/image_projection_based_fusion/

### Steps to reproduce

1. check the link

idorobotics, 2023-11-17:
@sezan92 thank you for reporting this. Are you able to create a PR in the docs repo for the quick fix?

sezan92, 2023-11-20:
@idorobotics someone already created a PR in #5625

idorobotics, 2024-01-30:
Closed, #5625 merged.

closed 2024-01-30 by idorobotics