autowarefoundation/autoware_universe #5632: behavior path planner dies in endless `getLeftMostLineString` recursion

VRichardJP, 2023-11-20:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

Sorry for the lack of information (yet)

I have caught this crash of the behavior path planner container with gdb:

```
(lost upper trace)
#4 lanelet::routing::RoutingGraph::previous()
#5 route_handler::RouteHandler::getPreviousLaneletsWithRoute()
#6 route_handler::RouteHandler::getLeftLanelet()
#7 route_handler::RouteHandler::getLeftMostLineString()
#8 route_handler::RouteHandler::getLeftMostLineString()
#9 route_handler::RouteHandler::getLeftMostLineString()
#10 route_handler::RouteHandler::getLeftMostLineString()
...and so on
```

My map is just a big circular loop, with sometimes multiple lanes and some cross walks (nothing fancy).

It might be the same crash as in https://github.com/autowarefoundation/autoware.universe/issues/5631

I will update when I have more information.

### Expected behavior

Never crash

### Actual behavior

crash

### Steps to reproduce

Stay tuned

VRichardJP, 2023-11-29:
I got the same crash today after setting a new goal point. The route handler died after 52000 recursions

I was too clumsy to find the root cause with gdb this time, but from the backtrace and the code, I can think of 2 cases:

- either `getLeftLanelet(lanelet, enable_same_root)` may return the input `lanelet`, and we have an easy recursion
- or the same with a cycle (e.g. lanelet B is left of lanelet A, which is left of lanelet B).

https://github.com/autowarefoundation/autoware.universe/blob/b65bfe03b241001c5476003d722b43b1c995ac4f/planning/route_handler/src/route_handler.cpp#L1329-L1341

I am surprised how seldom this bug is (3rd time I see it over 2 weeks). So I guess it might be related to the start/goal point. The lanelet map is just a big one way loop, with sometimes multiple lanes, sometimes a road shoulder on the left, and a few crosswalk. I don't see how such recursion could happen.

In any case, I don't think it is sane to keep unbounded recursion like this one in the code. Most roads are only a few lanelets wide, so it would not be very costy to keep a list of all explored lanelets and stop if a loop is detected.

VRichardJP, 2023-11-29:
What I managed to scrape:
```
#52263 0x00007fffec7dcb69 in route_handler::RouteHandler::getLeftMostLinestring (this=0x55555564ef70, lanelet=..., enable_same_root=true) at /usr/include/boost/optional/detail/optional_aligned_storage.hpp:48
#52264 0x00007fffec7dcb69 in route_handler::RouteHandler::getLeftMostLinestring (this=0x55555564ef70, lanelet=..., enable_same_root=true) at /usr/include/boost/optional/detail/optional_aligned_storage.hpp:48
#52265 0x00007fffec7dcb69 in route_handler::RouteHandler::getLeftMostLinestring (this=0x55555564ef70, lanelet=..., enable_same_root=true) at /usr/include/boost/optional/detail/optional_aligned_storage.hpp:48
#52266 0x00007fffec7dcb69 in route_handler::RouteHandler::getLeftMostLinestring (this=0x55555564ef70, lanelet=..., enable_same_root=true) at /usr/include/boost/optional/detail/optional_aligned_storage.hpp:48
#52267 0x00007fffec7dcb69 in route_handler::RouteHandler::getLeftMostLinestring (this=0x55555564ef70, lanelet=..., enable_same_root=true) at /usr/include/boost/optional/detail/optional_aligned_storage.hpp:48
#52268 0x00007fffec7dcb69 in route_handler::RouteHandler::getLeftMostLinestring (this=0x55555564ef70, lanelet=..., enable_same_root=true) at /usr/include/boost/optional/detail/optional_aligned_storage.hpp:48
#52269 0x00007fffec7dcb69 in route_handler::RouteHandler::getLeftMostLinestring (this=0x55555564ef70, lanelet=..., enable_same_root=true) at /usr/include/boost/optional/detail/optional_aligned_storage.hpp:48
#52270 0x00007fffec7dcb69 in route_handler::RouteHandler::getLeftMostLinestring (this=0x55555564ef70, lanelet=..., enable_same_root=true) at /usr/include/boost/optional/detail/optional_aligned_storage.hpp:48
#52271 0x00007fffec7dcb69 in route_handler::RouteHandler::getLeftMostLinestring (this=0x55555564ef70, lanelet=..., enable_same_root=true) at /usr/include/boost/optional/detail/optional_aligned_storage.hpp:48
#52272 0x00007fffec7dcb69 in route_handler::RouteHandler::getLeftMostLinestring (this=0x55555564ef70, lanelet=..., enable_same_root=true) at /usr/include/boost/optional/detail/optional_aligned_storage.hpp:48
#52273 0x00007fffec7dcb69 in route_handler::RouteHandler::getLeftMostLinestring (this=0x55555564ef70, lanelet=..., enable_same_root=true) at /usr/include/boost/optional/detail/optional_aligned_storage.hpp:48
#52274 0x00007fffec7dcb69 in route_handler::RouteHandler::getLeftMostLinestring (this=0x55555564ef70, lanelet=..., enable_same_root=true) at /usr/include/boost/optional/detail/optional_aligned_storage.hpp:48
#52275 0x00007fffec7dcb69 in route_handler::RouteHandler::getLeftMostLinestring (this=this@entry=0x55555564ef70, lanelet=..., enable_same_root=enable_same_root@entry=true) at /usr/include/boost/optional/detail/optional_aligned_storage.hpp:48
#52276 0x00007fffec7dd117 in route_handler::RouteHandler::getFurthestLinestring (this=0x55555564ef70, lanelet=..., is_right=<optimized out>, is_left=<optimized out>, is_opposite=<optimized out>, enable_same_root=true) at /home/computer/autoware/src/universe/autoware.universe/planning/route_handler/src/route_handler.cpp:1366
#52277 0x00007fffc5c1b672 in operator()<lanelet::ConstLanelet> (__closure=0x7fffc9fee9a0, target_lanelet=...) at /home/computer/autoware/src/universe/autoware.universe/planning/behavior_path_planner/src/utils/avoidance/utils.cpp:1029
#52278 0x00007fffc5c1fe07 in behavior_path_planner::utils::avoidance::filterTargetObjects (objects=std::vector of length 4, capacity 4 = {...}, data=..., debug=..., planner_data=std::shared_ptr<const behavior_path_planner::PlannerData> (use count 16, weak count 0) = {...}, parameters=std::shared_ptr<behavior_path_planner::AvoidanceParameters> (use count 4, weak count 0) = {...}) at /home/computer/autoware/src/universe/autoware.universe/planning/behavior_path_planner/src/utils/avoidance/utils.cpp:1040
#52279 0x00007fffc5b318b4 in behavior_path_planner::AvoidanceByLaneChange::fillAvoidanceTargetObjects (this=0x7fff981820e0, data=..., debug=...) at /home/computer/autoware/src/universe/autoware.universe/planning/behavior_path_planner/src/scene_module/lane_change/avoidance_by_lane_change.cpp:179
#52280 0x00007fffc5b31d1a in behavior_path_planner::AvoidanceByLaneChange::calcAvoidancePlanningData (this=0x7fff981820e0, debug=...) at /home/computer/autoware/src/universe/autoware.universe/planning/behavior_path_planner/src/scene_module/lane_change/avoidance_by_lane_change.cpp:133
#52281 0x00007fffc5b31e60 in behavior_path_planner::AvoidanceByLaneChange::updateSpecialData (this=0x7fff981820e0) at /home/computer/autoware/src/universe/autoware.universe/planning/behavior_path_planner/src/scene_module/lane_change/avoidance_by_lane_change.cpp:97
#52282 0x00007fffc5afd894 in behavior_path_planner::LaneChangeInterface::updateData (this=0x7fff98176850) at /home/computer/autoware/src/universe/autoware.universe/planning/behavior_path_planner/src/scene_module/lane_change/interface.cpp:192
#52283 0x00007fffc596d3f2 in behavior_path_planner::SceneModuleManagerInterface::isExecutionRequested (this=0x5555558a2820, previous_module_output=...) at /home/computer/autoware/src/universe/autoware.universe/planning/behavior_path_planner/include/behavior_path_planner/scene_module/scene_module_manager_interface.hpp:92
#52284 0x00007fffc595638a in behavior_path_planner::PlannerManager::getRequestModules (this=0x5555557c4a80, previous_module_output=...) at /usr/include/c++/11/bits/shared_ptr_base.h:1295
[... cut ...] elet::ConstLineStringImpl<lanelet::Point3d>> = {<lanelet::ConstPrimitive<lanelet::LineStringData>> = {constData_ = std::shared_ptr<const lanelet::LineStringData> (use count 14, weak count 0) = {get() = 0x7fff68194630}}, inverted_ = false}, <No data fields>}, <No data fields>}, <No data fields>}, <No data fields>}, regulatoryElements_ = std::vector of length 0, capacity 0, centerline_ = std::shared_ptr<lanelet::ConstLineString3d> (use count 1, weak count 0) = {get() = 0x7fff68257130}}
(gdb) 

```

kyoichi-sugahara, 2024-02-29:
@VRichardJP 
Sorry for the late reply as well, and thank you for reporting the bug! 
I've looked into the code a bit on my end but couldn't reach the direct cause of the recursion, sorry about that. 
As a temporary measure to prevent the bug from occurring, as you mentioned, keeping a list of all explored lanelets and stopping if a loop is detected seems like a good idea to me too.
However, despite being assigned, I apologize for not having the time to address this issue. Could you possibly handle it on your side?

However, I'm still curious about the conditions under which the recursion occurs and which part of the process is causing the issue. So, how about providing information that could help reproduce the problem and adding a "help-wanted" tag?

VRichardJP, 2024-03-01:
@kyoichi-sugahara 
Unfortunately, I don't remember the exact setup nor the map. However, I think the problem only occur with AvoidanceByLaneChange (which appear in the backtrace). I don't have access to the vehicle anymore, but the module has been disabled for a long time and the crash has never happened again.

I usually don't enable this kind of advanced module (because too risky), except from time to time during the vehicle integration to test a bit. So it is possible the crash has only ever happened with that module (normal lane change has never been an issue).

mkquda, 2024-05-24:
The function in which the endless recursion occurred is no longer used anywhere in the latest version and has been removed along with other obsolete functions in a recent PR. so this issue can be closed.

closed 2024-05-24 by mkquda

kyoichi-sugahara, 2024-05-24:
@mkquda 
Thank you so much for your report and contribution. Could you paste the link of related PR in this page?

mkquda, 2024-05-26:
https://github.com/autowarefoundation/autoware.universe/pull/7012