autowarefoundation/autoware_universe #5642: surround obstacle checker dies due to nullptr dereference if all obstacle targets are disabled

VRichardJP, 2023-11-21:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

Here, there is an early return if any dynamic target is enabled and the `object_ptr_` is null:

https://github.com/autowarefoundation/autoware.universe/blob/8e209a146357ec5d613f03b54b9711f0d980bcc6/planning/surround_obstacle_checker/src/node.cpp#L295-L310

However the pointer is dereferenced right after in getNearestObstacle() -> getNearestObstacleByDynamicObject():

https://github.com/autowarefoundation/autoware.universe/blob/8e209a146357ec5d613f03b54b9711f0d980bcc6/planning/surround_obstacle_checker/src/node.cpp#L463

Resulting in a segmentation fault.

### Expected behavior

No crash

### Actual behavior

Crash

### Steps to reproduce

Disable all dynamic targets in surround obstacle checker configuration.

kyoichi-sugahara, 2023-11-26:
@VRichardJP cc: @danielsanchezaran 
Can I close this issue with this PR's merge?
https://github.com/autowarefoundation/autoware.universe/pull/5671

danielsanchezaran, 2023-11-28:
@VRichardJP can I close this issue? I believe it has been solved with: https://github.com/autowarefoundation/autoware.universe/pull/5671

closed 2023-11-28 by VRichardJP