autowarefoundation/autoware_universe #5645: Predicted obstacle box is too far from the actual points, which impacts obstacle stop planner

VRichardJP, 2023-11-21:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

I have observed that predicted objects are always "further in time" than their tracked/detected counterpart. Typically, the faster the object is and the farther it will appear from th raw point cloud or raw detection. For example:

https://github.com/autowarefoundation/autoware.universe/assets/18645627/a354804d-3aef-467a-b57b-eff26e3adf14

I could not find any kind of delay compensation, but I guess this is what the prediction algorithm is doing. 

One annoying side effect is that obstacle stop planner is sometimes not able to associate points on the path with the tracked objects they come from (because they are too far from each other), and so autoware stops.

One way I have found to "fix" this issue is to increase the `object_polygon_length_margin` in the adaptive cruise control parameters (which is the case in the video). However it is not a great solution, because it increases the likeliness autoware would associate points to the wrong object.

### Expected behavior

obstacle stop planner can reliabily associate points to objects of any speed

### Actual behavior

when the object is fast, the predicted obstacle is too far from the pointcloud obstacle stop planner is using for comparison.

### Steps to reproduce

use lidar based detection.

VRichardJP, 2023-11-30:
For whoever gets this issue in the future: obstacle cruise planner does not have such issue at all because it does not use pointcloud data. It is also way more reliable than obstacle stop planner module but that's another topic.

technolojin, 2024-04-05:
Hi!

I am recently working on perception latency reduction.

What actually happening is, the tracking objects are in "current time" and the point clouds are "past time".  
From the sensor time to detection may takes several hundreds micro seconds.
(in case of 36 kph = 10 m/s of target related speed, 400 ms latency causes 4 m difference) 

> I could not find any kind of delay compensation

the delay compensation is done in object tracking - object publish time.
Config: https://github.com/autowarefoundation/autoware.universe/blob/804600599c546e5947d34666e842d6e29840909a/perception/multi_object_tracker/config/multi_object_tracker_node.param.yaml#L15
Prediction: https://github.com/autowarefoundation/autoware.universe/blob/804600599c546e5947d34666e842d6e29840909a/perception/multi_object_tracker/src/tracker/model/normal_vehicle_tracker.cpp#L377

You can disable this option, but you will get old objects.

It is impossible to reduce the latency to zero, but can be mitigated.
This is a case to be improved.
https://github.com/autowarefoundation/autoware.universe/pull/6710
The improvement will continue.

closed 2024-04-09 by technolojin

technolojin, 2024-04-17:
@VRichardJP

I linked this issue to my latest work for feedback.
I didn't know that the issue is closed by the PR is merged.

As I mentioned, the positions of point cloud, detected object, and tracked object cannot be the same unless there is no latency. 
If it is not acceptable from your side, please reopen the Issue.