autowarefoundation/autoware_universe #5649: behavior path planner sometimes dies when vehicle is on a small road shoulder

VRichardJP, 2023-11-22:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

Encountered this crash twice today, when the vehicle starts from a small road shoulder is asked to drive in a loop:

[image]

Backtrace:

```gdb
[INFO] [planning.scenario_planning.lane_driving.behavior_planning.behavior_path_planner]: waiting for scenario_topic (operator()())
[WARN] [behavior_path_planner.path_shifter]: Acc limit is too small to be applied. Tj: 0.258131, Ta: 3.157973, j: 0.581100, a_max: 0.278924, acc_limit: 0.150000 (calcBaseLengths())
[INFO] [planning.scenario_planning.lane_driving.behavior_planning.behavior_velocity_planner]: register task: module = out_of_lane, id = 0 (registerModule())
[WARN] [behavior_path_planner.path_shifter]: shift start point and end point can't be adjoining Maybe shift length is too short? (generate())
[WARN] [behavior_path_planner.path_shifter]: shift start point and end point can't be adjoining Maybe shift length is too short? (generate())
[WARN] [behavior_path_planner.path_shifter]: Acc limit is too small to be applied. Tj: 0.258131, Ta: 3.157973, j: 0.581100, a_max: 0.278924, acc_limit: 0.150000 (calcBaseLengths())
[WARN] [behavior_path_planner.path_shifter]: Acc limit is too small to be applied. Tj: 0.273621, Ta: 3.666367, j: 0.548203, a_max: 0.280519, acc_limit: 0.150000 (calcBaseLengths())
[WARN] [behavior_path_planner.path_shifter]: shift start point and end point can't be adjoining Maybe shift length is too short? (generate())
[WARN] [behavior_path_planner.path_shifter]: shift start point and end point can't be adjoining Maybe shift length is too short? (generate())
[WARN] [behavior_path_planner.path_shifter]: Acc limit is too small to be applied. Tj: 0.291301, Ta: 3.225285, j: 0.514932, a_max: 0.277050, acc_limit: 0.150000 (calcBaseLengths())
[WARN] [planning.scenario_planning.lane_driving.behavior_planning.behavior_path_planner.start_planner]: Start plan for a backward goal is not supported now (planWaitingApproval())
[WARN] [planning.scenario_planning.lane_driving.behavior_planning.behavior_path_planner.start_planner]: Start plan for a backward goal is not supported now (plan())
[ERROR] [planning.scenario_planning.lane_driving.behavior_planning.behavior_path_planner.start_planner]: Path is not safe against static objects (isExecutionReady())
terminate called after throwing an instance of 'lanelet::NoSuchPrimitiveError'
  what():  Tried to lookup an element with id InvalId!

Thread 29 "component_conta" received signal SIGABRT, Aborted.
[Switching to Thread 0x7fffcc7f0640 (LWP 31126)]
__pthread_kill_implementation (no_tid=0, signo=6, threadid=140736624264768) at ./nptl/pthread_kill.c:44
44	./nptl/pthread_kill.c: No such file or directory.
(gdb) bt
#0  __pthread_kill_implementation (no_tid=0, signo=6, threadid=140736624264768) at ./nptl/pthread_kill.c:44
#1  __pthread_kill_internal (signo=6, threadid=140736624264768) at ./nptl/pthread_kill.c:78
#2  __GI___pthread_kill (threadid=140736624264768, signo=signo@entry=6) at ./nptl/pthread_kill.c:89
#3  0x00007ffff7442476 in __GI_raise (sig=sig@entry=6) at ../sysdeps/posix/raise.c:26
#4  0x00007ffff74287f3 in __GI_abort () at ./stdlib/abort.c:79
#5  0x00007ffff78a2b9e in ?? () from /lib/x86_64-linux-gnu/libstdc++.so.6
#6  0x00007ffff78ae20c in ?? () from /lib/x86_64-linux-gnu/libstdc++.so.6
#7  0x00007ffff78ae277 in std::terminate() () from /lib/x86_64-linux-gnu/libstdc++.so.6
#8  0x00007ffff78ae4d8 in __cxa_throw () from /lib/x86_64-linux-gnu/libstdc++.so.6
#9  0x00007fffe4096e65 in ?? () from /opt/ros/humble/lib/x86_64-linux-gnu/liblanelet2_core.so.1
#10 0x00007ffff42d2ee9 in route_handler::RouteHandler::getLaneletsFromId (this=<optimized out>, id=<optimized out>) at /home/computer/autoware/src/universe/autoware.universe/planning/route_handler/src/route_handler.cpp:512
#11 0x00007fffc63463eb in behavior_path_planner::utils::getCurrentLanesFromPath (path=..., planner_data=std::shared_ptr<const behavior_path_planner::PlannerData> (use count 14, weak count 0) = {...}) at /home/computer/autoware/src/universe/autoware.universe/planning/behavior_path_planner/src/utils/utils.cpp:3065
#12 0x00007fffc630a944 in behavior_path_planner::NormalLaneChange::getCurrentLanes (this=0x7fff600217d0) at /home/computer/autoware/src/universe/autoware.universe/planning/behavior_path_planner/src/scene_module/lane_change/normal.cpp:406
#13 behavior_path_planner::NormalLaneChange::isLaneChangeRequired (this=0x7fff600217d0) at /home/computer/autoware/src/universe/autoware.universe/planning/behavior_path_planner/src/scene_module/lane_change/normal.cpp:117
#14 0x00007fffc615638a in behavior_path_planner::PlannerManager::getRequestModules (this=0x5555557c4ec0, previous_module_output=...) at /usr/include/c++/11/bits/shared_ptr_base.h:1295
#15 0x00007fffc615e25b in operator() (__closure=0x7fffcc7ee8e0) at /home/computer/autoware/src/universe/autoware.universe/planning/behavior_path_planner/src/planner_manager.cpp:95
#16 0x00007fffc615efc8 in behavior_path_planner::PlannerManager::run (this=this@entry=0x5555557c4ec0, data=std::shared_ptr<behavior_path_planner::PlannerData> (use count 14, weak count 0) = {...}) at /home/computer/autoware/src/universe/autoware.universe/planning/behavior_path_planner/src/planner_manager.cpp:55
#17 0x00007fffc619530e in behavior_path_planner::BehaviorPathPlannerNode::run (this=0x5555556a7490) at /usr/include/c++/11/bits/shared_ptr_base.h:1295
#18 0x00007fffc6199fa5 in std::__invoke_impl<void, void (behavior_path_planner::BehaviorPathPlannerNode::*&)(), behavior_path_planner::BehaviorPathPlannerNode*&> (__t=@0x5555558943c0: 0x5555556a7490, __f=@0x5555558943b0: (void (behavior_path_planner::BehaviorPathPlannerNode::*)(behavior_path_planner::BehaviorPathPlannerNode * const)) 0x7fffc6194ee0 <behavior_path_planner::BehaviorPathPlannerNode::run()>) at /usr/include/c++/11/bits/invoke.h:71
#19 std::__invoke<void (behavior_path_planner::BehaviorPathPlannerNode::*&)(), behavior_path_planner::BehaviorPathPlannerNode*&> (__fn=@0x5555558943b0: (void (behavior_path_planner::BehaviorPathPlannerNode::*)(behavior_path_planner::BehaviorPathPlannerNode * const)) 0x7fffc6194ee0 <behavior_path_planner::BehaviorPathPlannerNode::run()>) at /usr/include/c++/11/bits/invoke.h:96
#20 std::_Bind<void (behavior_path_planner::BehaviorPathPlannerNode::*(behavior_path_planner::BehaviorPathPlannerNode*))()>::__call<void, , 0ul>(std::tuple<>&&, std::_Index_tuple<0ul>) (__args=..., this=0x5555558943b0) at /usr/include/c++/11/functional:420
#21 std::_Bind<void (behavior_path_planner::BehaviorPathPlannerNode::*(behavior_path_planner::BehaviorPathPlannerNode*))()>::operator()<, void>() (this=0x5555558943b0) at /usr/include/c++/11/functional:503
#22 rclcpp::GenericTimer<std::_Bind<void (behavior_path_planner::BehaviorPathPlannerNode::*(behavior_path_planner::BehaviorPathPlannerNode*))()>, (void*)0>::execute_callback_delegate<std::_Bind<void (behavior_path_planner::BehaviorPathPlannerNode::*(behavior_path_planner::BehaviorPathPlannerNode*))()>, (void*)0>() (this=0x555555894380) at /opt/ros/humble/include/rclcpp/rclcpp/timer.hpp:244
#23 rclcpp::GenericTimer<std::_Bind<void (behavior_path_planner::BehaviorPathPlannerNode::*(behavior_path_planner::BehaviorPathPlannerNode*))()>, (void*)0>::execute_callback() (this=0x555555894380) at /opt/ros/humble/include/rclcpp/rclcpp/timer.hpp:230
#24 0x00007ffff7d63472 in rclcpp::Executor::execute_any_executable (this=this@entry=0x5555555d1410, any_exec=...) at /home/computer/autoware/src/rclcpp/rclcpp/src/rclcpp/executor.cpp:518
#25 0x00007ffff7d695ea in rclcpp::executors::MultiThreadedExecutor::run (this=0x5555555d1410, this_thread_number=<optimized out>) at /home/computer/autoware/src/rclcpp/rclcpp/src/rclcpp/executors/multi_threaded_executor.cpp:93
#26 0x00007ffff78dc253 in ?? () from /lib/x86_64-linux-gnu/libstdc++.so.6
#27 0x00007ffff7494ac3 in start_thread (arg=<optimized out>) at ./nptl/pthread_create.c:442
#28 0x00007ffff7526a40 in clone3 () at ../sysdeps/unix/sysv/linux/x86_64/clone3.S:81
(gdb) 
```

### Expected behavior

no crash

### Actual behavior

crash

### Steps to reproduce

TBD

VRichardJP, 2023-11-22:
How to reproduce:

[lanelet2_map.osm.txt](https://github.com/autowarefoundation/autoware.universe/files/13434465/lanelet2_map.osm.txt)

[Screencast from 2023-11-22 13-29-22.webm](https://github.com/autowarefoundation/autoware.universe/assets/18645627/bb247836-eae1-4510-8c3c-d131406ea160)

The bug seems caused by the the goal point being behind the ego vehicle but on the same lanelet. When I set the goal point on a different lanelet there is no problem.

mkquda, 2024-05-17:
Cannot reproduce this issue on latest version of autoware (using the same map and following the same steps shown in https://github.com/autowarefoundation/autoware.universe/issues/5649#issuecomment-1822192857 )

@VRichardJP can you confirm if issue still exists on latest version, or are you able to provide the version or commit hash on which you encountered the issue

closed 2024-06-03 by mkquda