autowarefoundation/autoware_universe #5657: `behavior_planning_container` dies from the start of the driving

Kim-mins, 2023-11-23:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

Hi team, 
I'm currently running Autoware on Carla, and I found the `behavior_planning_container` dies from the the start of the driving sometimes.
Here are the videos for better description: [[video: rviz](https://koreaoffice-my.sharepoint.com/:v:/g/personal/minsu_korea_ac_kr/Eer_5zzJIClFgd9cvvGNoqQBDaqS7vGFibR5ZP-rMNTOaw?nav=eyJyZWZlcnJhbEluZm8iOnsicmVmZXJyYWxBcHAiOiJPbmVEcml2ZUZvckJ1c2luZXNzIiwicmVmZXJyYWxBcHBQbGF0Zm9ybSI6IldlYiIsInJlZmVycmFsTW9kZSI6InZpZXciLCJyZWZlcnJhbFZpZXciOiJNeUZpbGVzTGlua0RpcmVjdCJ9fQ&e=H4NSll)] [[video: frontview](https://koreaoffice-my.sharepoint.com/:v:/g/personal/minsu_korea_ac_kr/EZKOTY9ofN5FnjoDQkeyBesBHo21sW_5wVshiPP0fNwcLA?nav=eyJyZWZlcnJhbEluZm8iOnsicmVmZXJyYWxBcHAiOiJPbmVEcml2ZUZvckJ1c2luZXNzIiwicmVmZXJyYWxBcHBQbGF0Zm9ybSI6IldlYiIsInJlZmVycmFsTW9kZSI6InZpZXciLCJyZWZlcnJhbFZpZXciOiJNeUZpbGVzTGlua0RpcmVjdCJ9fQ&e=rLQzlD)]

According to the [[launch.log](https://koreaoffice-my.sharepoint.com/:u:/g/personal/minsu_korea_ac_kr/Ec6rs0RGFZFBoT8fcm7WROYBl3EQArlXVLOFmnnXu1E2xg?e=p13mZ2)] file, I could see the log below regarding the error from `behavior_planning_container` module:
```
1700705716.2011483 [ERROR] [component_container_mt-60]: process has died [pid 6806, exit code -11, cmd '/opt/ros/humble/lib/rclcpp_components/component_container_mt --ros-args -r __node:=behavior_planning_container -r __ns:=/planning/scenario_planning/lane_driving/behavior_planning -p use_sim_time:=True -p wheel_radius:=0.383 -p wheel_width:=0.235 -p wheel_base:=2.79 -p wheel_tread:=1.64 -p front_overhang:=1.0 -p rear_overhang:=1.1 -p left_overhang:=0.128 -p right_overhang:=0.128 -p vehicle_height:=2.5 -p max_steer_angle:=0.7'].
```

### Expected behavior

I expect the ego vehicle to make proper trajectory continuously.

### Actual behavior

The car cannot plan the trajectory due to the error.

### Steps to reproduce

From docker image [[ghcr.io/autowarefoundation/autoware-universe:humble-20230715-prebuilt-cuda-amd64](https://github.com/autowarefoundation/autoware/pkgs/container/autoware-universe/109875667?tag=humble-20230715-cuda)], I ran Autoware and replayed the situation with the [[rosbag file](https://koreaoffice-my.sharepoint.com/:u:/g/personal/minsu_korea_ac_kr/EV2tjpnukg1Cq-tKizGsJukBIoh2gE2RDyaLFMfff7LPlw?e=llh8lX)]

### Versions

- Autoware from docker image [[ghcr.io/autowarefoundation/autoware-universe:humble-20230715-prebuilt-cuda-amd64](https://github.com/autowarefoundation/autoware/pkgs/container/autoware-universe/109875667?tag=humble-20230715-cuda)]
- Carla: 0.9.13

### Possible causes

I cannot pinpoint the error location, but it seems the error is from `behavior_planning_container` according to the `launch.log` file above.

VRichardJP, 2023-11-23:
A simple way to dig more information is to start the container with gdb and see where it crashes.

Until recently the behavior planner launch was python based and you could do this:

```python
    container = ComposableNodeContainer(
        name="behavior_planning_container",
        namespace="",
        package="rclcpp_components",
        executable=LaunchConfiguration("container_executable"),
        composable_node_descriptions=[
            behavior_path_planner_component,
            behavior_velocity_planner_component,
            glog_component,
        ],
        output="screen",
        prefix="gnome-terminal -- gdb -ex=run --args",  # << add this line
    )
```

If you updated the code recently, it is possible you only have this xml launch file:

https://github.com/autowarefoundation/autoware.universe/blob/c171d1c80c13d64da73e5cdf9c81c19df5d29e95/launch/tier4_planning_launch/launch/scenario_planning/lane_driving/behavior_planning/behavior_planning.launch.xml#L114C4-L114C18

To which you can add `launch_prefix="gnome-terminal -- gdb -ex=run --args"` to the `node_container` (or maybe just `prefix=...`, I am not so sure about the `node_container` syntax)

Kim-mins, 2023-11-24:
Thank you for the response @VRichardJP!!

I'm currently running a bit old version now, so I could find the python file and modified it as you told.
Here's a new [[launch.log](https://koreaoffice-my.sharepoint.com/:u:/g/personal/minsu_korea_ac_kr/EaU3KJLiYZ9AuZFxPEnWmjABOhgo0gu3xIQ99htymZyYBg?e=oOBOuY)] file, but I could not get any detailed information about `behavior_path_planner` when I replayed the log file.
Can another component be the cause of the error? Or should I run the driving again for more log, not just replaying it?

By the way, I could find some logs that can be related to the bug (regarding `obstacle_stop_planner`, `path_smoother` and `obstacle_avoidance_planner`. Can they be the cause?
```1700802176.8211129 [component_container_mt-62] [WARN] [1700802176.820604084] [planning.scenario_planning.lane_driving.motion_planning.obstacle_stop_planner]: size of vel que is 0. Something has wrong.
1700802176.8520107 [component_container_mt-62] [WARN] [1700802176.851752597] [planning.scenario_planning.lane_driving.motion_planning.obstacle_stop_planner]: size of vel que is 0. Something has wrong.
1700802176.8573928 [component_container_mt-62] [WARN] [1700802176.857080597] [planning.scenario_planning.lane_driving.motion_planning.obstacle_stop_planner]: size of vel que is 0. Something has wrong.
1700802176.9685121 [component_container_mt-62] [WARN] [1700802176.968224703] [planning.scenario_planning.lane_driving.motion_planning.obstacle_stop_planner]: size of vel que is 0. Something has wrong.
1700802176.9718113 [component_container_mt-62] [WARN] [1700802176.970929631] [planning.scenario_planning.lane_driving.motion_planning.obstacle_stop_planner]: size of vel que is 0. Something has wrong.
1700802176.9842675 [component_container_mt-62] [WARN] [1700802176.980416359] [planning.scenario_planning.lane_driving.motion_planning.obstacle_stop_planner]: size of vel que is 0. Something has wrong.
1700802177.0556555 [component_container_mt-62] [WARN] [1700802177.040514216] [path_smoother.trajectory_utils]: Fixed point will not be inserted due to the error during calculation.
1700802177.0558257 [component_container_mt-62] [WARN] [1700802177.053943136] [planning.scenario_planning.lane_driving.motion_planning.obstacle_stop_planner]: size of vel que is 0. Something has wrong.
1700802177.0979228 [component_container_mt-62] [WARN] [1700802177.097674864] [planning.scenario_planning.lane_driving.motion_planning.obstacle_stop_planner]: size of vel que is 0. Something has wrong.
1700802177.1584635 [component_container_mt-62] [WARN] [1700802177.158132562] [planning.scenario_planning.lane_driving.motion_planning.obstacle_stop_planner]: size of vel que is 0. Something has wrong.
1700802177.1762669 [component_container_mt-62] [WARN] [1700802177.174972242] [planning.scenario_planning.lane_driving.motion_planning.obstacle_stop_planner]: size of vel que is 0. Something has wrong.
1700802177.1784666 [component_container_mt-62] [WARN] [1700802177.178307523] [planning.scenario_planning.lane_driving.motion_planning.obstacle_stop_planner]: size of vel que is 0. Something has wrong.
1700802177.1826227 [component_container_mt-62] [WARN] [1700802177.181389454] [planning.scenario_planning.lane_driving.motion_planning.obstacle_stop_planner]: size of vel que is 0. Something has wrong.
1700802177.2319012 [component_container_mt-62] [WARN] [1700802177.231623728] [planning.scenario_planning.lane_driving.motion_planning.obstacle_stop_planner]: size of vel que is 0. Something has wrong.
1700802177.3735209 [component_container_mt-62] [WARN] [1700802177.372711749] [planning.scenario_planning.lane_driving.motion_planning.obstacle_stop_planner]: size of vel que is 0. Something has wrong.
1700802177.3771455 [component_container_mt-62] [WARN] [1700802177.376220058] [planning.scenario_planning.lane_driving.motion_planning.obstacle_stop_planner]: size of vel que is 0. Something has wrong.
1700802177.3931603 [component_container_mt-62] [WARN] [1700802177.392943635] [planning.scenario_planning.lane_driving.motion_planning.obstacle_stop_planner]: size of vel que is 0. Something has wrong.
1700802177.4860768 [component_container_mt-62] [WARN] [1700802177.485826454] [planning.scenario_planning.lane_driving.motion_planning.obstacle_stop_planner]: size of vel que is 0. Something has wrong.
1700802178.8114269 [component_container_mt-62] [WARN] [1700802178.806527287] [obstacle_avoidance_planner.trajectory_utils]: Fixed point will not be inserted due to the error during calculation.
```

Kim-mins, 2023-11-24:
Oh sorry, I just found that I modified another python file.
Here's a new [[launch.log](https://koreaoffice-my.sharepoint.com/:u:/g/personal/minsu_korea_ac_kr/EWFOxr9ln5dFqx0JIOtO3GQBH96THSSt3BjW3Zbo8-J5sw?e=n4Asq4)] file, and I could find logs regarding `localization`, not `behavior_planning_container`:
```
1700803896.3757224 [ndt_scan_matcher-31] [ERROR] [1700803896.367688053] [localization.pose_estimator.ndt_scan_matcher]: Cannot find the reference position for map update. Please check if the EKF odometry is provided to NDT.
1700803896.3763189 [service_log_checker-3] [ERROR] [1700803896.367737108] [system.service_log_checker]: /localization/initialize: status code 1 'The vehicle is not stopped.' (/localization/util/pose_initializer_node)
1700803896.3763847 [service_log_checker-3] [ERROR] [1700803896.367868030] [system.service_log_checker]: /localization/initialize: status code 1 'The vehicle is not stopped.' (/default_ad_api/node/localization)
1700803896.3764348 [service_log_checker-3] [ERROR] [1700803896.368173635] [system.service_log_checker]: /api/localization/initialize: status code 1 'The vehicle is not stopped.' (/default_ad_api/node/localization)
1700803896.3764811 [service_log_checker-3] [ERROR] [1700803896.368353735] [system.service_log_checker]: /api/localization/initialize: status code 1 'The vehicle is not stopped.' (/localization/util/default_ad_api/helpers/automatic_pose_initializer)
```

VRichardJP, 2023-11-24:
With the launch prefix, the behavior planning container should start in a new terminal, like here:
https://github.com/autowarefoundation/autoware.universe/issues/5649#issuecomment-1822192857

Be careful that due to default gdb pagination, sometimes the program does not start until you press Enter (you can avoid this by creating a `~/.gdbinit` file with `set pagination off` inside).

Regarding your second issue, either the vehicle is not stopped, or Autoware has received any velocity feedback from the vehicle (the wording is not clear I know). The simulator should sends this data to `/vehicle/velocity_status`

Kim-mins, 2023-11-24:
Thank you for the kind instruction @VRichardJP!!

Here are the [[full log](https://koreaoffice-my.sharepoint.com/:u:/g/personal/minsu_korea_ac_kr/EZ_liUO5ug9HgIb789x8JvQBqtLpn3sUUZgyuO84OHsGEg?e=TG3aYU)] and the [[video](https://koreaoffice-my.sharepoint.com/:v:/g/personal/minsu_korea_ac_kr/EfEQjDfb2TtLoWTT2dNPkScB8dZKFcp6xrtkmiw4I1netg?nav=eyJyZWZlcnJhbEluZm8iOnsicmVmZXJyYWxBcHAiOiJPbmVEcml2ZUZvckJ1c2luZXNzIiwicmVmZXJyYWxBcHBQbGF0Zm9ybSI6IldlYiIsInJlZmVycmFsTW9kZSI6InZpZXciLCJyZWZlcnJhbFZpZXciOiJNeUZpbGVzTGlua0RpcmVjdCJ9fQ&e=PUZcj8)] while logging.
In the log file, I could observe the message below:
```
...
Thread 26 "component_conta" received signal SIGSEGV, Segmentation fault.
...
```

I don't know why but after the segmentation fault occurs, the planner seems not work.

VRichardJP, 2023-11-24:
Once the segmentation fault has occured, you should be able to inspect the state of the program with gdb. The first thing to do would be to generate a backtrace with the `bt` command. Something like this:
```
Thread 29 "component_conta" received signal SIGABRT, Aborted.
[Switching to Thread 0x7fffcc7f0640 (LWP 31126)]
__pthread_kill_implementation (no_tid=0, signo=6, threadid=140736624264768) at ./nptl/pthread_kill.c:44
44	./nptl/pthread_kill.c: No such file or directory.
(gdb) bt
#0  __pthread_kill_implementation (no_tid=0, signo=6, threadid=140736624264768) at ./nptl/pthread_kill.c:44
#1  __pthread_kill_internal (signo=6, threadid=140736624264768) at ./nptl/pthread_kill.c:78
#2  __GI___pthread_kill (threadid=140736624264768, signo=signo@entry=6) at ./nptl/pthread_kill.c:89
#3  0x00007ffff7442476 in __GI_raise (sig=sig@entry=6) at ../sysdeps/posix/raise.c:26
#4  0x00007ffff74287f3 in __GI_abort () at ./stdlib/abort.c:79
#5  0x00007ffff78a2b9e in ?? () from /lib/x86_64-linux-gnu/libstdc++.so.6
#6  0x00007ffff78ae20c in ?? () from /lib/x86_64-linux-gnu/libstdc++.so.6
#7  0x00007ffff78ae277 in std::terminate() () from /lib/x86_64-linux-gnu/libstdc++.so.6
#8  0x00007ffff78ae4d8 in __cxa_throw () from /lib/x86_64-linux-gnu/libstdc++.so.6
#9  0x00007fffe4096e65 in ?? () from /opt/ros/humble/lib/x86_64-linux-gnu/liblanelet2_core.so.1
#10 0x00007ffff42d2ee9 in route_handler::RouteHandler::getLaneletsFromId (this=<optimized out>, id=<optimized out>) at /home/computer/autoware/src/universe/autoware.universe/planning/route_handler/src/route_handler.cpp:512
#11 0x00007fffc63463eb in behavior_path_planner::utils::getCurrentLanesFromPath (path=..., planner_data=std::shared_ptr<const behavior_path_planner::PlannerData> (use count 14, weak count 0) = {...}) at /home/computer/autoware/src/universe/autoware.universe/planning/behavior_path_planner/src/utils/utils.cpp:3065
#12 0x00007fffc630a944 in behavior_path_planner::NormalLaneChange::getCurrentLanes (this=0x7fff600217d0) at /home/computer/autoware/src/universe/autoware.universe/planning/behavior_path_planner/src/scene_module/lane_change/normal.cpp:406
#13 behavior_path_planner::NormalLaneChange::isLaneChangeRequired (this=0x7fff600217d0) at /home/computer/autoware/src/universe/autoware.universe/planning/behavior_path_planner/src/scene_module/lane_change/normal.cpp:117
#14 0x00007fffc615638a in behavior_path_planner::PlannerManager::getRequestModules (this=0x5555557c4ec0, previous_module_output=...) at /usr/include/c++/11/bits/shared_ptr_base.h:1295
#15 0x00007fffc615e25b in operator() (__closure=0x7fffcc7ee8e0) at /home/computer/autoware/src/universe/autoware.universe/planning/behavior_path_planner/src/planner_manager.cpp:95
#16 0x00007fffc615efc8 in behavior_path_planner::PlannerManager::run (this=this@entry=0x5555557c4ec0, data=std::shared_ptr<behavior_path_planner::PlannerData> (use count 14, weak count 0) = {...}) at /home/computer/autoware/src/universe/autoware.universe/planning/behavior_path_planner/src/planner_manager.cpp:55
#17 0x00007fffc619530e in behavior_path_planner::BehaviorPathPlannerNode::run (this=0x5555556a7490) at /usr/include/c++/11/bits/shared_ptr_base.h:1295
#18 0x00007fffc6199fa5 in std::__invoke_impl<void, void (behavior_path_planner::BehaviorPathPlannerNode::*&)(), behavior_path_planner::BehaviorPathPlannerNode*&> (__t=@0x5555558943c0: 0x5555556a7490, __f=@0x5555558943b0: (void (behavior_path_planner::BehaviorPathPlannerNode::*)(behavior_path_planner::BehaviorPathPlannerNode * const)) 0x7fffc6194ee0 <behavior_path_planner::BehaviorPathPlannerNode::run()>) at /usr/include/c++/11/bits/invoke.h:71
#19 std::__invoke<void (behavior_path_planner::BehaviorPathPlannerNode::*&)(), behavior_path_planner::BehaviorPathPlannerNode*&> (__fn=@0x5555558943b0: (void (behavior_path_planner::BehaviorPathPlannerNode::*)(behavior_path_planner::BehaviorPathPlannerNode * const)) 0x7fffc6194ee0 <behavior_path_planner::BehaviorPathPlannerNode::run()>) at /usr/include/c++/11/bits/invoke.h:96
#20 std::_Bind<void (behavior_path_planner::BehaviorPathPlannerNode::*(behavior_path_planner::BehaviorPathPlannerNode*))()>::__call<void, , 0ul>(std::tuple<>&&, std::_Index_tuple<0ul>) (__args=..., this=0x5555558943b0) at /usr/include/c++/11/functional:420
#21 std::_Bind<void (behavior_path_planner::BehaviorPathPlannerNode::*(behavior_path_planner::BehaviorPathPlannerNode*))()>::operator()<, void>() (this=0x5555558943b0) at /usr/include/c++/11/functional:503
#22 rclcpp::GenericTimer<std::_Bind<void (behavior_path_planner::BehaviorPathPlannerNode::*(behavior_path_planner::BehaviorPathPlannerNode*))()>, (void*)0>::execute_callback_delegate<std::_Bind<void (behavior_path_planner::BehaviorPathPlannerNode::*(behavior_path_planner::BehaviorPathPlannerNode*))()>, (void*)0>() (this=0x555555894380) at /opt/ros/humble/include/rclcpp/rclcpp/timer.hpp:244
#23 rclcpp::GenericTimer<std::_Bind<void (behavior_path_planner::BehaviorPathPlannerNode::*(behavior_path_planner::BehaviorPathPlannerNode*))()>, (void*)0>::execute_callback() (this=0x555555894380) at /opt/ros/humble/include/rclcpp/rclcpp/timer.hpp:230
#24 0x00007ffff7d63472 in rclcpp::Executor::execute_any_executable (this=this@entry=0x5555555d1410, any_exec=...) at /home/computer/autoware/src/rclcpp/rclcpp/src/rclcpp/executor.cpp:518
#25 0x00007ffff7d695ea in rclcpp::executors::MultiThreadedExecutor::run (this=0x5555555d1410, this_thread_number=<optimized out>) at /home/computer/autoware/src/rclcpp/rclcpp/src/rclcpp/executors/multi_t [... cut ...]
colcon build --symlink-install --cmake-args -DCMAKE_BUILD_TYPE=Debug --packages-select behavior_velocity_planner  # and other crashing modules...
```

If you can catch the same crash in gdb with debug build, it should be pretty easy to find the cause:
- from the segfault, use `up` until you reach some autoware code (source file should be something like `/home/username/autoware/src/...`)
- use `p` to print local variables and see if anything looks suspicious. For example:
```gdb
> p some_ptr
0x0
> p some_vector[0]
NaN
```

Kim-mins, 2023-11-27:
Thank you for the kind response @VRichardJP!!

I checked the gdb terminal again, typed `bt` and could get the log below:
```
Thread 22 "component_conta" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffc77fe640 (LWP 7356)]
0x00007fffc05a13f3 in behavior_velocity_planner::IntersectionModule::checkCollision(autoware_auto_planning_msgs::msg::PathWithLaneId_<std::allocator<void> > const&, std::vector<lanelet::ConstLanelet, std::allocator<lanelet::ConstLanelet> > const&, std::vector<lanelet::ConstLanelet, std::allocator<lanelet::ConstLanelet> > const&, std::optional<boost::geometry::model::polygon<tier4_autoware_utils::Point2d, true, true, std::vector, std::vector, std::allocator, std::allocator> > const&, std::vector<lanelet::ConstLanelet, std::allocator<lanelet::ConstLanelet> > const&, int, double) () from /autoware/install/behavior_velocity_intersection_module/lib/libbehavior_velocity_intersection_module.so
(gdb) bt
#0  0x00007fffc05a13f3 in behavior_velocity_planner::IntersectionModule::checkCollision(autoware_auto_planning_msgs::msg::PathWithLaneId_<std::allocator<void> > const&, std::vector<lanelet::ConstLanelet, std::allocator<lanelet::ConstLanelet> > const&, std::vector<lanelet::ConstLanelet, std::allocator<lanelet::ConstLanelet> > const&, std::optional<boost::geometry::model::polygon<tier4_autoware_utils::Point2d, true, true, std::vector, std::vector, std::allocator, std::allocator> > const&, std::vector<lanelet::ConstLanelet, std::allocator<lanelet::ConstLanelet> > const&, int, double) () from /autoware/install/behavior_velocity_intersection_module/lib/libbehavior_velocity_intersection_module.so
#1  0x00007fffc05a7cf0 in behavior_velocity_planner::IntersectionModule::modifyPathVelocityDetail(autoware_auto_planning_msgs::msg::PathWithLaneId_<std::allocator<void> >*, tier4_planning_msgs::msg::StopReason_<std::allocator<void> >*) () from /autoware/install/behavior_velocity_intersection_module/lib/libbehavior_velocity_intersection_module.so
#2  0x00007fffc05a94a3 in behavior_velocity_planner::IntersectionModule::modifyPathVelocity(autoware_auto_planning_msgs::msg::PathWithLaneId_<std::allocator<void> >*, tier4_planning_msgs::msg::StopReason_<std::allocator<void> >*) () from /autoware/install/behavior_velocity_intersection_module/lib/libbehavior_velocity_intersection_module.so
#3  0x00007fffc058e6d2 in behavior_velocity_planner::SceneModuleManagerInterface::modifyPathVelocity(autoware_auto_planning_msgs::msg::PathWithLaneId_<std::allocator<void> >*) () from /autoware/install/behavior_velocity_intersection_module/lib/libbehavior_velocity_intersection_module.so
#4  0x00007fffc056e736 in behavior_velocity_planner::PluginWrapper<behavior_velocity_planner::IntersectionModuleManager>::plan(autoware_auto_planning_msgs::msg::PathWithLaneId_<std::allocator<void> >*) () from /autoware/install/behavior_velocity_intersection_module/lib/libbehavior_velocity_intersection_module.so
#5  0x00007fffc258a73e in behavior_velocity_planner::BehaviorVelocityPlannerManager::planPathVelocity(std::shared_ptr<behavior_velocity_planner::PlannerData const> const&, autoware_auto_planning_msgs::msg::PathWithLaneId_<std::allocator<void> > const&) () from /autoware/install/behavior_velocity_planner/lib/libbehavior_velocity_planner.so
#6  0x00007fffc24ab4a7 in behavior_velocity_planner::BehaviorVelocityPlannerNode::generatePath(std::shared_ptr<autoware_auto_planning_msgs::msg::PathWithLaneId_<std::allocator<void> > const>, behavior_velocity_planner::PlannerData const&) () from /autoware/install/behavior_velocity_planner/lib/libbehavior_velocity_planner.so
#7  0x00007fffc24ababb in behavior_velocity_planner::BehaviorVelocityPlannerNode::onTrigger(std::shared_ptr<autoware_auto_planning_msgs::msg::PathWithLaneId_<std::allocator<void> > const>) () from /autoware/install/behavior_velocity_planner/lib/libbehavior_velocity_planner.so
#8  0x00007fffc24f3f47 in std::_Function_handler<void (std::shared_ptr<autoware_auto_planning_msgs::msg::PathWithLaneId_<std::allocator<void> > const>), std::_Bind<void (behavior_velocity_planner::BehaviorVelocityPlannerNode::*(behavior_velocity_planner::BehaviorVelocityPlannerNode*, std::_Placeholder<1>))(std::shared_ptr<autoware_auto_planning_msgs::msg::PathWithLaneId_<std::allocator<void> > const>)> >::_M_invoke(std::_Any_data const&, std::shared_ptr<autoware_auto_planning_msgs::msg::PathWithLaneId_<std::allocator<void> > const>&&) () from /autoware/install/behavior_velocity_planner/lib/libbehavior_velocity_planner.so
#9  0x00007fffc24f0cc2 in std::__detail::__variant::__gen_vtable_impl<std::__detail::__variant::_Multi_array<std::__detail::__variant::__deduce_visit_result<void> (*)(rclcpp::AnySubscriptionCallback<autoware_auto_planning_msgs::msg::PathWithLaneId_<std::allocator<void> >, std::allocator<void> >::dispatch(std::shared_ptr<autoware_auto_planning_msgs::msg::PathWithLaneId_<std::allocator<void> > >, rclcpp::MessageInfo const&)::{lambda(auto:1&&)#1}&&, std::variant<std::function<void (autoware_auto_planning_msgs::msg::PathWithLaneId_<std::allocator<void> > const&)>, std::function<void (autoware_auto_planning_msgs::msg::PathWithLaneId_<std::allocator<void> > const&, rclcpp::MessageInfo const&)>, std::function<void (rclcpp::SerializedMessage const&)>, std::function<void (rclcpp::SerializedMessage const&, rclcpp::MessageInfo const&)>, std::function<void (std::unique_ptr<autoware_auto_planning_msgs::msg::PathWithLaneId_<std::allocator<void> >, std::default_delete<autoware_auto_p [... cut ...]
#13 0x00007ffff7e7890a in rclcpp::executors::MultiThreadedExecutor::run(unsigned long) () from /opt/ros/humble/lib/librclcpp.so
#14 0x00007ffff7bf9253 in ?? () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#15 0x00007ffff7967ac3 in start_thread (arg=<optimized out>) at ./nptl/pthread_create.c:442
#16 0x00007ffff79f9a40 in clone3 () at ../sysdeps/unix/sysv/linux/x86_64/clone3.S:81
```

I'm, currently running prebuilt version of Autoware in docker, and I could not get the error line in gdb directly.
Could you please provide me some idea for this?

Kim-mins, 2023-11-27:
Thank you for the kind instruction @VRichardJP!!

I could finally localize the fault line with gdb by recompiling the Autoware with the option you provided!!
Here's the [[full log](https://koreaoffice-my.sharepoint.com/:u:/g/personal/minsu_korea_ac_kr/EaERML45BAJNqG7m9lnpJuYBSaOrgX4fWMWpMbYtscUncA?e=q6Ir6Q)] of gdb, and I think the log below is the most important:
```
...
Thread 20 "component_conta" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffd2ffd640 (LWP 6019)]
0x00007fff674a0dd4 in Eigen::PlainObjectBase<Eigen::Matrix<double, 2, 1, 0, 2, 1> >::PlainObjectBase (this=0x7fffa40303c0, other=...) at /usr/include/eigen3/Eigen/src/Core/PlainObjectBase.h:512
512	      : Base(), m_storage(other.m_storage) { }
(gdb) backtrace
#0  0x00007fff674a0dd4 in Eigen::PlainObjectBase<Eigen::Matrix<double, 2, 1, 0, 2, 1> >::PlainObjectBase (this=0x7fffa40303c0, other=...) at /usr/include/eigen3/Eigen/src/Core/PlainObjectBase.h:512
#1  0x00007fff67498877 in Eigen::Matrix<double, 2, 1, 0, 2, 1>::Matrix (this=0x7fffa40303c0, other=...) at /usr/include/eigen3/Eigen/src/Core/Matrix.h:414
#2  0x00007fff674a927b in tier4_autoware_utils::Point2d::Point2d (this=0x7fffa40303c0) at /home/mins/autoware/install/tier4_autoware_utils/include/tier4_autoware_utils/geometry/boost_geometry.hpp:53
#3  0x00007fff674ac249 in __gnu_cxx::new_allocator<tier4_autoware_utils::Point2d>::construct<tier4_autoware_utils::Point2d, tier4_autoware_utils::Point2d&> (this=0x7fffd2ff8a80, __p=0x7fffa40303c0) at /usr/include/c++/11/ext/new_allocator.h:162
#4  0x00007fff674a45bb in std::allocator_traits<std::allocator<tier4_autoware_utils::Point2d> >::construct<tier4_autoware_utils::Point2d, tier4_autoware_utils::Point2d&> (__a=..., __p=0x7fffa40303c0) at /usr/include/c++/11/bits/alloc_traits.h:516
#5  0x00007fff674a468f in std::vector<tier4_autoware_utils::Point2d, std::allocator<tier4_autoware_utils::Point2d> >::_M_realloc_insert<tier4_autoware_utils::Point2d&> (this=0x7fffd2ff8a80, __position=non-dereferenceable iterator for std::vector) at /usr/include/c++/11/bits/vector.tcc:449
#6  0x00007fff6749bf0e in std::vector<tier4_autoware_utils::Point2d, std::allocator<tier4_autoware_utils::Point2d> >::emplace_back<tier4_autoware_utils::Point2d&> (this=0x7fffd2ff8a80) at /usr/include/c++/11/bits/vector.tcc:121
#7  0x00007fff6748a4db in behavior_velocity_planner::IntersectionModule::checkCollision (this=0x7fff840788a0, path=..., attention_area_lanelets=std::vector of length 24, capacity 32 = {...}, adjacent_lanelets=std::vector of length 1, capacity 1 = {...}, 
    intersection_area=std::optional<boost::geometry::model::polygon<tier4_autoware_utils::Point2d, true, true, std::vector, std::vector, std::allocator, std::allocator>> [no contained value], ego_lane_with_next_lane=std::vector of length 2, capacity 2 = {...}, closest_idx=4, time_delay=0)
    at /home/mins/autoware/src/universe/autoware.universe/planning/behavior_velocity_intersection_module/src/scene_intersection.cpp:1015
...
```

Since I'm currently running somewhat old version of Autoware, the code line of 7th stack is here:
https://github.com/autowarefoundation/autoware.universe/blob/4c47a5f994f599c0070030ff1e010503ef9fc4a3/planning/behavior_velocity_intersection_module/src/scene_intersection.cpp#L1015
and the header file `PlaneObjectBase.h` seems an external library.

So, I guess the library makes a segmentation fault due to some illegal values provided to the library function.

If you need information other than the ones above, please let me know.

VRichardJP, 2023-11-27:
You just found the problem:

```C++
        Polygon2d polygon{};
        for (const auto & p : trimmed_ego_polygon) {
          polygon.outer().emplace_back(p.x(), p.y());
        }

        polygon.outer().emplace_back(polygon.outer().front());
```

We have a `trimmed_ego_polygon` which we use to build another `polygon`. After that we try to close that polygon by adding a copy of the first point at the end (so that `polygon.first() == polygon.last()`)... and this is where the program crash.

If by any change, the `trimmed_ego_polygon` happens to be empty, then our polygon is empty. So accessing its first element with `polygon.outer().front()` is undefined behavior.

@soblin @takayuki5168 @tkimura4 @shmpwk 
Could you check this?

mehmetdogru, 2024-02-15:
@Kim-mins Could you please provide the videos again since they don't work for me. As well could you please check if the issue is reproducable with planning simulator?

@maxime-clem to assign to a proper person that can reproduce the issue using Carla Sim.

Kim-mins, 2024-02-16:
Hi @mehmetdogru!

Sorry for the inconvenience.. I should have noticed that the links are not working currently.
I just restored every important link like video, log, ros2bag file.

Please let me know if you need more information during the debugging.

Thank you!

soblin, 2024-02-19:
This issue is already fixed at latest version.

soblin, 2024-02-19:
https://github.com/autowarefoundation/autoware.universe/blob/3f216833e62f3f4b21b9a287cdce3df82da66eab/planning/behavior_velocity_intersection_module/src/scene_intersection_collision.cpp#L278

soblin, 2024-02-19:
https://github.com/autowarefoundation/autoware.universe/pull/4655

closed 2024-02-19 by soblin

Kim-mins, 2024-02-19:
Thank for checking @soblin and sorry for bothering with the issue resolved already.