autowarefoundation/autoware_universe #5668: NPC vehicle's predicted path is unfeasible wrt HD-Map

mehmetdogru, 2023-11-24:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

**map_based_prediction** node predicts that NPC vehicles can make a lane change even though it is restricted in the HD-Map with:
 
`<tag k="lane_change" v="no"/>`

[image]

### Purpose

NPC vehicles predicted paths should not include unfeasible paths wrt HD-Map for some use cases.

### Possible approaches

Add a parameter that restricts unfeasible (non-routable) predictions and define the necessary condition.

### Definition of done

- [ ] Add a parameter that restricts unfeasible (non-routable) predictions and define the necessary condition

YoshiRi, 2023-11-25:
@mehmetdogru 
Hi. Thanks for your suggestion.

I have several opinion about this issue.

- In our regular automated driving, it is a regular occurrence for other vehicles to straddle the no lane change line.
- For example, it is often the case that a vehicle temporarily crosses the line to avoid a car parked in the shoulder lane.

Therefore, we would like to keep this lane change prediction if possible, because it is possible that other vehicles may cross the line even if it is a boundary that they are not legally allowed to cross.

YoshiRi, 2023-11-25:
Oh, now I just found your PR met all of our requirements :+1:

closed 2023-11-25 by shmpwk