autowarefoundation/autoware_universe #5776: `Predicted Trajectory` generates the path ignoring the NPC vehicle stopping at the downhill

Kim-mins, 2023-12-05:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

Hi team,
I'm currently running Autoware on Carla 0.9.13, and I found the situation that, `predicted trajectory` generates the path ignoring the NPC vehicle stopping on the downhill.

Here are the videos for more details: [[video: rviz](https://drive.google.com/file/d/1CO-e99aQGsKNVitw2FQCyLU3mYPwzc0Q/view?usp=sharing)]

In the video, I turned off viewing predicted trajectory at 0:25, and I turn it on at 0:34.
At 0:34, there's a message `obstacle on the path`, but the predicted trajectory generates the path(white polygon) ignoring the NPC vehicle and the speed does not decreases enough.

### Expected behavior

The car to stop behind the vehicle properly.

### Actual behavior

The car does not consider the vehicle in front of it, and eventually collides.

### Steps to reproduce

From docker image [[ghcr.io/autowarefoundation/autoware-universe:humble-20230715-prebuilt-cuda-amd64](https://github.com/autowarefoundation/autoware/pkgs/container/autoware-universe/109875667?tag=humble-20230715-cuda)], I ran Autoware and replayed the situation with the [[rosbag file](https://drive.google.com/drive/folders/1Gy0K_SNm5z-Nw0k_o3to6gdxrZ-zXy_w?usp=sharing)]

### Versions

- Autoware from docker image [[ghcr.io/autowarefoundation/autoware-universe:humble-20230715-prebuilt-cuda-amd64](https://github.com/autowarefoundation/autoware/pkgs/container/autoware-universe/109875667?tag=humble-20230715-cuda)]
- Carla: 0.9.13

VRichardJP, 2023-12-07:
Is it only for downhill vehicles? From the video it looks like your machine/setup is struggling to run the simulation, maybe it is just that everything is so slow that autoware just can't brake properly?

Kim-mins, 2023-12-07:
Thank you for the response @VRichardJP.

I intentionally slowed down(`↓` while replaying) or stopped(space bar while replaying) the rosbag replay, just for the video recording.
I'm currently running Autoware and simulator(carla) synchronously, so I think there's no sync problem.
You can check the video below, showing a normal behavior of the car.
https://github.com/autowarefoundation/autoware.universe/assets/50267797/105c1cb5-f74e-455f-b3a2-9d5415393966

VRichardJP, 2023-12-07:
In that case I can think of 2 possible causes for your issue:
- Autoware wants to brake for the obstacle (we see the obstacle "wall" in Rviz), but fails to because the slope is too strong. For example, you could try to create a goal point in the downward slope (withuot any obstacle) and see if Autoware manages to stop there or not. Normally, slopes are handled by the "slope compensation" algorithm in the longitudinal controller. [Could you check whether it is enabled in the parameters](https://github.com/autowarefoundation/autoware_launch/blob/36804c87f61a0a13a4e6221c4effc1fa183b71aa/autoware_launch/config/control/trajectory_follower/longitudinal/pid.param.yaml#L8)? It is also possible your lanelet map does not follow the slope (e.g. is flat), in which case Autoware thinks there is no slope.
- otherwise you found a strange bug. Congrats! :-)

Kim-mins, 2023-12-07:
Thank you for the response @VRichardJP! I'll try your suggestion soon.
I checked the value for the parameter you told is `true`(enabled), but I have no idea to check if there's any slope in my lanelet map.
Could you please provide me the way to check it?(any parameters to check on lanelet map file?)

VRichardJP, 2023-12-07:
If you change the camera view in Rviz to orbit or 3d follower, you should be able to check visually whether the lanelet points follow the PCD map slope or not.

VRichardJP, 2023-12-08:
Autoware target acceleration is in m/s^2. 

If I remember correctly, human-like smooth brake is usually below -0.5, while -1.0 is already quite strong (like stopping for a traffic light that just turned red).

Here it looks like the slope compensation is not strong enough (there might be a scaling factor for that). What happens if you disable the slope compensation? Do you get a similar profile (or worse?)

VRichardJP, 2023-12-11:
Yes, it looks like slope compensation is not working as expected. Actually, it's behavior is quite strange:

This is with slope compensation, right before the downward slope (23km/h):
[image]

This is without slope compensation, at the same spot (20km/h):
[image]

The remaining distance to goal point is more or less the same in both situations. However we can see that in the case with the slope compensation the Autoware has barely pushed the brake by the time it arrived at the downward slope, while in the case without slope compensation Autoware already started to brake rather strongly (-2m/s^2) before. In other words, slope compensation made things worse before the slope. There is definitely something fishy there...

@maxime-clem 
Can anyone from the planning team investigate this?

VRichardJP, 2023-12-15:
Regarding the incorrect pitch/z localization, I think ekf is responsible for that. Both Z and orientation are smoothened over multiple NDT poses by the `updateSimple1DFilters` function in EKF. The behavior can be tuned by changing the parameters `pose_additional_delay` (default 0.0) and `pose_smoothing_steps` (default 5) in `ekf_localizer.param.yaml`.

Assuming you are using the default parameters and your simulated LIDAR sensor runs at 10Hz, it means it will take at least 0.5 seconds for z and pitch values to converge on a sudden slope variation. Actually it may even be worse, because it means that the initial pose sent to NDT is not that good during the transition pahse, so the localization result may not be that accurate.

I think you should be able to get better localization by setting `pose_smoothen_steps` to 1 or 2.

maxime-clem, 2023-12-22:
Thank you for reporting the issue and for the initial investigation.
I looked at the bag and the slope information (as calculated by the longitudinal controller) does not match the map seen in the video.
[image]
At the end of the bag when ego collides with the obstacle, the measured slope (black) is positive but in the video the slope was negative (downward slope).

We can also see that there is a lot of error between the target velocity (orange) and the actual velocity (blue).
[image]

So my guess is that the issue is caused by a wrong slope information. Can you share your map so test it locally ?
On your side you can try to set `use_trajectory_for_pitch_calculation` to `true` and see if it makes a difference.

I will try to setup Carla to reproduce the issue but I may not be able to do so before a while.

Kim-mins, 2023-12-22:
Hi @maxime-clem! Thank you for the answer!

Here's the `.pcd` and `.osm` files I used(`Town03` of Carla): [[lanelet2_map.osm](https://drive.google.com/file/d/1U867V_L0DAPzDPPhANzRlhNTgU_gPWtv/view?usp=sharing)] [[pointcloud_map.pcd](https://drive.google.com/file/d/1kBsPE0TGwWfSi_9L2JABuzUWw4ijDBRm/view?usp=sharing)]

I have one question for the trial on my side. Do you mean to enable the parameter `use_trajectory_for_pitch_calculation` [here](https://github.com/autowarefoundation/autoware_launch/blob/36804c87f61a0a13a4e6221c4effc1fa183b71aa/autoware_launch/config/control/trajectory_follower/longitudinal/pid.param.yaml#L73)?
For the setting with Carla, I'm using the bridge [here(humble branch)](https://github.com/evshary/autoware_carla_launch/tree/humble). I hope it would help you.

maxime-clem, 2023-12-25:
Thank you for sharing your map. I was able to reproduce the issue using the `simple_planning_simulator` (after activating the road slope simulation with this PR https://github.com/autowarefoundation/sample_vehicle_launch/pull/55).

https://github.com/autowarefoundation/autoware.universe/assets/78338830/2244bf1b-f839-4ab3-95da-f80996df92a3

There was no difference when enabling parameter `use_trajectory_for_pitch_calculation`.
For reference, here is the behavior with no slope compensation in the longitudinal controller.

https://github.com/autowarefoundation/autoware.universe/assets/78338830/3e0120dc-3247-4e32-ada4-7b0e5fdbb862

So in conclusion, it looks like a problem with the slope compensation not adjusting the deceleration enough to allow stopping in a downward slope.

I am not sure if someone will be able to work on this issue before a while. I will at least make sure it is discussed at the next [working group meeting](https://github.com/orgs/autowarefoundation/discussions/categories/working-group-meetings?discussions_q=is%3Aopen+category%3A%22Working+group+meetings%22++label%3Ameeting%3Aplanning-control-wg+) on January 18 ([google calendar link](https://www.google.com/calendar/event?eid=Z3YxcnY3cjBuYWxxM28xZXVnNzE5dWVsajdfMjAyNDAxMDRUMDgwMDAwWiBhdXRvd2FyZS5vcmdfNmxvbDBobzVmdDAyMTdoOGM2MHBpMWZtMzBAZw)).

Kim-mins, 2023-12-26:
Thank you for the detailed investigation @maxime-clem!
Then I'll wait for it. Thanks.

mehmetdogru, 2024-01-18:
@brkay54 to test if any PR related slop compensation solves the issue or not

brkay54, 2024-01-31:
Hi @Kim-mins, I can reproduce the issue like Maxime did. I checked the slope values of the road and I saw that it was higher than 0.1 rad (which is max and min slope for pitch). When I changed it to a higher value (0.5 and -0.5). The vehicle can stop at the goal point. Could you try to change [these parameters](https://github.com/autowarefoundation/autoware_launch/blob/b7880fea1a03e86874a348ea6511024b08e3168c/autoware_launch/config/control/trajectory_follower/longitudinal/pid.param.yaml#L75-L76) to a higher value?

Kim-mins, 2024-02-05:
Hi @brkay54! Thank you for the answer!

Unfortunately, the ego vehicle still go through the goal point for me..
I changed the parameter `max_pitch_rad` from `0.1` to `0.5`, and `min_pitch_rad` from `-0.1` to `-0.5`, as you suggested, but I could not observe any difference.
I'm currently running the Autoware with the prebuilt version (docker, 20230715). Could you please tell me the version you are running?

maxime-clem, 2024-02-07:
I tested the solution of @brkay54 and it completely solves the issue. The version of Autoware you are using (`20230715`) is missing some fixes to the slope compensation feature and this is probably why changing the parameters does not work in your case.

Can you try switching to a more recent version of Autoware ? Otherwise you can try cherry picking some of the fix PRs (I think all PRs to the `pid_longitudinal_controller` since July: https://github.com/autowarefoundation/autoware.universe/pulls?q=is%3Apr+is%3Aclosed+slope+sort%3Aupdated-desc).

Kim-mins, 2024-02-07:
Thank you for the suggestion @maxime-clem!

I just realized that I can run the planning simulation of the latest version, and I could check the ego vehicle stops right at the goal point with the parameter @brkay54 provided.
So I think the issue is gone now! I really appreciate to the hard work!! Thank you all!

closed 2024-02-08 by Kim-mins