autowarefoundation/autoware_universe #5825: Avoidance module fails because of wrong drivable area generation

brkay54, 2023-12-09:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

While testing the avoidance module, we realized that avoidance generates non-valid paths in some conditions, you can see the behavior below: 

https://github.com/autowarefoundation/autoware.universe/assets/45468306/c7c353be-d733-4246-90a4-e14f4f28493e

After some investigation, this behavior is occurring on neighbor lanes. When the path passes the neighbor lanes, we can see that the maximum drivable area marker shows non-valid lines for the drivable area as below:

[image]
[image]

It can not close the drivable area correctly.

In areas like above, when the avoidance module needs to be run, it creates wrong drivable area lines for the condition below:

**The route does not include the neighbor lanes:**
- Place 1:
 
https://github.com/autowarefoundation/autoware.universe/assets/45468306/dfbeccda-3a68-43e4-9b3c-8b05bcd53fa3

- Place 2:

https://github.com/autowarefoundation/autoware.universe/assets/45468306/5447ae7d-8bed-44b3-aae3-d4a660b5aab1

It can create the drivable lane (blue) successfully, but it can not create the maximum drivable area properly.

**The route includes the neighbor lanes:**
- Place 1: 

https://github.com/autowarefoundation/autoware.universe/assets/45468306/cc9e72d4-c524-4b08-97c4-378b402b440d

- Place 2: 

https://github.com/autowarefoundation/autoware.universe/assets/45468306/6c09c545-b87c-4d48-b51d-256366acfb15

In place 2, we do not observe the failed behavior but we can see the drivable area is not proper. In place 1, we can see that the avoidance module creates a non-valid path because of the drivable area.

To reproduce:

[avoidance_scenario.tar.gz](https://github.com/autowarefoundation/autoware.universe/files/13625935/avoidance_scenario.tar.gz)

### Expected behavior

Autoware creates a proper drivable area and avoids the vehicle successfully. 

### Actual behavior

Autoware does not create a proper drivable area and fails to avoid the vehicle.

### Steps to reproduce

1. Download the scenario and map file from above
2. Run it with the latest Autoware

satoshi-ota, 2023-12-21:
@brkay54 Thanks for your report. And sorry for inconvenient. Since I can reproduce it, I'll investigate the cause of it.

[image]

satoshi-ota, 2023-12-21:
Hi @brkay54 @mehmetdogru I found two bugs in bpp.

1. output broken drivable bound.
2. output inconsistent drivable area marker.

The former one effects wierd vehicle behavior. Fortunatelly, I already could find the cause of it and guess this issue will be fixed by https://github.com/autowarefoundation/autoware.universe/pull/5928.

On the other hand, the latter one does NOT effect vehicle behavior because it is generated dependently and used for only visualization. It seems that this marker is not maintained recently.

@zulfaqar-azmi-t4 I think you added this feature in https://github.com/autowarefoundation/autoware.universe/pull/499. So, could you fix latter issue...? Or, I think we can remove it if nobody uses this marker.

zulfaqar-azmi-t4, 2023-12-21:
@brkay54 @mehmetdogru 
With the drivable area generation function getting complicated, at the moment, we dont use the visualization anymore. So I am considering to remove it. 

Therefore, before I make a PR to remove, just to confirm, did you use the the maximum drivable area visualization often, or do you think it is useful? if yes, we will think of a way to maintain the function. If not, I will proceed with removing it.

brkay54, 2023-12-21:
@zulfaqar-azmi-t4 Thank you, sure, you can remove it.

closed 2023-12-25 by satoshi-ota