autowarefoundation/autoware_universe #5938: motion velocity smoother generate invalid trajectory from freespace planner output:

VRichardJP, 2023-12-22:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

freespace_planner trajectory:
[image]

```bash
clear; ros2 topic echo /planning/scenario_planning/parking/trajectory --once
header:
  stamp:
    sec: 1703230235
    nanosec: 694643853
  frame_id: map
points:
- time_from_start:
    sec: 0
    nanosec: 0
  pose:
    position:
      x: 3704.4996056985747
      y: 73740.54148407953
      z: 0.0
    orientation:
      x: 0.0
      y: 0.0
      z: 0.8876844177650136
      w: 0.46045235850974725
  longitudinal_velocity_mps: -1.3888888359069824
  lateral_velocity_mps: 0.0
  acceleration_mps2: 0.0
  heading_rate_rps: 0.0
  front_wheel_angle_rad: 0.0
  rear_wheel_angle_rad: 0.0
- time_from_start:
    sec: 0
    nanosec: 0
  pose:
    position:
      x: 3704.4930889210755
      y: 73740.55103609574
      z: 0.0
    orientation:
      x: 0.0
      y: 0.0
      z: 0.8882163516239227
      w: 0.45942541582708296
  longitudinal_velocity_mps: -1.3888888359069824
  lateral_velocity_mps: 0.0
  acceleration_mps2: 0.0
  heading_rate_rps: 0.0
  front_wheel_angle_rad: 0.0
  rear_wheel_angle_rad: 0.0
- time_from_start:
    sec: 0
    nanosec: 0
  pose:
    position:
      x: 3704.5292234154686
      y: 73740.50098668145
      z: 0.0
    orientation:
      x: 0.0
      y: 0.0
      z: 0.8910379629250171
      w: 0.45392879246246953
[... 1273 lines cut ...]

It may be related to the "partial trajectory" logic implemented in the freespace planner.

I observe this behavior typically happens in the following situation:

[image]

where the freespace algorithm wants to drive slightly forward first in order to adjust the vehicle orientation, and then drive backward to the goal point. If the turn point happen to be close to the current location (based on `th_arrived_distance_m` parameter), then the freespace planner decide to skip the forward part and only send the backward trajectory. I have added a couple printf in the code to verify that. In particular I print for each point of the trajectory whether `is_back` is set (so here from index 0 to 4 we are supposed to drive forward), and the indices of the partial trajectory generated:

```
[component_container_mt-31] is_back: 000001111111111111111100000
[component_container_mt-31] indices: 4 21 
[component_container_mt-31] target_index_: 4
[component_container_mt-31] trajectory from 4 to 21
```

So here the freespace planner decided to skip the small forward section.

I am not sure why this would cause an issue for the velocity smoother:
- maybe because the trajectory starts in front the vehicle while the vehicle is asked to drive backward
- maybe because because without the forward section it becomes impossible to adjust the vehicle orientation properly

I am not sure this is the problem though, because I still get the issue when I reduce `th_arrived_distance_m`.

any idea?

VRichardJP, 2023-12-27:
One strange behavior I observe is that when the path is broken while the vehicle is asked to go reverse, the trajectory orientation is reversed.

Here the generation is correct:

[image]

Custom freespace planner trajectory:
```
header:
  stamp:
    sec: 1703640772
    nanosec: 956512242
  frame_id: map
points:
- time_from_start:
    sec: 0
    nanosec: 0
  pose:
    position:
      x: 3725.650146484375
      y: 73740.2265625
      z: 0.0
    orientation:
      x: 0.0
      y: 0.0
      z: 0.20226494505023526
      w: 0.979330838891447
  longitudinal_velocity_mps: -0.25
  lateral_velocity_mps: 0.0
  acceleration_mps2: 0.10000000149011612
  heading_rate_rps: 0.05014345049858093
  front_wheel_angle_rad: 0.0
  rear_wheel_angle_rad: 0.0
[...]
```

`motion_velocity_smoother` trajectory:
```
header:
  stamp:
    sec: 1703640773
    nanosec: 356973420
  frame_id: map
points:
- time_from_start:
    sec: 0
    nanosec: 0
  pose:
    position:
      x: 3725.650146484375
      y: 73740.2265625
      z: 0.0
    orientation:
      x: 0.0
      y: 0.0
      z: 0.20226494505023526
      w: 0.979330838891447
  longitudinal_velocity_mps: -0.6013365983963013
  lateral_velocity_mps: 0.0
  acceleration_mps2: -0.7235000133514404
  heading_rate_rps: 0.05264103040099144
  front_wheel_angle_rad: 0.0
  rear_wheel_angle_rad: 0.0
```

As you can see, orientation field is the same in both case. However when the path is broken I get this:

freespace planner trajectory:
```
header:
  stamp:
    sec: 1703640202
    nanosec: 83022253
  frame_id: map
points:
- time_from_start:
    sec: 0
    nanosec: 342219741
  pose:
    position:
      x: 3707.8560223578856
      y: 73733.6997291461
      z: 0.0
    orientation:
      x: 0.0
      y: 0.0
      z: 0.6741909033788862
      w: 0.7385571242640351
  longitudinal_velocity_mps: -0.1900017261505127
  lateral_velocity_mps: 0.0
  acceleration_mps2: -1.0888986587524414
  heading_rate_rps: 0.012745929881930351
  front_wheel_angle_rad: -0.0
  rear_wheel_angle_rad: 0.0
```

[image]

`motion_velocity_smoother` trajectory:
```
header:
  stamp:
    sec: 1703640201
    nanosec: 683058660
  frame_id: map
points:
- time_from_start:
    sec: 0
    nanosec: 342219741
  pose:
    position:
      x: 3707.8560223578856
      y: 73733.6997291461
      z: 0.0
    orientation:
      x: 0.0
      y: 0.0
      z: -0.7288375537869278
      w: 0.6846866583992178
  longitudinal_velocity_mps: -0.0
  lateral_velocity_mps: 0.0
  acceleration_mps2: 0.10000000149011612
  heading_rate_rps: 0.012745929881930351
  front_wheel_angle_rad: 0.0
  rear_wheel_angle_rad: 0.0
```

The position is exactly the same, but orientation has been reversed:

[image]

ralwing, 2025-10-24:
This problem still persists in the current version of Autoware. The trajectory generated by the `freespace planner` sometimes becomes invalid after smoothing.
Sample trajectory generated from the **lanedriving pipeline**:

[image]

The trajectory generated from the **freespace planner** is **not smoothed** and **contains forth-and-back x values**, which results in a *reverted yaw angle* when processed by the velocity smoother.
[image]

I wonder why the path_smoother and path_optimiser are not utilized in the freespace planner pipeline.