autowarefoundation/autoware_universe #5966: `route_handler` sometimes refuse goal on parking

VRichardJP, 2023-12-26:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

Example here on the sample map extended parking ([lanelet2_map.txt])(https://github.com/autowarefoundation/autoware.universe/files/13769175/lanelet2_map.txt)

The ego vehicle is somewhere on the parking and I try to set the goal somewhere else on the parking:

[Screencast from 12-26-2023 03_16_00 PM.webm](https://github.com/autowarefoundation/autoware.universe/assets/18645627/b7de3a26-abce-441d-adfc-c3a587fe29e1)

All the goals should be reachable and accepted but many are refused by the `route_handler` node. It looks like this mostly depend on the ego pose.

### Expected behavior

All goal set in the video above should be accepted

### Actual behavior

sometimes refused

### Steps to reproduce

planning simulator with a parking lot polygon (e.g. the one linked above.)

VRichardJP, 2023-12-26:
So the root cause is the way [`RouteHandler::planPathLaneletsBetweenCheckpoints`](https://github.com/autowarefoundation/autoware.universe/blob/1d6e7e82ffc948fcf8c8f325947eec1785259088/planning/route_handler/src/route_handler.cpp#L2091) is implemented or/and used in `default_planner`:

https://github.com/autowarefoundation/autoware.universe/blob/1d6e7e82ffc948fcf8c8f325947eec1785259088/planning/mission_planner/src/lanelet2_plugins/default_planner.cpp#L411-L415

The `planPathLaneletsBetweenCheckpoints` get the closest lanelets from start and goal poses and try to generate a path between the lanelets. If no path is found the goal is refused. The thing is that this does not makes sense at all if the vehicle is on a parking area, outside lanelets.

I think the `default_planner` should not use `RouteHandler::planPathLaneletsBetweenCheckpoints` if the start and goal poses happen to be on the same parking area. In the later case, the goal pose should be accepted for any initial/goal pose.

VRichardJP, 2023-12-26:
Quick and dirty fix by trying the planning up to 3 times:
1. from start to goal current lanelets (must be on lanelet)
2. from start to goal if on the same parking lot
3. from start to goal closest lanelets (legacy behavior)

This way lanelet based routing is prioritized if start/goal are both on lanelets and parking lots (like on the sample map), but "loose" routing (i.e. with the closest lanelet) is only tried if the parking "routing" is not possible.

[Screencast from 12-26-2023 05_20_56 PM.webm](https://github.com/autowarefoundation/autoware.universe/assets/18645627/8a6342e3-cc9e-4411-8ce7-449847a8234e)

There is still an issue if the vehicle is on a lanelet but not oriented properly.

I don't know what would be the clean way to support this (have route_handler consider parking lots in the routing?)

```diff
diff --git a/planning/mission_planner/src/lanelet2_plugins/default_planner.cpp b/planning/mission_planner/src/lanelet2_plugins/default_planner.cpp
index d074e9cc83..fda14e2212 100644
--- a/planning/mission_planner/src/lanelet2_plugins/default_planner.cpp
+++ b/planning/mission_planner/src/lanelet2_plugins/default_planner.cpp
@@ -411,10 +411,29 @@ PlannerPlugin::LaneletRoute DefaultPlanner::plan(const RoutePoints & points)
     const auto start_check_point = points.at(i - 1);
     const auto goal_check_point = points.at(i);
     lanelet::ConstLanelets path_lanelets;
+    // first try to find "strict" path from start to goal, i.e. for which start and goal poses are expected to be on lanelets
     if (!route_handler_.planPathLaneletsBetweenCheckpoints(
-          start_check_point, goal_check_point, &path_lanelets, param_.consider_no_drivable_lanes)) {
-      RCLCPP_WARN(logger, "Failed to plan route.");
-      return route_msg;
+          start_check_point, goal_check_point, &path_lanelets, param_.consider_no_drivable_lanes, true)) {
+      // otherwise check if both points belong to the same parking area
+      lanelet::BasicPoint2d start_point{start_check_point.position.x, start_check_point.position.y};
+      lanelet::BasicPoint2d goal_point{goal_check_point.position.x, goal_check_point.position.y};
+      const auto all_parking_lots = lanelet::utils::query::getAllParkingLots(route_handler_.getLaneletMapPtr());
+      lanelet::ConstPolygon3d start_parking_lot;
+      lanelet::ConstPolygon3d goal_parking_lot;
+      bool start_on_parking = lanelet::utils::query::getLinkedParkingLot(start_point, all_parking_lots, &start_parking_lot);
+      bool goal_on_parking = lanelet::utils::query::getLinkedParkingLot(goal_point, all_parking_lots, &goal_parking_lot);
+      if (start_on_parking || goal_on_parking || start_parking_lot.id() == goal_parking_lot.id()) {
+        // stay within the parking area
+        path_lanelets.clear();
+      } else {
+        // otherwise try to find a "non-strict" path from start to goal, i.e. for which start and goal poses may be outside lanelets. In such case the closest lanelet is used.
+        if (!route_handler_.planPathLaneletsBetweenCheckpoints(
+          start_check_point, goal_check_point, &path_lanelets, param_.consider_no_drivable_lanes, false)) {
+          // tried our best :(
+          RCLCPP_WARN(logger, "Failed to plan route.");
+          return route_msg;
+        }
+      }
     }
     for (const auto & lane : path_lanelets) {
       all_route_lanelets.push_back(lane);
@@ -455,6 +474,7 @@ PlannerPlugin::LaneletRoute DefaultPlanner::plan(const RoutePoints & points)
 geometry_msgs::msg::Pose DefaultPlanner::refine_goal_height(
   const Pose & goal, const RouteSections & route_sections)
 {
+  if (route_sections.empty()) return goal; // free goal
   const auto goal_lane_id = route_sections.back().preferred_primitive.id;
   lanelet::Lanelet goal_lanelet = lanelet_map_ptr_->laneletLayer.get(goal_lane_id);
   const auto goal_lanelet_pt = lanelet::utils::conversion::toLaneletPoint(goal.position);
diff --git a/planning/mission_planner/src/mission_planner/mission_planner.cpp b/planning/mission_planner/src/mission_planner/mission_planner.cpp
index a1536a1beb..d2db266cba 100644
--- a/planning/mission_planner/src/mission_planner/mission_planner.cpp
+++ b/planning/mission_planner/src/mission_planner/mission_planner.cpp
@@ -389,8 +389,9 @@ void MissionPlanner::on_set_route_points(
 
   // Check planned routes
   if (route.segments.empty()) {
-    throw component_interface_utils::ServiceException(
-      ResponseCode::ERROR_PLANNER_FAILED, "The planned route is empty.");
+    // throw component_interface_utils::ServiceException(
[... 102 lines cut ...]
+        is_route_found = true;
 
-      if (optional_route->length2d() < shortest_path_length2d) {
-        shortest_path_length2d = optional_route->length2d();
-        shortest_path = optional_route->shortestPath();
-        start_lanelet = st_llt;
+        if (optional_route->length2d() < shortest_path_length2d) {
+          shortest_path_length2d = optional_route->length2d();
+          shortest_path = optional_route->shortestPath();
+          start_lanelet = st_llt;
+          goal_lanelet = gl_llt;
+        }
       }
     }
   }
@@ -2225,6 +2235,7 @@ std::vector<LaneletSegment> RouteHandler::createMapSegments(
 lanelet::ConstLanelets RouteHandler::getMainLanelets(
   const lanelet::ConstLanelets & path_lanelets) const
 {
+  if (path_lanelets.empty()) return {}; // prevent crash on empty route
   auto lanelet_sequence = getLaneletSequence(path_lanelets.back());
   lanelet::ConstLanelets main_lanelets;
   while (!lanelet_sequence.empty()) {

```

iumida, 2025-05-15:
I would like to ask if this problem still exists?