autowarefoundation/autoware_universe #6027: `motion_velocity_smoother`: `stopping_distance` and `stopping_velocity` don't work anymore

VRichardJP, 2024-01-09:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

I have a feeling these parameters used to work in the past, but not anymore.

Here for example `stopping_distance` is set to 5 meters and `stopping_velocity` is set to 3.6 km/h, but Autoware seems not to slow down at all:

[Screencast from 01-09-2024 12_58_56 PM.webm](https://github.com/autowarefoundation/autoware.universe/assets/18645627/1fffa8e5-6a85-4d13-baf5-46114905887a)

### Expected behavior

With the parameters above, the vehicle velocity should be less than 3.6km/h 5 meters away from the goal pose.

### Actual behavior

stopping distance/velocity have no effect

### Steps to reproduce

planning simulator and set parameters with RQT reconfigure or in the yaml file.

brkay54, 2024-01-09:
Thank you for the report @VRichardJP,

When I looked at the debug trajectories, I saw that `applyStopApproachingVelocity()` function does not work as we wanted because it can not find stop_idx. 

While debugging, I saw that the input trajectory of the `applyStopApproachingVelocity()` function does not have zero velocity on the last point. Because there is no zero velocity on the last point, `applyStopApproachingVelocity()` function returns none.

When I set zero velocity to the last point manually, the function behaves as we want.
For stopping_velocity: 0.0, stopping_distance: 5.0 :

https://github.com/autowarefoundation/autoware.universe/assets/45468306/4c2c8811-0a47-49d3-9576-839baaa10e3f

Vehicle stops 5.0 meters before the goal point.

brkay54, 2024-01-11:
@VRichardJP After PR was merged, the issue was automatically closed. Sorry for that, it is my mistake. If this PR solved the problem, could you close the issue? If any problem arises, please let me know.

VRichardJP, 2024-01-11:
Yes, no problem!

closed 2024-01-11 by VRichardJP