autowarefoundation/autoware_universe #6073: xml launch `node_container` with `launch-prefix` sometimes does not start

VRichardJP, 2024-01-15:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

Really the same vibes than https://github.com/autowarefoundation/autoware.universe/issues/2940#issuecomment-1801203074, except here it's the container which is not started and not the components.

I want to run `motion_velocity_smoother` and `freespace_planner` in their own terminal for debugging, so I modified:

https://github.com/autowarefoundation/autoware.universe/blob/a4c92079cd1fcc80e4a9b78032a2be11ffdbb606/launch/tier4_planning_launch/launch/scenario_planning/scenario_planning.launch.xml#L27

and 

https://github.com/autowarefoundation/autoware.universe/blob/a4c92079cd1fcc80e4a9b78032a2be11ffdbb606/launch/tier4_planning_launch/launch/scenario_planning/parking.launch.xml#L6

by adding launch-prefix="gnome-terminal -- "`

When I run the planning simulator, sometimes the terminals appear and it works just fine, and sometimes they don't. When they don't, the containers are not visible from CLI either:

```
sig@vr-desktop:~/autoware$ ros2 component list /planning/scenario_planning/parking/parking_container
Unable to find container node '/planning/scenario_planning/parking/parking_container'
```

It is most likely a ROS2 problem but I have not been able to isolate it yet. That said, I often use `launch-prefix` to run stuff in separate terminals, and I have never had issues doing so with xml `<node>` or python based launch files. So maybe it only happens with `<node_container>` inside xml launch?

Can anyone else reproduce the problem

### Expected behavior

Everything launched

### Actual behavior

`launch-prefix`-ed containers sometimes missing.

### Steps to reproduce

Add `launch-prefix="gnome-terminal -- "` to misc `<node_container>` and run the autoware?

maxime-clem, 2024-01-30:
I am not encountering this issue and I am using `launch-prefix="konsole -e gdb -ex run --args"` with the `behavior_planning` node container.
Is it possible the issue only occurs with `gnome-terminal` ?

VRichardJP, 2024-02-15:
Can't reproduce anymore. Will close for now.

closed 2024-02-15 by VRichardJP