autowarefoundation/autoware_universe #6191: Failure of recording a scenario simulation

lethal233, 2024-01-27:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

The scenario simulator cannot perform normally when I set the record to true, which means:

it cannot do the scenario test simulation
it cannot record the simulation.

### Expected behavior

The scenario test simulator should run the scenario normally and record the whole simulation messages.

### Actual behavior

[image]

### Steps to reproduce

1. Run the scenario test simulation with the command `ros2 launch scenario_test_runner scenario_test_runner.launch.py   architecture_type:=awf/universe record:=false scenario:='t4v2.yml'   sensor_model:=sample_sensor_kit   vehicle_model:=sample_vehicle`
2. The scenario is simulated and played on the rviz.
3. Run the scenario test simulation again with the command `ros2 launch scenario_test_runner scenario_test_runner.launch.py   architecture_type:=awf/universe record:=true scenario:='t4v2.yml'   sensor_model:=sample_sensor_kit   vehicle_model:=sample_vehicle`
5. Rviz is successfully started, but the scenario is not simulated, along with console error output.

### Versions

 - OS: Ubuntu 22.04
 - ROS 2 version: humble

### Additional context

I raised an issue in https://github.com/tier4/scenario_simulator_v2/issues/1179 with no response. I think it is related to autoware.universe so I raise it here

lethal233, 2024-01-30:
With help of the tier4 developer, I am able to record and replay the scenario. But here are two additional issues:
1. No steering angle and speed displayed. 
[image]
2. When recording the scenario, here is some error message. It seems that ros bag cannot record the topic  '/perception/traffic_light_recognition/traffic_signals'
```sh
[ERROR 1706636596.198758876] [ROSBAG2_TRANSPORT]: Topic '/perception/traffic_light_recognition/traffic_signals' has more than one type associated. Only topics with one type are supported (has_single_type() at ./src/rosbag2_transport/topic_filter.cpp:45)
```

lethal233, 2024-01-31:
Here I encountered another problem: the rviz cannot be started

```sh
[rviz2-84] qt.qpa.xcb: could not connect to display :0
[rviz2-84] qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "" even though it was found.
[rviz2-84] This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.
[rviz2-84] 
[rviz2-84] Available platform plugins are: eglfs, linuxfb, minimal, minimalegl, offscreen, vnc, xcb.
[rviz2-84] 

```

maxime-clem, 2024-02-01:
It looks like you have an issue with your environment. Are you using a computer with a graphic card and are the drivers properly installed ?
I do not think it is related to the original issue, so if the problem with recording is fixed, I recommend closing this issue and opening a new one.

lethal233, 2024-02-01:
Oh I think I can close this issue. I forgot to update it. After I restarted my computer, the issue was resolved. Thanks!

closed 2024-02-01 by lethal233