autowarefoundation/autoware_universe #6192: confused TrackedObject‘s speed direction

xunkaixin, 2024-01-28:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

in map_based_prediction_node.cpp, the code
// assumption: the object vx is much larger than vy
 if (object.kinematics.twist_with_covariance.twist.linear.x >= 0.0) return; 
seems not correct, as the vx and vy is not aways >= 0  in  map/world coordinates system;
at same time, vx may not necessarily be the main direction

### Expected behavior

Clearer definition of coordinate systems

### Actual behavior

confused TrackedObject‘s coordinates system

### Steps to reproduce

see code about use of twist

miursh, 2024-01-29:
@YoshiRi 
Could you take a look at this?

YoshiRi, 2024-01-30:
Let me share the background:
- The object twist and acceleration are represented in the object frame (Not in map frame. Sorry, I think this information should written in somewhere.)
- `updateObjectData` is used to fix object orientation so that the planning module do not consider negative velocity

So the code intention is that normally vehicle lateral motion won't be faster than longitudinal one.
I think we need to add the custom message description to [somewhere in our document](https://autowarefoundation.github.io/autoware-documentation/release-v1.0_beta/design/autoware-architecture/perception/).

@xunkaixin 
Was this what you want to discuss about?

xunkaixin, 2024-01-30:
perfect, thanks. the document is important