autowarefoundation/autoware_universe #6204: BlindSpotModule setStopLineInx fail

ahmeddesokyebrahim, 2024-01-29:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

Fixes #5539 

Currently, we are working on a task for fixing and improving Autoware logging system as Autoware creates a noisy debugging environment, making it challenging to identify and address genuine issues.

For more details about the issue, plan how to tackle it, and linked PR, please have a look [here](https://github.com/autowarefoundation/autoware.universe/issues/5539).

For this bug, when using planning simulator, blind spot module throws a warning that is not able to set stop index by the stop pose.

This [video](https://youtu.be/UrVD9SdY_Bk) demonstrates the issue.

### Expected behavior

Blind spot module should be able to set stop line index by stop pose

### Actual behavior

Blind spot module is not able to set stop line index by stop pose

### Steps to reproduce

1. Open planning simulator
2. Plan vehicle start pose and goal pose in a way that blind spot module will be triggered, such as having turns like in this [video](https://youtu.be/UrVD9SdY_Bk)
3. Let Autoware drive autonomously and check for the warning message.

You may use the PRs mentioned [here](https://github.com/autowarefoundation/autoware.universe/issues/5539#issuecomment-1869696363) for having more comfortable logging messages, if you want.

soblin, 2024-02-15:
https://github.com/autowarefoundation/autoware.universe/pull/6410

soblin, 2024-02-15:
I fixed the blind_spot module in https://github.com/autowarefoundation/autoware.universe/pull/6405 and confirmed
- debug polygon appears
- it reacts to bicycles
- the debug print does not show up

mehmetdogru, 2024-02-15:
@ahmeddesokyebrahim Could you please confirm if the issue is resolved with @soblin 's PR so that we could close the issue.

ahmeddesokyebrahim, 2024-02-19:
I confirm that the warning message is no longer logged using planning simulator.
I will close the issue.
cc : @mehmetdogru @soblin

closed 2024-02-19 by ahmeddesokyebrahim