autowarefoundation/autoware_universe #6244: Pose initialization has become slow with dynamic map loading

VRichardJP, 2024-01-31:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

When I use the planning simulator with default parameters (i.e. with dynamic map loading) and a 250MB PCD map, pose initialization with Rviz takes around 6-7 seconds:

[Screencast from 01-31-2024 01_27_45 PM.webm](https://github.com/autowarefoundation/autoware.universe/assets/18645627/47ef2ba8-2b43-4f28-8a09-d757ac8c7553)

When I use the same setup but with the dynamic map loading disabled (see below), pose initialization is done instantly:

[Screencast from 01-31-2024 01_30_36 PM.webm](https://github.com/autowarefoundation/autoware.universe/assets/18645627/f4d698ab-9775-4c69-a76b-3535bac89af7)

- `autoware_launch/config/map/pointcloud_map_loader.param.yaml`:
  - `enable_partial_load: false`
- `autoware_launch/config/perception/object_recognition/detection/pointcloud_map_filter.param.yaml`:
  - `use_dynamic_map_loading: False`

I guess what happens is the PCD map is reloaded every single time the pose is initialized, despite being the exact same map. AFAIK, this only impacts pose initialization so it is not a huge issue at the moment, but I think current approach may impact other modules as well.

One simple way to improve this would be to split the map loading into 2 queries:
1. a query to get the chunk ID at given position, or all chunk IDs within given radius. This one should be fairly fast
2. a query to retrieve chunks by ID. Obvi

In such case, nodes using dynamic map loading feature would simply need to track which chunk ID is currently stored in their own cache.

### Expected behavior

No delay on pose initialization

### Actual behavior

4-5 seconds latency

### Steps to reproduce

planning simulator with "big" PCD map

isamu-takagi, 2024-01-31:
This is because the whole map is transferred when correcting the Z coordinate. The pose from rviz is not accurate in height, so the [initial_pose_adaptor](https://github.com/autowarefoundation/autoware.universe/blob/main/system/default_ad_api_helpers/ad_api_adaptors/src/initial_pose_adaptor.cpp#L51) uses [map_height_fitter](https://github.com/autowarefoundation/autoware.universe/blob/main/map/map_height_fitter/src/map_height_fitter.cpp#L105) to fix it.

Since dynamic_map_loading feature sends point clouds in file units, it is necessary to split the point cloud file in advance when using this function. Or I don't know if it's possible, but I think it can be solved by implementing a function in the map_loader that automatically splits a single file.

VRichardJP, 2024-04-03:
@isamu-takagi @Motsu-san 
If the point cloud is split into chunks, would map height fitter only fetch the most relevant chunk? 

The PCD I used happened to be very dense. Down sampling the map reduced the service time significantly, but it still remained slower than the previous design.

VRichardJP, 2024-04-03:
Thinking back at my cache idea: if both the map loader and height fitting nodes are deployed in the some container, I think it should be possible to write the code in a way the map could be accessed by the height fitter node using intra process (so no copy or in the worse case 1 copy when the message is created, which shouldn't take long)

isamu-takagi, 2024-04-03:
>If the point cloud is split into chunks, would map height fitter only fetch the most relevant chunk?

Yes. When the map is splitted according to [this prerequisites](https://autowarefoundation.github.io/autoware.universe/main/map/map_loader/#prerequisites), only the nearest chunks will be requested.

> Thinking back at my cache idea: if both the map loader and height fitting nodes are deployed in the some container, I think it should be possible to write the code in a way the map could be accessed [...]

Since the service does not seem to support intra-process communication, I don't think the processing time will change if dynamic loading is enabled. The most efficient way is for the map_loader to have the function of the map_height_fitter, but I'm not sure if it's a good idea from the perspective of dividing packages.