autowarefoundation/autoware_universe #6296: missing parameters of perception.launch.xml called from tier4_perception_component.launch.xml

maitelorente, 2024-02-02:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

I just updated Autoware to the current version. When running autoware with "ros2 launch autoware_launch e2e_simulator.launch.xml map_path:=$HOME/my_workspaces/my_project_ws/maps/carla vehicle_model:=sample_vehicle sensor_model:=sample_sensor_kit", I get this error message: [ERROR] [launch]: Caught exception in launch (see debug for traceback): Included launch description missing required argument 'object_recognition_detection_roi_cluster_fusion_param_path' (description: 'no description given')...

I have checked the perception.launch.xml file and apart from object_recognition_detection_roi_cluster_fusion_param_path, these other parameters are also missing: 
object_recognition_detection_roi_pointcloud_fusion_param_path
object_recognition_detection_roi_detected_object_fusion_param_path
object_recognition_detection_radar_lanelet_filtering_range_param_path
object_recognition_detection_radar_crossing_objects_noise_filter_param_path
object_recognition_detection_object_velocity_splitter_radar_fusion_param_path
object_recognition_detection_object_range_splitter_radar_param_path
object_recognition_detection_object_range_splitter_radar_fusion_param_path

Also, when trying to launch other examples like "ros2 launch autoware_launch planning_simulator.launch.xml map_path:=$HOME/autoware_map/sample-map-planning vehicle_model:=sample_vehicle sensor_model:=sample_sensor_kit", the same error occurs.

I do not know which values they should have as default values or if i have missed something from the update. Please, could anyone advise me on this matter?

Thanks!

### Expected behavior

Being able to launch "ros2 launch autoware_launch e2e_simulator.launch.xml map_path:=$HOME/my_workspaces/my_project_ws/maps/carla vehicle_model:=sample_vehicle sensor_model:=sample_sensor_kit", and having autoware running.

### Actual behavior

I do not get to launch autoware from the launch file (autoware.launch.xml) provided in the repository code.

### Steps to reproduce

Update Autoware following the step in the documentation page (https://autowarefoundation.github.io/autoware-documentation/main/installation/autoware/source-installation/), and launch autoware.

maitelorente, 2024-02-06:
The problem was that the repository of autoware_launch package had not been updated correctly, now it is solved. Sorry for the inconvenience!

cyn-liu, 2024-02-07:
> The problem was that the repository of autoware_launch package had not been updated correctly, now it is solved. Sorry for the inconvenience!
@maitelorente 
Are you sure the problem already solved?
I update my code, but when I run the command:
`
ros2 launch autoware_launch planning_simulator.launch.xml map_path:=../sample-map-planning/ vehicle_model:=sample_vehicle sensor_model:=sample_sensor_kit
`
I get same error:
```
[INFO] [launch]: All log files can be found below /root/.ros/log/2024-02-07-16-41-57-742203-cynthia-liu-278041
[INFO] [launch]: Default logging verbosity is set to INFO
[ERROR] [launch]: Caught exception in launch (see debug for traceback): Included launch description missing required argument 'crop_box_filter_measurement_range_param_path' (description: 'no description given'), given: [map_path, map_path, vehicle_model, sensor_model, data_path, planning_module_preset, launch_sensing, launch_localization, launch_perception, use_pointcloud_container, launch_vehicle_interface, system_run_mode, launch_system_monitor, launch_dummy_diag_publisher, lanelet2_map_file, pointcloud_map_file, rviz, rviz_config, rviz_respawn, enable_all_modules_auto_mode, use_sim_time, vehicle_model, use_multithread, container_name, vehicle_model, sensor_model, vehicle_id, launch_vehicle_interface, config_dir, ...
```

cyn-liu, 2024-02-07:
I no idea about this error, how to solve it? @maitelorente  if you could help me, I would greatly appreciate it.

maitelorente, 2024-02-09:
Hi @cyn-liu,

Did you manage to solve the error? I have checked that the parameter crop_box_filter_measurement_range_param_path is required from launch file localization.launch.xml in package tier4_localization_launch, which is called from tier4_localization_component.launch.xml in package autoware_launch. You may need to add the argument in tier4_localization_component.launch.xml  or if you just updated your repository, as in my case, maybe the autoware_launch package could not properly update. To solve the problem, I did "git checkout" to discard any changes I had made in the original repository (in my case, the configuration file for RViz) and "git pull". Then, the autoware_launch package was updated correctly and the problem solved.

cyn-liu, 2024-02-09:
> Hi @cyn-liu,
>
> Did you manage to solve the error? I have checked that the parameter crop_box_filter_measurement_range_param_path is required from launch file localization.launch.xml in package [...]

thanks for your reply, I no idea to solve this error?

idorobotics, 2024-02-13:
cc: @xmfcx

xmfcx, 2024-02-13:
Please make sure you updated everything. For these kind of issues, my recommended route:

```bash
cd autoware
git pull # make sure it succeeds and says "Already up to date."
```

Pull the sub repositories to latest:
```bash
vcs pull src
```

The output should look like:

```bash
mfc@mfc-leo:~/projects/autoware$ vcs pull src
............................
=== src/core/autoware.core (git) ===
Already up to date.
=== src/core/autoware_adapi_msgs (git) ===
Already up to date.
=== src/core/autoware_common (git) ===
Already up to date.
=== src/core/autoware_msgs (git) ===
Already up to date.
=== src/core/external/autoware_auto_msgs (git) ===
Already up to date.
=== src/launcher/autoware_launch (git) ===
Already up to date.
=== src/param/autoware_individual_params (git) ===
Already up to date.
=== src/sensor_component/external/nebula (git) ===
Already up to date.
=== src/sensor_component/external/sensor_component_description (git) ===
Already up to date.
=== src/sensor_component/external/tamagawa_imu_driver (git) ===
Already up to date.
=== src/sensor_component/external/transport_drivers (git) ===
Already up to date.
=== src/sensor_kit/external/awsim_sensor_kit_launch (git) ===
Already up to date.
=== src/sensor_kit/sample_sensor_kit_launch (git) ===
Already up to date.
=== src/simulator/scenario_simulator (git) ===
Already up to date.
=== src/universe/autoware.universe (git) ===
Already up to date.
=== src/universe/external/ament_cmake (git) ===
Already up to date.
=== src/universe/external/data_tamer (git) ===
Already up to date.
=== src/universe/external/eagleye (git) ===
Already up to date.
=== src/universe/external/llh_converter (git) ===
Already up to date.
=== src/universe/external/morai_msgs (git) ===
Already up to date.
=== src/universe/external/muSSP (git) ===
Already up to date.
=== src/universe/external/ndt_omp (git) ===
Already up to date.
=== src/universe/external/pointcloud_to_laserscan (git) ===
Already up to date.
=== src/universe/external/rtklib_ros_bridge (git) ===
Already up to date.
=== src/universe/external/tier4_ad_api_adaptor (git) ===
Already up to date.
=== src/universe/external/tier4_autoware_msgs (git) ===
Already up to date.
=== src/vehicle/external/pacmod_interface (git) ===
[... 124 lines cut ...]
```

Here you want to see "Already on" for each of them, make sure you are on the recommended branches.

Then clean your workspace:

```bash
rm -rf build install log
```

Then to clear up your shell environment variables, open up a new terminal tab with `CTRL+Shift+T` and close the current terminal tab (you will still be on the same directory).

Update your system and rosdep and then install rosdep dependencies if they are missing.

```bash
sudo apt update
sudo apt dist-upgrade # always be careful when upgrading your packages
rosdep update
rosdep install -y --from-paths src --ignore-src --rosdistro humble
```

Then build the Autoware and hopefully it should work:
```bash
colcon build --symlink-install  --cmake-args -DCMAKE_BUILD_TYPE=RelWithDebInfo -DCMAKE_EXPORT_COMPILE_COMMANDS=1
```

closed 2024-02-19 by cyn-liu