autowarefoundation/autoware_universe #6348: ML-based Motion Prediction

ktro2828, 2024-02-07:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

Integrate ML-based motion prediction to the Autoware perception stack to improve the reliability and accuracy of the predicted paths.

Current rule-based prediction can only estimate paths aligned to the center line of the road for vehicles or straight paths for other traffic participants (e.g. pedestrian, cyclists).

ML-based module allow us to predict their future behaviors by considering observed state and their interactions.

### Purpose

To improve accuracy, adaptability and be able to handle uncertanity of future behaviors of traffic participants.

### Possible approaches

Now we are trying to integrate MTR(Motion TRnsformer), which achieved the 1st place for the motion prediction task in 2022 Waymo Open Dataset Challege.
- Paper: https://arxiv.org/abs/2209.13508
- Original repo: https://github.com/sshaoshuai/MTR

### Definition of done

- Implementation to Autoware

sezan92, 2024-02-07:
is it possible to work on this ?

ktro2828, 2024-02-07:
@sezan92 Thanks for your interest. I've finished creating the training pipeline of MTR. Now I'm working on the task to support 
 inference with TensorRT, and integrate to Autoware as ROS 2 node.

sezan92, 2024-02-07:
> @sezan92 Thanks for your interest. I've finished creating the training pipeline of MTR. Now I'm working on the task to support inference with TensorRT, and integrate to Autoware as ROS 2 node.

damn, missed it! :(

zehranrgi, 2024-03-21:
Hi, what was your steps to integrate another model to autoware?