autowarefoundation/autoware_universe #6373: tracking_object_merger continues to publish old detected object when the number of detected objects are zero

brkay54, 2024-02-09:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

While testing the perception pipeline, I faced a weird behavior. If the detected_objects is empty, the perception pipeline continues to publish old detected objects all the time.

In my test, I am publishing the lidars pointclouds to the pipeline. There are 2 different pointcloud messages: empty area pointcloud (pointcloud A) and pointcloud with an object in front of the ego (pointcloud B).

In the test, I started to publish pointcloud A to the perception pipeline and I can see there are no detected, predicted, or tracking objects. After a while, I started to publish pointcloud B and we saw the detected and predicted objects in the topics.

After that, I started to publish pointcloud A again, concat pointcloud was changed, and I can't see any objects in DetectedObjects topics, however, PredictedObjects is still publishing the old object.

https://github.com/autowarefoundation/autoware.universe/assets/45468306/c8b01da4-0a30-4ffe-85da-8ea629433f5b

As you can see in the video above, when we published pointcloud A after pointcloud B, the Predicted Object was not removed.

After some investigation, I realized `tracking_object_merger` is publishing the old detected objects although there are no detected objects in the pipeline.

In [here](https://github.com/autowarefoundation/autoware.universe/blob/main/perception/tracking_object_merger/src/decorative_tracker_merger.cpp#L253-L255), if there are no detected_objects, it returns false without any operation and it continues to publish the old object. Therefore, `map_based_prediction` continues to publish an object.

After I removed the protection of the empty objects, it worked as we wanted as shown below:

https://github.com/autowarefoundation/autoware.universe/assets/45468306/66d27a6d-7cfa-4e8c-952f-4d90c36c2a43

### Expected behavior

It should publish empty objects after detected objects are removed.

### Actual behavior

It continues to publish the old detected object as a tracking object although it was removed.

### Steps to reproduce

-

YoshiRi, 2024-02-11:
Thanks for your report. This issue is related with tracklet management problem.

### About quick fix

> After I removed the protection of the empty objects, it worked as we wanted as shown below:

I agree the short-term solution mentioned above. By removing the following section, we can update the survival probability even if the main topic is empty. This change requires minimal code modifications and should suffice as a solution for the current issue.

```cpp
if (main_topic_objects->objects.empty())
{
  return false;
}
```

### Additional solution
Ideally, a sanity check of the tracklets should be performed before issuing an object, ensuring a more comprehensive management of tracklet survival without omissions.

Fortunately, there is a parameter named `max_dt`, which can be utilized to manage the survival of tracklets before publishing. I will add code for this purpose as a precaution.

I will try to add some sanitizer around following part:

https://github.com/autowarefoundation/autoware.universe/blob/1121f3b2a5a298ac474835522b97e84d9301dd9a/perception/tracking_object_merger/src/utils/tracker_state.cpp#L274

### About future solution

There's a fundamental issue with entrusting the update() function with tracklet survival management, which should be re-evaluated in the long term. Currently, the survival management of tracklets directly manipulates the existence_probability, which may not be ideal as the number of topics to merge increases, requiring changes to this value each time.

A less disruptive proposal involves not directly manipulating the existence_probability. Instead, for survival management, the last_updated_time could be referenced to determine survival or demise based on a decayed value.

I've shared my thoughts, but I'm open to any opinions you might have.

YoshiRi, 2024-02-13:
@brkay54 Hi, could you test this [PR](https://github.com/autowarefoundation/autoware.universe/pull/6386) when you have time?

YoshiRi, 2024-02-13:
Thanks for your quick response! :+1:

closed 2024-02-13 by YoshiRi