autowarefoundation/autoware_universe #6378: system_error_monitor Error Messages while Parking

ahmeddesokyebrahim, 2024-02-12:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

Fixes #5539 

Currently, we are working on a task for fixing and improving Autoware logging system as Autoware creates a noisy debugging environment, making it challenging to identify and address genuine issues.

For more details about the issue, plan how to tackle it, and linked PR, please have a look [here](https://github.com/autowarefoundation/autoware.universe/issues/5539).

For this bug, when using planning simulator for a parking scenario, Autoware throws some error messages relevant to `system_error_montior`: 
- [system_error_monitor /autoware/planning/node_alive_monitoring]: [Single Point Fault]: Error (logThrottledNamed() at ~/autoware/src/universe/autoware.universe/system/system_error_monitor/src/system_error_monitor_core.cpp:54)
- [system_error_monitor /autoware/planning/node_alive_monitoring/topic_status/topic_state_monitor_mission_planning_route: planning_topic_status]: [Single Point Fault]: OK (logThrottledNamed() at ~/autoware/src/universe/autoware.universe/system/system_error_monitor/src/system_error_monitor_core.cpp:54
- [system_error_monitor /autoware/planning/node_alive_monitoring/topic_status/topic_state_monitor_scenario_planning_trajectory: planning_topic_status]: [Single Point Fault]: Error (logThrottledNamed() at ~/autoware/src/universe/autoware.universe/system/system_error_monitor/src/system_error_monitor_core.cpp:54)

This [video](https://youtu.be/UgCDS1_il0c) demonstrates the issue.

### Expected behavior

Autoware system_error_monitor should not having error messages while parking

### Actual behavior

Autoware system_error_monitor has error messages while parking

### Steps to reproduce

1. Open planning simulator
2. Plan vehicle start pose and goal pose achieving parking scenario like in this [video](https://youtu.be/UgCDS1_il0c)
3. Let Autoware drive autonomously and check for the error messages for `system_error_monitor`

You may use the PRs mentioned [here](https://github.com/autowarefoundation/autoware.universe/issues/5539#issuecomment-1869696363) for having more comfortable logging messages, if you want.

mehmetdogru, 2024-02-15:
@mitsudome-r to make a quick investigation on the issue.

ahmeddesokyebrahim, 2024-02-29:
@ito-san : I would highly appreciate if you can have a look and support in this issue or let us know who is the current package owner for `system_error_monitor` that can support here.

PR #6696 (2024-03-28): fix(freespace_planner): fix parking trajectory errors and warnings when parking completed
## Description
fixes #6380 
fixes #6378 
fixes #6381 

## Related links

This PR is one of group of PRs that aim to fix Autoware logging system to achieve the goal of reducing excessive error and warning logs on Autoware launch. 

Part of:
- https://github.com/autowarefoundation/autoware.universe/issues/5539

## Related links
More details in the [issue](https://github.com/autowarefoundation/autoware.universe/issues/5539).

## Tests performed
- Before this PR : https://youtu.be/UgCDS1_il0c
- After this PR  [... cut]
closed 2024-03-28 by ahmeddesokyebrahim