autowarefoundation/autoware_universe #6379: motion_velocity_smoother Error Message while Parking

ahmeddesokyebrahim, 2024-02-12:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I'm convinced that this is not my fault but a bug.

### Description

Fixes #5539 

Currently, we are working on a task for fixing and improving Autoware logging system as Autoware creates a noisy debugging environment, making it challenging to identify and address genuine issues.

For more details about the issue, plan how to tackle it, and linked PR, please have a look [here](https://github.com/autowarefoundation/autoware.universe/issues/5539).

For this bug, when using planning simulator for a parking scenario, Autoware throws some error messages relevant to `motion_velocity_smoother`: 
- [planning.scenario_planning.motion_velocity_smoother]: input trajectory size must > 1. Skip computation. (checkData() at ~/autoware/src/universe/autoware.universe/planning/motion_velocity_smoother/src/motion_velocity_smoother_node.cpp:427)

This [video](https://youtu.be/UgCDS1_il0c) demonstrates the issue.

### Expected behavior

Autoware `motion_velocity_smoother` should not having error message while parking

### Actual behavior

Autoware `motion_velocity_smoother` has error message while parking

### Steps to reproduce

1. Open planning simulator
2. Plan vehicle start pose and goal pose achieving parking scenario like in this [video](https://youtu.be/UgCDS1_il0c)
3. Let Autoware drive autonomously and check for the error message for `motion_velocity_smoother`

You may use the PRs mentioned [here](https://github.com/autowarefoundation/autoware.universe/issues/5539#issuecomment-1869696363) for having more comfortable logging messages, if you want.

mehmetdogru, 2024-02-15:
It seems even though this error is published, actions continue. 

@mehmetdogru to contact the code owner or responsible person to this code.

ahmeddesokyebrahim, 2024-02-29:
@rej55  I would highly appreciate if you can have a look and support in this issue.

closed 2024-05-09 by ahmeddesokyebrahim